node-opcua/node-opcua #421: More complete typescript for ClientSession methods

crussell52, 2018-01-29:
**I'm submitting a ...**  (check one with "x")

 - ( ) bug report => search github for a similar issue or PR before submitting
 - (x) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

Code-hints and code inspections against `ClientSession` methods have varied success.  The two which I've run into are:

 * No entry for Promise signature of `ClientSession.write()` (causes "Invalid number of paramters" in IntelliJ IDEs).
 * No entry for any variation of `ClientSession.readVariableValue()`

**Expected behavior**

All signatures for all ClientSession methods would be defined within node-opcua.d.ts

  
**What is the motivation / use case for changing the behavior?**

Simpler development with the aid of improved code hinting.  Easier to see real problems via fewer false-positives w/ code-inspections.

**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - (x) MacOs   : version : 10.13.2 
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:** ``

v0.2.0
* **Node:** 
   `node --version` = 8.9.4

erossignon, 2018-01-31:
@crussell52, can you confirm that this is now fixed with node-opcua@0.2.1 ?

crussell52, 2018-01-31:
@erossignon Confirmed, thank you. 👍

closed 2018-01-31 by erossignon

hdnn, 2019-04-15:
It is still incomplete TypeScript declaration of ClientSession. I am strongly need the getBuiltInDataType function. But it's not in node-opcua.d.ts file.