node-opcua/node-opcua #425: can't connect uaAnsiC with security SIGN Basic256

zombinary, 2018-02-06:
I using the example 'simple_client.js' to connect the uaAnsiC Server with security options.
But the 'CreateSessionRequest()' is responsed with an 'ServiceFault'.
With the security option '-s=NONE -P=None' the server works fine.

For a fix test from the server i used the UAExpert with the settings Basic256 and SIGN and it run without problems.

```
> $ node simple_client.js -e=opc.tcp://127.0.0.1:48020 -s=SIGN -P=Basic256
> ... BadCertificateUseNotAllowed (0x80180000)
```

The same script works with the option -s=SIGN -P=Basic256 and a connection to the KepServer and Prosys UA Simulation Server. 

All programms run on the same machine (win7).
node-opcua -v --> 0.1.1-0
Unified Automation UA SDK ANSI C Server Professional 1.7.0 (Binary  Edition)

I have no idea whats going wrong, the pcap file looks good also the Certificate?
Is ther a problem within the certificate?

erossignon, 2018-02-11:
Please make sure that your client certificate appears in the trusted  certificate folder of your server.

By default, UAC Demo server rejects certificate by adding them the 
```C:\ProgramData\UnifiedAutomation\UaAnsiCServer\pki\rejected``` folder.
You need to move trusted certificate to 
```C:\ProgramData\UnifiedAutomation\UaAnsiCServer\pki\trusted```

zombinary, 2018-02-13:
I'm sure that the Certificate is moved to the trusted\cert folder. 
I cleared the reject and trusted folder and test the client again. The client print out the same error.
Bevore the certificate ist trusted i get the error...

```
ERROR RECEIVED FROM SENDER BadSecurityChecksFailed (0x80130000) null
...
disconnected
client : process terminated with an error
error Error: _socket has been disconnected by third party

```
... after moving the certification the error is 

```
Options =  SIGN Basic256
 reconnecting to  opc.tcp://127.0.0.1:48020
 disconnected
 client : process terminated with an error
 error { Error:  ServiceFault returned by server { /*ServiceFault*/
 responseHeader                /* ResponseHeader                   */: {
   timeStamp                   /* UtcTime                          */: 2018-02-1
3T13:08:33.996Z
   requestHandle               /* IntegerId                        */: 3
       0x3
   serviceResult               /* StatusCode                       */: BadCertif
icateUseNotAllowed (0x80180000)
   serviceDiagnostics          /* DiagnosticInfo                   */: {
     namespaceUri              /* Int32                            */: -1
     symbolicId                /* Int32                            */: -1
     locale                    /* Int32                            */: -1
     localizedText             /* Int32                            */: -1
     additionalInfo            /* String                           */: null
     innerStatusCode           /* StatusCode                       */: Good (0x0
0000)
     innerDiagnosticInfo       /* DiagnosticInfo                   */: {
     }
   }
   stringTable                 /* String                        [] */: [ ] (l=1)

   additionalHeader            /* ExtensionObject                  */: null
 }
};
    at process_request_callback (D:\workspace\visualStudio2015\opcua-client\node
-opcua-0.1.1-0\packages\node-opcua-samples\node_modules\node-opcua-secure-channe
l\src\client\client_secure_channel_layer.js:67:15)
    at ClientSecureChannelLayer._on_message_received (D:\workspace\visualStudio2
015\opcua-client\node-opcua-0.1.1-0\packages\node-opcua-samples\node_modules\nod
e-opcua-secure-channel\src\client\client_secure_channel_layer.js:122:5)
    at emitMany (events.js:147:13)
    at MessageBuilder.emit (events.js:224:7)
    at MessageBuilder._decode_message_body (D:\workspace\visualStudio2015\opcua-
client\node-opcua-0.1.1-0\packages\node-opcua-samples\node_modules\node-opcua-se
cure-channel\src\message_builder.js:442:22)
    at MessageBuilderBase._feed_messageChunk (D:\workspace\visualStudio2015\opcu
a-client\node-opcua-0.1.1-0\packages\node-opcua-samples\node_modules\node-opcua-
transport\src\message_builder_base.js:192:18)
    at PacketAssembler.<anonymous> (D:\workspace\visualStudio2015\opcua-client\n
ode-opcua-0.1.1-0\packages\node-opcua-samples\node_modules\node-opcua-transport\
src\message_builder_base.js:52:14)
    at emitOne (events.js:116:13)
    at PacketAssembler.emit (events.js:211:7)
    at PacketAssembler.feed (D:\workspace\visualStudio2015\opcua-client\node-opc
ua-0.1.1-0\packages\node-opcua-samples\node_modules\node-opcua-packet-assembler\
src\packet_assembler.js:105:14)
  response:
   ServiceFault {
     responseHeader:
      ResponseHeader {
        timeStamp: 2018-02-13T13:08:33.996Z,
[... 29 lines cut ...]
    at process_request_callback (D:\workspace\visualStudio2015\opcua-client\node
-opcua-0.1.1-0\packages\node-opcua-samples\node_modules\node-opcua-secure-channe
l\src\client\client_secure_channel_layer.js:67:15)
    at ClientSecureChannelLayer._on_message_received (D:\workspace\visualStudio2
015\opcua-client\node-opcua-0.1.1-0\packages\node-opcua-samples\node_modules\nod
e-opcua-secure-channel\src\client\client_secure_channel_layer.js:122:5)
    at emitMany (events.js:147:13)
    at MessageBuilder.emit (events.js:224:7)
    at MessageBuilder._decode_message_body (D:\workspace\visualStudio2015\opcua-
client\node-opcua-0.1.1-0\packages\node-opcua-samples\node_modules\node-opcua-se
cure-channel\src\message_builder.js:442:22)
    at MessageBuilderBase._feed_messageChunk (D:\workspace\visualStudio2015\opcu
a-client\node-opcua-0.1.1-0\packages\node-opcua-samples\node_modules\node-opcua-
transport\src\message_builder_base.js:192:18)
    at PacketAssembler.<anonymous> (D:\workspace\visualStudio2015\opcua-client\n
ode-opcua-0.1.1-0\packages\node-opcua-samples\node_modules\node-opcua-transport\
src\message_builder_base.js:52:14)
    at emitOne (events.js:116:13)
    at PacketAssembler.emit (events.js:211:7)
    at PacketAssembler.feed (D:\workspace\visualStudio2015\opcua-client\node-opc
ua-0.1.1-0\packages\node-opcua-samples\node_modules\node-opcua-packet-assembler\
src\packet_assembler.js:105:14)
Exiting

```

zombinary, 2018-02-13:
i get the same error with version 0.2.0 and 0.2.1

erossignon, 2018-02-13:
Interesting ... it works with UACPPDemoServer and fail with UACServer 
The interesting bit is 
```BadCertificateUseNotAllowed (0x80180000)````

erossignon, 2019-05-05:
Investigation: BadCertificateUseNotAllowed
https://apps.opcfoundation.org/profilereporting/testcases/Server/SecurityCertificateValidation.html

029 | Lab CTT | Unavailable  OpenSecureChannel() | Connect a Client that uses a CA certificate – which is intended for issuing certificates and not for use as an Application Instance Certificate. | ServiceResult = BadSecurityChecksFailed, but the Server's log shows BadCertificateUseNotAllowed
-- | -- | -- | -- | --

Looking at self-signed certificate generated by node-opcua-pki:

```
$ openssl x509 -in client_selfsigned_cert_1024.pem -text | more
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            ae:4a:da:13:7d:3f:5c:b1
        Signature Algorithm: sha256WithRSAEncryption
        Issuer: C = FR, ST = IDF, L = Paris, O = NodeOPCUA, CN = NodeOPCUA-TEST
        Validity
            Not Before: May  5 08:44:19 2019 GMT
            Not After : May  1 08:44:19 2034 GMT
        Subject: C = FR, ST = IDF, L = Paris, O = NodeOPCUA, CN = NodeOPCUA-TEST
        Subject Public Key Info:
            Public Key Algorithm: rsaEncryption
                RSA Public-Key: (2048 bit)
                Modulus:
                    00:d9:b4:2e:ad:96:a2:12:33:f3:40:53:61:01:e7:
                    ...
                    e4:d7
                Exponent: 65537 (0x10001)
        X509v3 extensions:
            X509v3 Subject Key Identifier:
                67:2A:58:D7:E9:5C:20:9D:02:13:6D:9D:B0:36:31:FC:0F:6C:7F:80
            X509v3 Authority Key Identifier:
                keyid:67:2A:58:D7:E9:5C:20:9D:02:13:6D:9D:B0:36:31:FC:0F:6C:7F:80
                DirName:/C=FR/ST=IDF/L=Paris/O=NodeOPCUA/CN=NodeOPCUA-TEST
                serial:AE:4A:DA:13:7D:3F:5C:B1

            X509v3 Basic Constraints:
                CA:TRUE
            X509v3 Key Usage: critical
                Certificate Sign, CRL Sign
            Netscape Comment:
```

X509v3 Key Usage: critical usage should be
```
Digital Signature, Non Repudiation, Key Encipherment, Data Encipherment, Key Agreement, Certificate Sign
```
and not
```
   Certificate Sign, CRL Sign
```

erossignon, 2019-05-06:
issue is now fixed by node-opcua-pki@1.5.0, certificates may need to be regenerated ...

closed 2019-05-06 by erossignon

samminen, 2019-05-13:
Hello Erossignon,
   I am facing same connectivity issue when I use below secure connectivity:
                "OPC_MESSAGE_SECURITY_MODE":"3",
                "OPC_CLIENT_SECURITY_POLICY": "Basic256Sha256",
When I start my OPC UA client first time, it connects to OPC server and all good. But If I restart OPC server then client is not connected to server on reconnect and getting below error client side:
***********************************************************
Verbose DEBUG=ALL logs of client

15:46:16.040Z :client_secure_channel_layer   :1267  ------------------------------------- Client Sending a request
15:46:16.041Z :client_secure_channel_layer   :1268   CHANNEL ID  0
15:46:16.043Z :client_secure_channel_layer   :1269  { /*OpenSecureChannelRequest*/
 ...                                                 requestHeader                 /* RequestHeader                    */: {
 ...                                                   authenticationToken         /* NodeId                           */: ns=0;i=0
 ...                                                   timeStamp                   /* UtcTime                          */: 2019-05-10T15:46:16.039Z
 ...                                                   requestHandle               /* IntegerId                        */: 1               0x1
 ...                                                   returnDiagnostics           /* UInt32                           */: 0               0x0
 ...                                                   auditEntryId                /* UAString                         */: null
 ...                                                   timeoutHint                 /* UInt32                           */: 0               0x0
 ...                                                   additionalHeader            /* ExtensionObject                  */: null
 ...                                                 }
 ...                                                 clientProtocolVersion         /* UInt32                           */: 0               0x0
 ...                                                 requestType                   /* SecurityTokenRequestType         */: ISSUE ( 0)
 ...                                                 securityMode                  /* MessageSecurityMode              */: SIGNANDENCRYPT ( 3)
 ...                                                 clientNonce                   /* ByteString                       */
 ...                                                BUFFER{00000000: d2 2d f7 07 c8 b3 6e 70 f1 3c e8 1a ae 9e d7 80 48 38 07 9e 87 4b 46 29 ec a3 2a 5c 15 d9 3f 6f    R-w.H3npq<h...W.H8...KF)l#*\.Y?o
 ...                                                }
 ...                                                 requestedLifetime             /* UInt32                           */: 600000               0x927c0
 ...                                                };
15:46:16.087Z :client_secure_channel_layer   :1144  CLIENT SEND done.
15:46:16.705Z :client_secure_channel_layer   :877   CLIENT RECEIVED {"msgType":"ERR","isFinal":"F","length":71}
15:46:16.707Z :client_secure_channel_layer   :878
 ...                                                00000000: 45 52 52 46 47 00 00 00 00 00 12 80 37 00 00 00 43 6f 75 6c 64 20 6e 6f 74 20 76 65 72 69 66 79    ERRFG.......7...Could.not.verify
 ...                                                00000020: 20 73 65 63 75 72 69 74 79 20 6f 6e 20 4f 70 65 6e 53 65 63 75 72 65 43 68 61 6e 6e 65 6c 20 72    .security.on.OpenSecureChannel.r
 ...                                                00000040: 65 71 75 65 73 74 2e                                                                               equest.
 ...                                                
15:46:16.714Z :client_secure_channel_layer   :879   ERR F length   = 71
 ERROR RECEIVED FROM SENDER BadCertificateInvalid (0x80120000) Could not verify security on OpenSecureChannel request.
00000000: 45 52 52 46 47 00 00 00 00 00 12 80 37 00 00 00 43 6f 75 6c 64 20 6e 6f 74 20 76 65 72 69 66 79    ERRFG.......7...Could.not.verify
00000020: 20 73 65 63 75 72 69 74 79 20 6f 6e 20 4f 70 65 6e 53 65 63 75 72 65 43 68 61 6e 6e 65 6c 20 72    .security.on.OpenSecureChannel.r
00000040: 65 71 75 65 73 74 2e                                                                               equest.
15:46:16.719Z :client_secure_channel_layer   :223   request id =  null Error: ERROR RECEIVED
 ...                                                    at MessageBuilderBase._report_error (C:\Edge\Solution\edge\rig-connectors\opc-ua-client-service\src\node_modules\node-opcua-transport\src\message_builder_base.js:221:24)
 ...                                                    at MessageBuilder._decode_message_body (C:\Edge\Solution\edge\rig-connectors\opc-ua-client-service\src\node_modules\node-opcua-secure-channel\src\message_builder.js:430:14)
 ...                                                    at MessageBuilderBase._feed_messageChunk (C:\Edge\Solution\edge\rig-connectors\opc-ua-client-service\src\node_modules\node-opcua-transport\src\message_builder_base.js:196:18)
 ...                                                    at PacketAssembler.<anonymous> (C:\Edge\Solution\edge\rig-connectors\opc-ua-client-service\src\node_modules\node-opcua-transport\src\message_builder_base.js:53:14)
 ...                                                    at PacketAssembler.emit (events.js:182:13)
[... cut ...]
*************************************************************

Node opc version I am using 0.7.2 on WIN OS.

OPC Server Docker Image:

  opcplc:
     image: mcr.microsoft.com/iotedge/opc-plc:latest
     container_name: opcplc
     hostname: opcplc
     ports:
       - "50000:50000"
     command: --aa
**********************************************

As per above comment, you fixed regenerating cerfificates in  now fixed by node-opcua-pki@1.5.2 (Node opc version 2.0 alpha), but how can i get this fix under node-opcua version 0.7.2 ?

erossignon, 2019-05-13:
Error is : ERROR RECEIVED FROM SENDER BadCertificateInvalid (0x80120000)

samminen, 2019-05-13:
correct; I have self signed cert and key. When I pass them as part of opc ua connect options, I am still getting same above error (when I restart OPC UA server) and opc ua client also creating its certificates every time when I restart my opc ua client instead of using the same certificate what I passed in options. 

 this.options = {
                applicationName: "Edge-opcua-client",
                endpoint_must_exist: false,
                keepSessionAlive: true,
                securityMode: this.securityMode,   //3
                securityPolicy: this.securityPolicy,   //Basic256Sha256
                certificateFile: "./Certificates/edge-server.bundle.crt",
                privateKeyFile: "./Certificates/edge-server.key"
            };
**************************************
2nd question on create session error:

When i start the opc ua client first time, it is connected to opc server and when i create session for subscription, I am getting below error.

TypeError: Cannot read property '0' of undefined
****************************************

Any help ?

erossignon, 2019-05-13:
I manage to connect multiple times with node-opcua@0.7.2 sample client using the following command:
``` node bin\simple_client.js -e opc.tcp://10.0.75.2:5000 -s SIGNANDENCRYPT -P Basic256Sha256```

can you send a minimalist client code that reproduces the issue ?

In your code extract above for the client parameter it seems that you pass a server key pair,
```certificateFile: "./Certificates/edge-server.bundle.crt",
privateKeyFile: "./Certificates/edge-server.key"
```
should be client certificate and not server certificate key pair.

Also make sure edge-server.bundle.crt contains a PEM file,

---- 
Second question:  TypeError: Cannot read property '0' of undefined 
a step by step instruction to reproduce and/or a full stack trace would help.
I suggest you  fill the template form on a dedicated issue,

Please also consider contacting sterfive for professional  support 🙏

samminen, 2019-05-14:
Thank you so much for quick updates, appreciate; 

First time when i start OPC client with certificates, It connects to OPC server and able to create subscriptions as well. But when I restart OPC server, client is trying reconnect and it failed with error:

Logs in case of reconnect:
connecting: Endpoint URL: opc.tcp://34.52.44.59:50000/ Security Mode: 3, Security Policy: Basic256Sha256
Connected to OPC Server
opcua server connection lost
start reconnection
Connection retry attempt #0 retrying in 1 seconds
Connection retry attempt #1 retrying in 2 seconds
Connection retry attempt #2 retrying in 4 seconds
Connection retry attempt #3 retrying in 8 seconds

ERROR RECEIVED FROM SENDER BadCertificateInvalid (0x80120000)

Repro steps:
*******************
OPC Client code with certificates (you can use certificates generated by NODE  OPC UA)

 this.options = {
                    applicationName: "Edge-opcua-client",
                    endpoint_must_exist: false,
                    keepSessionAlive: true,
                    securityMode: this.NodeOPCUA.MessageSecurityMode.SIGNANDENCRYPT,
                    securityPolicy: NodeOPCUA.SecurityPolicy.Basic256Sha256
                    certificateFile: this.client_certificate, 
                    privateKeyFile: this.client_certificate_key 
                };

this.client = new NodeOPCUA.OPCUAClient(this.options);

            console.info(`connecting: Endpoint URL: ${endpointUrl} Security Mode: ${this.securityMode}, Security Policy: ${this.securityPolicy}`);        
        
            this.client.on('connection_lost', () => {
                console.info("opcua server connection lost");
            })
    
            this.client.on("backoff", async (number, delay) => {
                this._isConnected = false;
                console.info(`Connection retry attempt #${number} retrying in ${delay/1000.0} seconds`);
            });
    
            this.client.on("start_reconnection", async (...args) => {
                this._isConnected = false;
                await MQTTConnectionInstance.instance.updateSession();
                console.info("start reconnection");
            });
            this.client.on("connection_reestablished", async () => {
                this._isConnected = true;
                console.info("reconnect successful !!");
            });
            this.client.on("close", async  () => {
                this._isConnected = false;
                console.info("close and disconnect");
                // try {
                //     await this.client.disconnect();
                // }catch(err) {
                //     console.info(`error occured while closing the client disconnect ${err}`);
                // }
            });
            this.client.on("timed_out_request", async (request) => {
                console.info(`Client timeout request Received: ${request.toString()}`);
                //this.initialize(endpointUrl);
            });
            await this.client.connect(endpointUrl);
            this._isConnected = true;
            console.info("Connected to OPC Server ");
***********************************

OPC Server details I used :

OPC Server Docker Image:

opcplc:
image: mcr.microsoft.com/iotedge/opc-plc:latest
container_name: opcplc
hostname: opcplc
ports:
- "50000:50000"
command: --aa

samminen, 2019-05-28:
Hello Etinne,
  Any update/help for this issue as per my latest post ?

erossignon, 2019-05-28:
Hi @samminen ,
Please delete certificate and update to latest node-opcua@0.7.2  verion (or 2.0.0-alpha)
Then certificate will be regenerated and will expose a update "X509v3 Key Usage" that should content OPCUA C sever.

commit 94cc0cb4 (2019-05-06): update package  fixes #425