node-opcua/node-opcua #426: timestamped_set doesn't send write request response

gnobre, 2018-02-06:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**
Client wants to write data to Server. When on the server side I add a variable to the addressSpace with a `set` method, everything works fine. When I change this `set` method to `timestamped_set`, the data is written fine but the client never gets the response from the server, then eventually it raises a time out error: `Timeout .... waiting for response for  WriteRequest`

**Expected behavior**
When the client writes a variable to the server, it should receive a response for its request in the same way that the `set` method does.

**Minimal reproduction of the problem with instructions**

  1. Start server with writable variable and timestamped_set
  2. Start client
  3. Write data from client to server
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * (X) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: Lenovo IdeaPad
  * OS version: Ubuntu 16.04 LTS
    
    - ( ) Windows : version : _________
    - (X) Linux   : version : 16.04 LTS
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Local OPCUA Server
     
    
* **node-opcua version:** ``
0.2.1

* **Node:** 
   `node --version` = v8.9.4

erossignon, 2018-02-11:
This need to be investigated.
In the meantime, as a workaround , try using the async version of timestamp_set:

``` javascript
   timestamped_set: function (dataValue, callback) {
       // do something with dataValue 
        callback(null, StatusCodes.Good);
    }
```

gnobre, 2018-02-12:
Thanks, just tested the async function and it's working well :)