node-opcua/node-opcua #427: Authenticating with a X509 Certificate (windows)

stevewillard, 2018-02-06:
- (x) feature request
  * (x) I have installed node-opcua as a package ( using npm install )
      
* **node-opcua version:** 
0.2.1

* **Node:** 
   `node --version` = 6.12.3

I'm looking to authenticate an OPC connection using a x509 certificate on Windows. I was just wondering what the status of this feature is - I noticed some older closed issues talking about it, but it is not marked as a supported authentication feature.

[image]

davidemess, 2018-03-14:
Follow

erossignon, 2018-10-31:
duplicates #506

closed 2018-10-31 by erossignon