node-opcua/node-opcua #435: is it possible to delete existing references

Sushifresser, 2018-02-12:
**I'm submitting a ...**  (check one with "x")

 - ( ) bug report => search github for a similar issue or PR before submitting
 - (x) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**
I cannot find any method in the source code of node-opcua that would allow me to delete a reference from a node in my UA Server application. I can perfectly find an existing reference via BaseNode.findReference() method and I get the Reference object back. But it doesn't seem to be possible for me to delete it.

**Expected behavior**
It would be very helpful to be able to delete a reference from a node. Maybe analogously to the findReference() implementation. If it can be successfully found by the given parameters it can then be deleted.

**Minimal reproduction of the problem with instructions**
  1. instantiate server and add a node ("myNode") to objects folder (e.g. by "componentOf)
  2. add another reference (e.g. "Organizes") from objects folder to "myNode"
  3. now try to delete the hasComponent reference (there doesn't seem to be a method for this, right?)

  
**What is the motivation / use case for changing the behavior?**
It should be possible in general to delete references from a node that have been added earlier.

**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: Windows PC
  * OS version: Windows 10
    
    - (x) Windows : version : Windows 10 Enterprise
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: node-opcua
     - Version: v0.2.1 (NPM)
     - Manufacturer/Software vendor: node-opcua
     
    
* **node-opcua version:** ``
v0.2.1 (NPM)

* **Node:** 
   `node --version` = v8.9.4

erossignon, 2018-02-22:
node-opcua@0.2.2 now provides a BaseNode#removeReference.
Please let me know how it fits with your requirements.

Sushifresser, 2018-02-24:
Hi @erossignon 
this does the trick. Thank you very much!
Best regards

closed 2018-02-24 by Sushifresser