node-opcua/node-opcua #440: Specify namespace index of browseName

FRules, 2018-02-27:
I might be wrong but it seems like the namespace of the browseName is always 0.
According to the API, browseName takes just a string as an argument, not an object where you could specify the namespace (see page 16 and 62 of opc ua specification part 3: address space).  

This is the (relevant) code where I add a node.
```js
var adapter = addressSpace.addObject({
  organizedBy: addressSpace.rootFolder.objects,
  browseName: "Adapter",
  nodeId: new opcua.NodeId(opcua.NodeIdType.STRING, "adapter", 2)
});
```

Shouldn't it be the case that the namespace of the browse name equals either the namespace of the nodeId, in this example 2, or a specified one?

erossignon, 2018-02-28:
browseName could be a string or a QualifiedName like object.
It will be coerce to the QualifiedName 
A QualifiedName contains a namespace index and a string 
When specified as a string a optional name space coud be prepended : ```36:MyName" ```

So the following variation are possible
```js
var adapter = addressSpace.addObject({
  browseName: "Adapter",
});
```
```js
var adapter = addressSpace.addObject({
  browseName: { name: "Adapter", namespaceIndex: 0};
});
```
```js
var adapter = addressSpace.addObject({
  browseName: new opcua.QualifiedName({ name: "Adapter", namespace: 1});
});
```
or even:
```js
var adapter = addressSpace.addObject({
    browseName: "1:Adapter", // Adapter on Namespace Index 1
});
```

FRules, 2018-02-28:
Thank you, works like a charm! 👍

closed 2018-02-28 by FRules