node-opcua/node-opcua #442: session MaxListenersExceededWarning on server

biancode, 2018-02-28:
**I'm submitting a ...**  (check one with "x")

 - (x ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

I set the server to max 20 sessions and add more sessions, but it raise a warning on reaching default max sessions of node.js.

```
  opcuaIIoT:server Session timeout:60000 +0ms
  opcuaIIoT:server Session id:undefined +0ms
(node:15313) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 abort listeners added. Use emitter.setMaxListeners() to increase limit
    at _addListener (events.js:280:19)
    at ServerSecureChannelLayer.addListener (events.js:297:10)
    at ServerSession._attach_channel (/node-red/node_modules/node-opcua-server/src/server_session.js:792:13)
    at OPCUAServer._on_CreateSessionRequest (/node-red/node_modules/node-opcua-server/src/opcua_server.js:829:13)
    at OPCUAServer.OPCUABaseServer.on_request (/node-red/node_modules/node-opcua-server/src/base_server.js:240:21)
    at OPCUAServerEndPoint.<anonymous> (/node-red/node_modules/node-opcua-server/src/opcua_server.js:277:14)
    at emitThree (events.js:136:13)
    at OPCUAServerEndPoint.emit (events.js:217:7)
    at ServerSecureChannelLayer.<anonymous> (/node-red/node_modules/node-opcua-server/src/server_end_point.js:287:18)
    at emitOne (events.js:116:13)
    at ServerSecureChannelLayer.emit (events.js:211:7)
    at ServerSecureChannelLayer._on_common_message (/node-red/node_modules/node-opcua-secure-channel/src/server/server_secure_channel_layer.js:1051:18)
    at emitMany (events.js:147:13)
    at MessageBuilder.emit (events.js:224:7)
    at MessageBuilder._decode_message_body (/node-red/node_modules/node-opcua-secure-channel/src/message_builder.js:459:22)
    at MessageBuilderBase._feed_messageChunk (/node-red/node_modules/node-opcua-transport/src/message_builder_base.js:196:18)
    at PacketAssembler.<anonymous> (/node-red/node_modules/node-opcua-transport/src/message_builder_base.js:53:14)
    at emitOne (events.js:116:13)
    at PacketAssembler.emit (events.js:211:7)
    at PacketAssembler.feed (/node-red/node_modules/node-opcua-packet-assembler/src/packet_assembler.js:107:14)
    at MessageBuilderBase.feed (/node-red/node_modules/node-opcua-transport/src/message_builder_base.js:159:30)
    at ServerTCP_transport.<anonymous> (/node-red/node_modules/node-opcua-secure-channel/src/server/server_secure_channel_layer.js:382:37)
```

**Expected behavior**

The sessions setMaxListeners should be the same as the server max sessions allow without warning, please.

**Minimal reproduction of the problem with instructions**

  1. set server options to max 20 sessions
  2. open more than 11 sessions to the server

**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( x) node-red
       - ( ) other : please specify 
        
  * Device: macbook
  * OS version:
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( x) MacOs   : version : 10.13
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: node-opcua
     - Version: 0.2.x
     - Manufacturer/Software vendor: you
     - link : https://
     
    
* **node-opcua version:** ``

0.2.2

* **Node:** 
   `node --version` = 8.9.4

biancode, 2018-02-28:
session and/or channel

erossignon, 2018-02-28:
see : https://nodejs.org/docs/latest/api/events.html#events_emitter_setmaxlisteners_n
max listeners can be set to inifinity by using setMaxListeners(0);
see also https://stackoverflow.com/a/26176922/406458

erossignon, 2018-02-28:
This being said, this warning is here for a reason .
It looks like you are reusing the same channel over and over again to create a session.
OPCUA client will rarely do this in practice. 
Please provide more insight about why you  need to create more than one session on a  given secure channel ?
Please provide more information about how to reproduce using vendor opcua client

erossignon, 2018-02-28:
Btw, there is one single unit test that causes this warning to be revealed
( please note that this is a warning, so it's not harmful anyway)
```
cd packages
node run_all_mocha_tests.js "should be possible to open  many sessions on a single connection"
````

biancode, 2018-03-01:
I just work with the monitor of subscription and it monitors more than 10 NodeIds

`return subscription.monitor(
    {
      nodeId: opcua.resolveNodeId(addressSpaceItem.nodeId),
      attributeId: opcua.AttributeIds.Value,
      origin: msg
    },
    {
      samplingInterval: interval,
      discardOldest: true,
      queueSize: queueSize
    },
    opcua.read_service.TimestampsToReturn.Both,
    function (err) {
      handleErrorCallback(err, addressSpaceItem, msg)
    }
  )`

biancode, 2018-03-01:
it is in list: 
* one session to the server and 
* one subscription in that session
* many monitor calls at the subscription

biancode, 2018-03-01:
client and server are node-opcua

biancode, 2018-03-01:
other case is to start a node-opcua client to node-opcua server with in list:

* one opcua client
* many sessions (more than 11)

closed 2019-08-03 by stale[bot]