node-opcua/node-opcua #446: Using opcua client to crash opcua server

wcchh, 2018-03-05:
**I'm submitting a ...**  (check one with "x")

 - (X) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

FATAL ERROR: CALL_AND_RETRY_LAST Allocation failed - JavaScript heap out of memory

**Expected behavior**

I guess the Server should not failed.

**Minimal reproduction of the problem with instructions**

  1. Using simple server/client from example. 
  2. One client repeats "connect -> create session -> close session ->disconnect"...
  3. Server will crashed by "JavaScript heap out of memory" after 2111 connection.
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * (X) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _Linux Mint 18.1_
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
The opcua simple server/client from example
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:** ``

"node-opcua": "^0.2.1"

* **Node:** 
   `node --version` = v6.12.3

erossignon, 2018-03-05:
what's about node-opcua@0.2.2  this version fixes a some memory leaks .

wcchh, 2018-03-06:
node-opcua@0.2.2 is better, but still failed after 3281 times connections.

<--- Last few GCs --->
 651764 ms: Mark-sweep 1135.3 (1418.2) -> 1123.3 (1418.2) MB, 2469.3 / 0.0 ms [allocation failure] [GC in old space requested].
 653697 ms: Mark-sweep 1123.3 (1418.2) -> 1123.1 (1418.2) MB, 2286.8 / 0.0 ms [allocation failure] [GC in old space requested].
 656207 ms: Mark-sweep 1123.1 (1418.2) -> 1132.5 (1403.2) MB, 2363.1 / 0.0 ms [last resort gc].
 658466 ms: Mark-sweep 1132.5 (1403.2) -> 1141.8 (1403.2) MB, 2363.1 / 0.0 ms [last resort gc].
<--- JS stacktrace --->
==== JS stack trace =========================================
Security context: 0x27cb71dcf781 <JS Object>
    1: _register [/media/wcc/75e63f32-2e90-45aa-ae0f-010f2dbca628/projects/imo/node_modules/node-opcua-address-space/src/address_space.js:~190] [pc=0x256120753a28] (this=0x3cd51ec460a1 <an AddressSpace with map 0x6e529ed10b1>,node=0x9a485576569 <a UAObject with map 0xc91e8b45391>)
    2: _createNode [/media/wcc/75e63f32-2e90-45aa-ae0f-010f2dbca628/projects/imo/node_modules/nod...
FATAL ERROR: CALL_AND_RETRY_LAST Allocation failed - JavaScript heap out of memory
...

erossignon, 2018-03-06:
Could you provide the sample client application that you are using to illustrate the issue ?
I have been using:
``` javascript
"use strict";

const opcua = require("node-opcua");
const endpointUrl = "opc.tcp://localhost:26543";
async function main()  {

    let counter = 1;
    while(true) {
        const client = new opcua.OPCUAClient({endpoint_must_exist: false});

        await client.connect(endpointUrl);

        const session = await client.createSession();
        await session.close();

        await client.disconnect();

        console.log(" Connected = ",counter++);
    }
}
main();
```
and could run it  up to at least 5000 time without memory consumption going up on the server side.

wcchh, 2018-03-07:
Thanks for reply, and your sample is good. 
I using promise since my node version is v6.12.3

Here's my client:
```
"use strict";
const opcua = require("node-opcua");
const endpointUrl = "opc.tcp://" + require("os").hostname() + ":4334/UA/MyLittleServer";

var promise = function (arg) {
	return new Promise(function (resolve, reject) {
		resolve(arg);
	});
};
function goSenario(client, session, done) {
	let the_session = session;
	promise()
		.then(() => {
			return new Promise(function (resolve, reject) {
					client.connect(endpointUrl, function (err) {
						if (err) {
							console.log(" cannot connect to endpoint :", endpointUrl);
							return reject(err);
						}
						return resolve();
					});
			});
		})
		.then(() => {
			return new Promise(function (resolve, reject) {
					client.createSession((err, session) => {
						if (err) {
							console.log(" cannot create session :", err.message);
							return reject(err);
						}
						the_session = session;
						return resolve(session);
					});
			});
		})
		.then(() => {
			return new Promise(function (resolve, reject) {

				the_session.browse("RootFolder", function (err, browseResult) {
					if (err) {
						console.log(" cannot browse RootFolder :", err.message);
						return reject(err);
					}
					return resolve();
				});
			});
		})
		.then(() => {
				if (the_session) {
					the_session.close(function (err) {
						the_session = null;
						if (!err) {
							return client.disconnect(() => { done(); });
						} else {
							console.error("session closed failed.#", err);
							return done(err);
						}
					});
				} else {
					return client.disconnect(() => { done(); });
				}
		}, (err) => {
			console.error(err);
			if (the_session) {
				the_session.close(function (err) {
					if (!err) {
						return client.disconnect(() => { done(err); });
					} else {
						console.error("session closed failed.#", err);
						return done(err);
					}
				});
			} else {
				return client.disconnect(() => { done(err); });
			}
		});
}

function main(counter) {
	var client = new opcua.OPCUAClient();

	var callCount = 0;
	var SenarioLoop = function (err = null, session = null) {
		callCount++;

		setTimeout(() => {
			goSenario(client, session, SenarioLoop);
			console.error("callCount:", callCount);
		}, 100);
	};
	SenarioLoop();
}
main();
```

wcchh, 2018-03-07:
I found the different part is the client object be generated.
`new opcua.OPCUAClient()`

> If I kept the client, server crashed. 
> If I regenerated the client everytime, server passed.

closed 2019-08-03 by stale[bot]