node-opcua/node-opcua #452: Updating from version 0.2.1 to 0.2.2 causes client to fail to connect.

davidAlittle, 2018-03-07:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 
**Current behavior**
Creating a new connection fails. 
In console:
`["opc.tcp://CX-1D8328:4840 NONE http://opcfoundation.org/UA/SecurityPolicy#None"]0: "opc.tcp://CX-1D8328:4840 NONE http://opcfoundation.org/UA/SecurityPolicy#None"length: 1__proto__: Array(0)
OPCclient.js?dce6:328  failure  Error:  End point must exist opc.tcp://10.88.30.104:4840`
**Expected behavior**
Client should connect normally.

**Minimal reproduction of the problem with instructions**

Update version.

**Please tell us about your environment:**

  * (x) I have installed node-opcua as a package ( using npm install )

        
  * Device: Desktop PC (HMI client)
  * OS version: Windows 7
    
    - (x ) Windows : version : _7
 
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: OPC Server running on Beckhoff CX5120
     - Version:_________
     - Manufacturer/Software vendor:Beckhoff
     - link : https://www.beckhoff.com/english.asp?embedded_pc/cx5120.htm and https://www.beckhoff.com/english.asp?twincat/twincat_opc_server.htm
     
    
* **node-opcua version:** ``
0.2.2

* **Node:** 
   `node --version` = v6.11.4

* **Code: **
```
var client = new opcua.OPCUAClient({keepSessionAlive: true});
var endpointUrl = "opc.tcp://10.88.30.104:4840";  
var the_session = null;
var the_subscription =null;
async.series([
    // step 1 : connect to
    function(callback)  {
        client.connect(endpointUrl,function (err) {
            if(err) {
                console.log(" cannot connect to endpoint :" , endpointUrl );
            } else {
                console.log("connected !");
            }
            callback(err);
        });
    },
    // step 2 : createSession
    function(callback) {
        client.createSession( function(err,session) {
            if(!err) {
                the_session = session;
            }
            callback(err);
        });

    },
```

erossignon, 2018-03-13:
New version 0.2.2 is more compliant to OPCUA security guideline and now require that endpoint is properly   written and client will now refuse to connect if the specified enpoint is not exactly one of the endpoints exposed by the server (see release notes https://github.com/node-opcua/node-opcua/releases/tag/v0.2.2 ) 
Please try
```javascript
var client = new opcua.OPCUAClient({
  keepSessionAlive: true,
  ensure_endpoint_exists:false
 });

```

closed 2018-03-13 by erossignon

davidAlittle, 2018-03-15:
in case anyone comes looking, it's actually
`endpoint_must_exist: false`
not
`ensure_endpoint_exists:false 
`
(or at least the 1st works but the second doesn't)