node-opcua/node-opcua #454: opc-ua server on galileo gen 2

cristina21r, 2018-03-11:
I want to make an opc-ua server on galileo gen 2, with a yocto built version of linux.
I've made, with some help from git, a complete version of that kind of server for raspberry pi 3, but i have some problems with porting it to galileo. I'm using for the server dht11 and ds18b20 sensors, but i can't manage to make it work on galileo.  I have also this error on trying to run the server on galileo: Proxy is not defined.

Here is my code:
 
var opcua = require("node-opcua");
var w1bus = require("node-w1bus");
var bus = w1bus.create();
var rpiDhtSensor = require("rpi-dht-sensor");

function read_sensor(sensor,callback) {
                bus.getValueFrom(sensor)
                .then(function(res){  callback(null,res.result); })
                .catch(callback);
        };

function start_server() {

        var server = new opcua.OPCUAServer({
                port: 1234
        });

        server.buildInfo.productName ="Rapsberry OPCUA Server"

        function post_initialize() {

                var addressSpace = server.engine.addressSpace;
                var devices = addressSpace.addFolder("ObjectsFolder",{ browseName: "Devices"});
                var temperatureSensor = "28-041460c2a3ff";
                install_sensor(temperatureSensor);
        };

        var tempValue = -20.0;
        var tempValue1 = -20.0;
        var humValue = -20.0;

        function install_sensor(sensor) {
                var addressSpace = server.engine.addressSpace;
                var sensorName = "ds18b20";
                var sensorName1 = "dht11";
                var devices = addressSpace.rootFolder.objects.devices;
                var sensor  = addressSpace.addObject({
                        organizedBy: devices,
                        browseName:  sensorName
                });

                var dht = new rpiDhtSensor.DHT11(27);

                setInterval(function() {
                        var temperatureSensor = "28-041460c2a3ff";
                        read_sensor(temperatureSensor,function(err,result) {
                                tempValue = result.value;
                        });
                        var readout = dht.read();
                        tempValue1 = readout.temperature.toFixed(2);
                        humValue = readout.humidity.toFixed(2);
                },3000);

                addressSpace.addVariable({
                        propertyOf: sensor,
                        browseName: "temperature1",
                        dataType:   "Double",
                        value: {
                                get: function () {
                                        return new opcua.Variant({dataType: opcua.DataType.Double, value: tempValue });
                                }
                        }
                });

                addressSpace.addVariable({
                        propertyOf: sensor,
                        browseName: "temperature2",
                        dataType:   "Double",
                        value: {
                                get: function () {
                                        return new opcua.Variant({dataType: opcua.DataType.Double, value: tempValue1 });
                                }
                        }
                });

                addressSpace.addVariable({
                        propertyOf: sensor,
                        browseName: "humidity",
                        dataType: "Double",
                        value: {
                                get: function() {
                                        return new opcua.Variant({dataType: opcua.DataType.Double, value: humValue });
                                }
                        }
                });
        };

        server.initialize(post_initialize);

        server.start(function() {
                console.log("\n\nTemperature - OPCUAServer");
                var endpointUrl = server.endpoints[0].endpointDescriptions()[0].endpointUrl;
                console.log("Server endpoint url is ", endpointUrl );
                console.log("\n");
        });
}

start_server();

Some help for making it work on Galileo, please??

erossignon, 2018-03-13:
Proxy is not defined : please try to upgrade to a more recent version of nodejs => 8 or 9

cristina21r, 2018-03-13:
I will do this..but i will still have the problem with reading the sensors connected to galileo..
Do you know something about this??

erossignon, 2018-03-14:
no sorry

cristina21r, 2018-03-22:
The problem is that my version of linux doesn't support a newer version of nodejs.. i have 4.4.3 and i can't make this server work and i must do that... some other ideas to fix the proxy issue?

erossignon, 2018-03-23:
You probably need to stick with an older version of node-opcua , until nodejs > 6.5 becomes available for galileo.

erossignon, 2018-03-23:
have you try upgrading nodejs version with nvm ?
https://github.com/hybridgroup/cylon-intel-iot

cristina21r, 2018-03-28:
I tried everything..this one just not support nodejs>4.4.3.. 
How could i get an older version of node-opcua??

cristina21r, 2018-03-28:
The thing is that this one is working just fine on rpi, with node 4.8.2..i don't think that i need node >6..
Maybe some settings for configure proxy? but i dont know what node-opcua needs in order to work..

erossignon, 2018-03-28:
you can install older version of node opcua by specifying the version  in your package.json
I would try 0.2.0.
Nodejs < 0.4.0 is not under maintenance any more and we have moved forward using es6 version of javascript and node 6.5.0 
see https://kangax.github.io/compat-table/es6/.
Proxy is a built-in javascript feature and cannot be emulated

cristina21r, 2018-03-29:
a node-opcua 0.2.0 doesn t use proxy?

cristina21r, 2018-04-02:
I tried with 0.2.0 and still the same problem..

erossignon, 2018-04-02:
It seems that npm install node-opcua@0.2.0 still install  node-opcua@0.2.3 unfortunalty. I suggest you go for node-opcua@0.0.65 then.
It looks like intel has discontinued galileo https://hardware.slashdot.org/story/17/06/19/1720201/intel-quietly-discontinues-galileo-joule-and-edison-development-boards

cristina21r, 2018-04-02:
0.0.65 told that doesn't exist on repository.
I managed to install only 0.0.50.. 
And  I have this error now:
/server/sample_server.js:24
            organizedBy: addressSpace.rootFolder.objects,
                                                ^

TypeError: Cannot read property 'objects' of undefined
    at construct_my_address_space (/server/sample_server.js:24:49)
    at post_initialize (/server/sample_server.js:95:5)
    at Immediate._onImmediate (/server/node_modules/node-opcua/lib/server/opcua_server.js:428:9)
    at processImmediate [as _immediateCallback] (timers.js:383:17)

cristina21r, 2018-04-02:
https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-server/src/opcua_server.js
the refference is at this file, at  done();

 */
OPCUAServer.prototype.initialize = function (done) {

    var self = this;
    assert(!self.initialized);// already initialized ?

    OPCUAServer.registry.register(self);

    self.engine.initialize(self.options, function () {
        self.emit("post_initialize");
        done();
    });
};

erossignon, 2018-04-02:
0.0.50 might be too old, sorry, you can list node-opcua published versionh here  https://www.npmjs.com/package/node-opcua

nospam2000, 2018-07-17:
For the Intel Quark Processor (GalileoGen2 board) the newest version of node.js which I was able to build is version 7. Version 8 doesn't compile because X87 support is broken in earlier 8 versions and has been removed for later 8 versions.
The main problem is that the Quark CPU doesn't support MMX.

nospam2000, 2019-07-30:
In the meanwhile there is a port of Node.js 8.15.0 to the SIMATIC IOT2000 which is based on GalileoGen2. See https://github.com/siemens/meta-iot2000/issues/107 for more details

closed 2020-08-05 by stale[bot]