node-opcua/node-opcua #455: ClientSession.prototype.readVariableValue error handling bug

JulZimmermann, 2018-03-12:
**I'm submitting a ...**  (check one with "x")

 - ( X) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**
ClientSession.prototype.readVariableValue called with ("ns=2;i=invalid") will cause:
Unhandled promise rejection (rejection id: 3): Error: String cannot be coerced to a nodeId : ns=2;i=invalid

This will even occur when wrapping the call in try catch block.

**Expected behavior**
Callback should have error object != undefined

**Minimal reproduction of the problem with instructions**
```
const opcua = require("node-opcua");

const client = new opcua.OPCUAClient();

client.connect("opc.tcp://opcua.demo-this.com:51210/UA/SampleServer", (err) => {
    if (!err) {
        client.createSession((err, session) => {
            if (!err) {
                session.readVariableValue("ns=2;i=invalid", (err, dataValue) => {
                    if (err) {
                        console.log("This won't be executed");
                    }
                });
            }

            client.disconnect();
        });
    }
});
```

**What is the motivation / use case for changing the behavior?**
Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code

**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * (X ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - (X ) Windows : version : 10
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
 
* **node-opcua version:** ``
v0.2.2

* **Node:** 
   `node --version` = v7.7.3

erossignon, 2018-03-13:
"ns=2;i=invalid"  cannot match a valid nodeId hence the error, please provide a valid nodeId.

erossignon, 2018-03-13:
if the nodeId is valid but doesn't exist you can check the dataValue.statusCode to detect this.
```` javascript
                session.readVariableValue("ns=2;i=10000", (err, dataValue) => {
                    dataValue.statusCode.should.eql(opcua.StatusCodes.BadNodeIdUnknown);
                    inner_done();
                });

````

commit 18886c8a (2018-03-13): fixes #455 : provide a example
closed 2018-03-13 by erossignon