node-opcua/node-opcua #457: UserIdentity token is invalid with open62541 server

w4tsn, 2018-03-13:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**
Invoking the opcua-commander to connect to the open62541 example server starts the commander, opens a SecureChannel, throws a warning and closes right after that without any further info on what happened.

The open62541 server responds with

```
Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
SecureChannel wants to activate, but UserIdentity token is invalid
```

The opcua-commander throws the warning `SERVER responseHeader.requestHandle is invalid: expecting 0x1 but got 0x0 Class=OpenSecureChannelResponse`

**Expected behavior**
The opcua-commander can connect to the example server of open62541 with basic connection parameters like `./opcua-commander -v -e opc.tcp://localhost:16664/`

**Minimal reproduction of the problem with instructions**

1. Download the [open62541-server package](https://open62541.org/) in Version 0.2. In my case for RaspberryPi
2. Start the `examples/server` binary
3. Call `./opcua-commander -e opc.tcp://localhost:16664/`
  
**What is the motivation / use case for changing the behavior?**
The opcua-commander is used in node-red-contrib-iiot-opcua and may be related to [this issue](https://github.com/biancode/node-red-contrib-iiot-opcua/issues/26). I want to use the node-red client nodes to connect to the open64521 server.

**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * (x) I am using an application that uses node-opcua 
  
       - (x) node-red
       - ( ) other : please specify 
        
  * Device: OrangePiZero
  * OS version: Linux 3.4.113-sun8i #18 SMP PREEMPT 2018 armv7l GNU/Linux
    
    - ( ) Windows : version : _________
    - (x) Linux   : version : Linux 3.4.113-sun8i #18 SMP PREEMPT 2018 armv7l GNU/Linux
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: open62541/examples/server
     - Version: 0.2
     - Manufacturer/Software vendor: open62541 project
     - link : https://open62541.org
     
    
* **node-opcua version:**
Release 0.2.2

* **Node:** 
   `node --version` = v8.9.4

erossignon, 2018-03-15:
Please check with a most recent version of open62541 ( 0.3-rc1 or master) and let us know if the problem still exists

w4tsn, 2018-03-16:
With the current release candidate 0.3-rc1 the server responds with `ActivateSession: Could not generate a server nonce`

```
[2018-03-16 07:48:06.176 (UTC+0000)] info/network	Connection 5 | New connection over TCP from 127.0.0.1
[2018-03-16 07:48:06.445 (UTC+0000)] info/channel	Creating a new SecureChannel
[2018-03-16 07:48:06.446 (UTC+0000)] warn/securitypolicy	No PKI plugin set. Accepting all certificates
[2018-03-16 07:48:06.446 (UTC+0000)] info/channel	Connection 5 | SecureChannel 1 | Opened SecureChannel
[2018-03-16 07:48:06.808 (UTC+0000)] info/session	Connection 5 | SecureChannel 1 | Session bea40cda-1275-744c-2fe2-2e52cf5c1531 | ActivateSession: Could not generate a server nonce
[2018-03-16 07:48:07.051 (UTC+0000)] info/network	Connection 5 | Closed by the server
```

erossignon, 2018-03-16:
@w4tsn , I can get the exact same behavior on open62541 0.3-rc1,
However switching to latest version on master of open62541  is working fine.
This issue seems to be on open62541  side and has been fix on their side now.
I suggest that you wait for open62541@0.3-rc2 that should be available soon on report an issue on their side.

erossignon, 2018-03-18:
see also https://github.com/biancode/node-red-contrib-iiot-opcua/issues/26

closed 2018-03-18 by w4tsn