node-opcua/node-opcua #458: Fails to create anonymous session after reconnect

sgmbh, 2018-03-15:
**I'm submitting a ...**  (check one with "x")

 - (x ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**
After reconnect to the opc ua server the application can't find anonymous user token policy and can't create a new session.

**Expected behavior**
It should be possible to lose the connection to the opc ua server and after reconnect to initialize a new session with all subscriptions.

**Minimal reproduction of the problem with instructions**
1. Start atvise as opc ua server (is running as a service on windows 10 pro)
2. Start client (following code)
3. Stop service
4. -> backoff Event 
5. Start service
6. -> connection_reestablished and after_reconnection Event
7. warning : disconnection : closing pending sessions
8. Error: Cannot find ANONYMOUS user token policy in end point description

```
let options = {
    connectionStrategy: {
        maxRetry: 100000
    },
    keepSessionAlive: true
};
let client = new opcua.OPCUAClient( options );
let session : opcua.ClientSession;

client.on( "start_reconnection", function () {
    winston.info( " ... start_reconnection" );
} );
client.on( "backoff", function ( nb : any, delay : any ) {
    winston.info( "  connection failed for the", nb, " time ... We will retry in ", delay, " ms" );
} );
client.on( "connection_reestablished", function () {
    winston.info( " ... connection is reestablished after a connection break" );
} );
client.on( "after_reconnection", function ( err : Error ) {
    winston.info( " ... reconnection process has been completed: ", err );
//without the following I will get an other error
    client = new opcua.OPCUAClient( options );
    client.connect( endpointUrl, ( err ) => {
        if ( err ) {
            let errorMsg : string = `Could not connect client: ${err}`;
            winston.error( errorMsg );
        } else {
            winston.info( "connected" );
            createOpcUaSession();
        }
    } );
} );
client.on( "timed_out_request", function ( request : any ) {
    winston.info( " ... request has timed out without receiving a response: ", request );
} );

const endpointUrl : string = 'opc.tcp://' + url + ':' + port;
winston.info( endpointUrl );

winston.info( "connect" );
client.connect( endpointUrl, ( err ) => {
    if ( err ) {
        let errorMsg : string = `Could not connect client: ${err}`;
        winston.error( errorMsg );
    } else {
        winston.info( "connected" );
        createOpcUaSession();
    }
} );

function createOpcUaSession() {
    let callback = ( err, ses ) => {
        if ( !err ) {
            winston.info( "created session" );
            session = ses;
            // this.initSubscription( session );
        } else {
            let errorMsg : string = `Could not create session: ${err}`;
            winston.error( errorMsg );
        }
    };
    winston.info( "create session" );
    client.createSession( callback );
}
...
```

Console Output:
  _info: start
info: opc.tcp://127.0.0.1:4840
info: connect
info: connected
info: create session
info: created session
info:  ... start_reconnection
info:   connection failed for the 0  time ... We will retry in  10  ms
info:   connection failed for the 1  time ... We will retry in  20  ms
info:   connection failed for the 2  time ... We will retry in  40  ms
info:   connection failed for the 3  time ... We will retry in  80  ms
info:  ... connection is reestablished after a connection break
info:  ... reconnection process has been completed: 
warning : disconnection : closing pending sessions
info: connected
info: create session
error: Could not create session: Error: Cannot find ANONYMOUS user token policy in end point description
 warning : ClientSessionKeepAliveManager#ping_server   ServiceFault returned by server { /*ServiceFault*/
 responseHeader                /* ResponseHeader                   */: {
   timeStamp                   /* UtcTime                          */: 2018-03-15T09:01:06.039Z
   requestHandle               /* IntegerId                        */: 4               0x4
   serviceResult               /* StatusCode                       */: BadSessionNotActivated (0x80270000)
   serviceDiagnostics          /* DiagnosticInfo                   */: {
     namespaceUri              /* Int32                            */: -1
     symbolicId                /* Int32                            */: -1
     locale                    /* Int32                            */: -1
     localizedText             /* Int32                            */: -1
     additionalInfo            /* String                           */: null
     innerStatusCode           /* StatusCode                       */: Good (0x00000)
     innerDiagnosticInfo       /* DiagnosticInfo                   */: {
     }
   }
   stringTable                 /* String                        [] */: [ ] (l=1)
   additionalHeader            /* ExtensionObject                  */: null
 }
};_

**What is the motivation / use case for changing the behavior?**
It should be possible to reconnect to the opc ua server.

**Please tell us about your environment:**
For development I use atvise as opc ua server

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: Dell
  * OS version: 16299.309
    
    - (x ) Windows : version : 10 pro v1709
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: atvise server
     - Version: 3.0.1
     - Manufacturer/Software vendor: Atvise
     - link : https://www.atvise.com/en/
     
    
* **node-opcua version:** ``
v0.1.1-0

* **Node:** 
   `node --version` = v6.9.1

erossignon, 2018-03-15:
To investigate further:
 * it looks like the script is recreating a client connection  + session inside the ```"after_reconnection```  event hander. this is not necessary as OPCUAClient should take care itself to restore the connection  and maintain the session. 
  * We need to check if the problem come from ATVISE or node-opcua, for this I would try to replay the same scenario on a different server.  can you try the same test scenario on some other server ( from various vendor) ?

sgmbh, 2018-03-16:
yes that's true. Because without the recreation of a client connection and session, the existing client will be closed. As example, without creating a new client, the output looks as following:

> info: connect
info: connected
info: create session
info: created session
info:  ... start_reconnection
info:   connection failed for the 0  time ... We will retry in  10  ms
info:   connection failed for the 1  time ... We will retry in  20  ms
info:  ... connection is reestablished after a connection break
info:  ... reconnection process has been completed: 
info: client exists
info: session exists
warning : disconnection : closing pending sessions
info:  ... client closed: 
Process finished with exit code 0

I've added to my code the following:
```
client.on( "close", function ( err : Error ) {
    winston.info( " ... client closed: ", err );
    setTimeout( () => {
        client = new opcua.OPCUAClient( options );
        winston.info( "new client" );
        winston.info( "connect" );
        client.connect( endpointUrl, ( err ) => {
            if ( err ) {
                let errorMsg : string = `Could not connect client: ${err}`;
                winston.error( errorMsg );
            } else {
                winston.info( "connected" );
                createOpcUaSession();
            }
        } );
    }, 3000);
} );
```

now the application doesn't crash anymore.

> info: start
info: opc.tcp://127.0.0.1:4840
info: connect
info: connected
info: create session
info: created session
info:  ... start_reconnection
info:   connection failed for the 0  time ... We will retry in  10  ms
info:   connection failed for the 1  time ... We will retry in  20  ms
info:   connection failed for the 2  time ... We will retry in  40  ms
info:  ... connection is reestablished after a connection break
info:  ... reconnection process has been completed: 
info: client exists
info: session exists
warning : disconnection : closing pending sessions
info:  ... client closed: 
info: new client
info: connect
info: connected
info: create session
info: created session

erossignon, 2018-04-16:
@sgmbh can you test with node-opcua@0.3.0 and tell us it is fixed now ?

sgmbh, 2018-04-18:
@erossignon thanks. I updated to 0.3.0, but unfortunately the behavior is the same.
If I don't create a new opcua.OPCUAClient after the "close" Event, the process will be finished.

```
client.on( "start_reconnection", function () {
   winston.info( " ... start_reconnection" );
} );
client.on( "backoff", function ( nb : any, delay : any ) {
    winston.info( "  connection failed for the", nb, " time ... We will retry in ", delay, " ms" );
} );
client.on( "connection_reestablished", function () {
    winston.info( " ... connection is reestablished after a connection break" );
} );
client.on( "after_reconnection", function ( err : Error ) {
    winston.info( " ... reconnection process has been completed: ", err );
    if ( client ) {
        winston.info( "client exists" );
        if ( session ) {
            winston.info( "session exists" );
        }
    }
} );
client.on( "timed_out_request", function ( request : any ) {
    winston.info( " ... request has timed out without receiving a response: ", request );
} );
client.on( "close", function ( err : Error ) {
    winston.info( " ... client closes: ", err );
} );
```
Output:

> info: start
info: opc.tcp://127.0.0.1:4840
info: connect
info: connected
info: create anonymous session
info: created session
info: init subscription
info: start subscribe
info: nodeNameXY false
**info:  ... start_reconnection
info:   connection failed for the 0  time ... We will retry in  10  ms
info:   connection failed for the 1  time ... We will retry in  20  ms
info:   connection failed for the 2  time ... We will retry in  40  ms
info:   connection failed for the 3  time ... We will retry in  80  ms
info:  ... connection is reestablished after a connection break
info:  ... reconnection process has been completed: 
info: client exists
info: session exists
info:  ... client closes: 
Process finished with exit code 0**

erossignon, 2018-04-18:
Hi ,
In fact node-opcua CLIENT takes care of recreating session and subscription+monitored item when reconnection is re-established. You should not need to create a new client in 
```javascript
client.on( "after_reconnection", function ( err : Error ) {
     /* DO NOT ATTEMPT TO RECREATE CLIENT/SESSION HERE */ 
    /* this will be handled automatically */
} );
````

sgmbh, 2018-04-23:
Hi @erossignon 
Yes I understood that. That's the reason why I tried it without creating a client on my own. But then, the client closes and the process finishes. Because of that my workaround is to create a new client in the close event handler.

Unfortunately I don't know how you can setup an atvise server without the license costs.

bompi88, 2018-04-25:
@erossignon I get similar behaviour using the Prosys simulator. The client is not able to reuse the session. After a connection drop, I get:

Connection reestablished -> Client closed

The session status I get in the client close listener is `BadSessionClosed (0x80260000)`.

erossignon, 2018-04-25:
@bompi88  could you be more specific ? node-opcua version involved ?  how was made the disconnection ( server being shutdown ? cable disconnected ? network driver disable ?). Is the server alive all along ? 
What is the node-opcuja client you're using ( node-red-iiot-opcua ? your own code ? any sample code ? which one ?) 
can you share a simple gist for a simple client which can reproduce the issue ?

bompi88, 2018-06-18:
@erossignon More information. I'm sorry for the late reply.
- Using the newest node-opcua `0.4.1`.
- The disconnection was made by killing the server.
- This happens on Prosys and UaAnsiCServer, as far as I know.
- Used in custom code, which wrap the opcua logic inside a final state machine.
- The down time is within seconds ~5 seconds.

Here's the debug log for most of the relevant node-opcua packages. It seems that the server does not accept transferring the subscription in this state, as the server has lost its' own state? The problem is that the client emits a close event, but nothing is done with the actual event as far as I can see. Should we recreate the client, session and subscription in these cases?

```
client_base                   :502    OPCUAClientBase emitting close Error: Connection Break
    at ClientTCP_transport.<anonymous> (/Users/bjorbrat/projects/edge-agent/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:594:45)
    at ClientTCP_transport.emit (events.js:127:13)
    at Socket._on_socket_error_after_connection (/Users/bjorbrat/projects/edge-agent/node_modules/node-opcua-transport/src/client_tcp_transport.js:196:18)
    at Socket.emit (events.js:132:15)
    at emitErrorNT (internal/streams/destroy.js:64:8)
    at process._tickCallback (internal/process/next_tick.js:114:19)
client_base                   :279   _recreate_secure_channel...
client_base                   :166    DESTROYING SECURE CHANNEL  false
client_base                   :181   byteWritten  =  0
client_base                   :182   byteRead     =  0
client_base                   :315   OPCUAClientBase:  RECONNECTED                                       !!!
client_base                   :525   secureChannel#on(close) => _recreate_secure_channel returns  OK
undefined
reconnection                  :259    Starting sessions reactivation
reconnection                  :237   TRYING TO REACTIVATE EXISTING SESSION  ns=1;i=3564249996
reconnection                  :238     SubscriptionIds : [ 3564249994 ]
reconnection                  :245   ActivateSession :   ServiceFault returned by server { /*ServiceFault*/
 responseHeader                /* ResponseHeader                   */: {
   timeStamp                   /* UtcTime                          */: 2018-06-18T14:08:46.256Z
   requestHandle               /* IntegerId                        */: 2               0x2
   serviceResult               /* StatusCode                       */: BadSessionIdInvalid (0x80250000)
   serviceDiagnostics          /* DiagnosticInfo                   */: {
     namespaceUri              /* Int32                            */: -1
     symbolicId                /* Int32                            */: -1
     locale                    /* Int32                            */: -1
     localizedText             /* Int32                            */: -1
     additionalInfo            /* String                           */: null
     innerStatusCode           /* StatusCode                       */: Good (0x00000)
     innerDiagnosticInfo       /* DiagnosticInfo                   */: {
     }
   }
   stringTable                 /* String                        [] */: [ ] (l=1)
   additionalHeader            /* ExtensionObject                  */: null
 }
};
reconnection                  :118       => suspend old session publish engine....
reconnection                  :124       => creating a new session ....
opcua_client                  :261   revised session timeout =  60000
reconnection                  :128       => creating a new session (based on old session data).... Done
reconnection                  :138       => activating a new session ....
reconnection                  :141       =>  activating a new session .... Done
reconnection                  :151     session subscriptionCount =  1
reconnection                  :156       => asking server to transfer subscriptions = [ 3564249994 ]
client_base                   :543   connection_reestablished has failed
client_base                   :925   warning : disconnection : closing pending sessions
client_session                :820   ClientSession#emitCloseEvent
client_base                   :166    DESTROYING SECURE CHANNEL  false
client_base                   :181   byteWritten  =  2085
client_base                   :182   byteRead     =  12320
client_base                   :953    EMIT NORMAL CLOSE
opcua_client                  :261   revised session timeout =  60000
client_subscription           :132   ClientSubscription created
client_subscription           :161   registering callback
client_subscription           :162   publishingInterval                1000
client_subscription           :163   lifetimeCount                     15
client_subscription           :164   maxKeepAliveCount                 5
client_subscription           :165   publish request timeout hint =    15000
client_subscription           :166   hasTimedOut                       false
```

biancode, 2018-06-18:
I had the same problem with v2.x of the ProSys server - now with v3.x it works mostly well.

[image]

biancode, 2018-06-18:
... and I had to try a bit with the strategy -  @erossignon is that a good or bad strategy?

[image]

bompi88, 2018-06-19:
@biancode Seems that upgrade to Prosys 3.0.0 did the trick, but restarting Unified Automation UaAnsiCServer does still create the same behaviour.

erossignon, 2018-06-20:
@bompi88 , @biancode  what is the version of node-opcua used ?

bompi88, 2018-06-20:
`0.4.1` in my case.

biancode, 2018-06-20:
0.4.1 in my case too

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

biancode, 2018-06-20:
Nodejs 9 because some packages doesn't work with 10 in my environment for now

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

bompi88, 2018-06-20:
Nodejs 9 here too, but I have tried it in nodejs 8 as well.

luisfelipe998, 2019-07-22:
Hi. I'm at the version 0.7.0 and I'm facing this problem too. I connect to the opcua server and then, remove the ethernet cable from my computer, after a while I reconnect it. It is called the 'after_reconnection' event, but the message 'closing pending sessions' appears and I am not able to reconnect to the server anymore if I don't restart the client.

To temporary solve this problem, I disconnect the previous session and create a new one on the 'after_reconnection' event. Also, I have noticed that sometimes, it takes about 2 minutes to the client to reconnect to the server (to call the 'after_reconnection') after I have put the ethernet cable back.

Is there any news about this?
Thanks.

erossignon, 2019-11-14:
node-opcua version > 2.2.0 addresses reconnection issue better

closed 2019-11-14 by erossignon

wakeupbits, 2021-03-28:
Hi I am also getting same issue after reconnect changes value is not receiving