node-opcua/node-opcua #459: install node-opcua

TiptoeOkin, 2018-03-19:
**I'm submitting a ...**  (check one with "x")

 - ( x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

tried to install node-opcua on an Ubuntu 16.04.4 LTS machine with:
sudo npm install node-opcua --save

> node-opcua-client@0.2.2 postinstall /home/nikolai/WNS-Weather/node_modules/node-opcua-client
> node test_helpers/create_certificates.js certificate -s -o certificates/client_selfsigned_cert_1024.pem

/home/nikolai/WNS-Weather/node_modules/yargs/yargs.js:1100
      else throw err
           ^

Error: EACCES: permission denied, mkdir '/home/nikolai/WNS-Weather/node_modules/node-opcua-client/certificates'
    at Object.fs.mkdirSync (fs.js:885:18)
    at mkdir (/home/nikolai/WNS-Weather/node_modules/node-opcua-pki/lib/pki/toolbox.js:76:12)
    at readConfiguration (/home/nikolai/WNS-Weather/node_modules/node-opcua-pki/lib/crypto_create_CA.js:165:5)
    at /home/nikolai/WNS-Weather/node_modules/async/dist/async.js:3866:24
    at replenish (/home/nikolai/WNS-Weather/node_modules/async/dist/async.js:998:17)
    at /home/nikolai/WNS-Weather/node_modules/async/dist/async.js:1002:9
    at eachOfLimit (/home/nikolai/WNS-Weather/node_modules/async/dist/async.js:1027:24)
    at /home/nikolai/WNS-Weather/node_modules/async/dist/async.js:1032:16
    at _parallel (/home/nikolai/WNS-Weather/node_modules/async/dist/async.js:3865:5)
    at Object.series (/home/nikolai/WNS-Weather/node_modules/async/dist/async.js:4721:5)
npm WARN wns-weather@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-opcua-client@0.2.2 postinstall: `node test_helpers/create_certificates.js certificate -s -o certificates/client_selfsigned_cert_1024.pem`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-opcua-client@0.2.2 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/nikolai/.npm/_logs/2018-03-19T10_14_24_276Z-debug.log

**Expected behavior**

**Minimal reproduction of the problem with instructions**

  1. ________________
  2. ________________
  3. ________________
  ...
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

Ubuntu 16.04.4 LTS 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:** ``

* **Node:** 
   `node --version` = 8.10.0

TiptoeOkin, 2018-03-23:
i have read to a lot of discussions about problems with the certificates.
But I could find nothing that matched.
open ssl is setup properly
It's not a mac
I'am not behind a company firewall
Could someone point me out what I'am missing?

erossignon, 2018-03-23:
This may be related to the old aging NPM EACCESS error https://docs.npmjs.com/getting-started/fixing-npm-permissions

closed 2018-08-25 by erossignon

sounavo, 2020-03-13:
I am running into the same issue.