node-opcua/node-opcua #462: Write into a variable with Write

davidemess, 2018-03-20:
**I'm submitting a ...**  (check one with "x")

 - ( ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - (x) support request 

**Current behavior**
Hi, 
I'm trying to write on a variable on my opcua server (installed on a PLC omron). The variable is writeable, a simple Int32 called opcContatore and setted 12, I'm trying to write 25. 
The code don't send me a error, but don't write into the variable.
Can you finde the mistake? Thank you in advance.

ps:   if i add " statusCode: Good, " to nodesToWrite it send me an error:
       ReferenceError: Good is not defined

**Minimal reproduction of the problem with instructions**

    let val = await session.read({nodeId: "ns=4;s=opcContatore"})
    console.log(val.value.value)
    let nodesToWrite = {
        nodeId: "ns=4;s=opcContatore",
        attributeIds: opcua.AttributeIds.Value,
        value: {
            value: {
            dataType: opcua.DataType.Int32,
            value: 25
            }
        }
    }
    let risultato = await session.write(nodesToWrite,function (err,Data) {
        if(err) { return callback(err);}
        else{return Data}
    });
    console.log(risultato)
    val = await session.read({nodeId: "ns=4;s=opcContatore"})
    console.log(val.value.value)
------------------------------------------------------------------------------------------------
output: 
     12
     undefined
     12

expected output:
     12
      .....(maybe undefined again)
     25
------------------------------------------------------------

**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : __10_______
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:___O______
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:** ``
0.2.2
* **Node:** 
   `node --version` = v8.10.0

davidemess, 2018-03-20:
Using writeSingleNode I do it:

    let val = await session.read({nodeId: "ns=4;s=opcContatore"})
    console.log(val.value.value)

    dataToWrite = {
        dataType: "Int32",
        value: 25
     }

    let risultato = await session.writeSingleNode("ns=4;s=opcContatore", dataToWrite)
    console.log(risultato)
    val = await session.read({nodeId: "ns=4;s=opcContatore"})
    console.log(val.value.value)

-------------------------------------------------------------
output:
        0
        { [Number: 0] value: 0, description: 'No Error', name: 'Good' }
        25

davidemess, 2018-03-21:
post scriptum:

I tryed this but doesn't send me error and doesn't change te variable: 
    var nodesToWrite = [
        {
            nodeId: "ns=4;s=opctest1_Copy1",
            attributeId: opcua.AttributeIds.Value,
            indexRange: null,
            value: { /* dataValue*/
                serverTimestamp: new Date(),
                serverPicoseconds: 20,
                sourceTimestamp: new Date(),
                sourcePicoseconds: 30,
                value: { /* Variant */
                    dataType: opcua.DataType.UInt16,
                    value: 5678        }
            }
        }
    ];
    await session.write(nodesToWrite);

erossignon, 2018-03-23:
* It looks like you're mixing the callback style with the async/await style.
* use ````statusCode: opcua.StatusCodes.Good ````

try this : 
```` javascript
let val = await session.read({nodeId: "ns=4;s=opcContatore"})
console.log(val.value.value);

let nodeToWrite = {
    nodeId: "ns=4;s=opcContatore",
    attributeIds: opcua.AttributeIds.Value,
    value: {
        value: {
        dataType: opcua.DataType.Int32,
        value: 25
        }
    },
    statusCode: opcua.StatusCodes.Good // <==== 
}
let risultato = await session.write(nodeToWrite);
console.log(risultato);

val = await session.read({nodeId: "ns=4;s=opcContatore"})
console.log(val.value.value)
````

davidemess, 2018-03-23:
Hi Etienne,

Thank you for your answer. I've tryed your code and the output is this:

25
{ [Number: 2151350272]
  value: 2151350272,
  description: 'The access level does not allow writing to the Node.',
  name: 'BadNotWritable' }
25

So yes, the variable isn't Writable. I was pretty sure that was. Thank you again, i will close the post.

davidemess, 2018-03-23:
I'm sorry but, I'm trying write the variable from a client opcua python, and the variable on the server change as expected.
So the variable is writeable. I don't know what's going on

erossignon, 2018-03-23:
It could depend on the session credential. Are you using anonymous or login/password to create a session ? It is possible that some users  may have read/only access and some other read/write access.

erossignon, 2018-03-23:
Also double check your nodeId.

davidemess, 2018-03-23:
Is an user/password session for both, nodejs and python. I've tryed with anonymous login but same story. The node Id is correct, I have try the code with different nodes and different dataType (of curse changing the datatype in the 'nodeToWrite').

erossignon, 2018-03-23:
OK, could you provide a full script of your node-opcua as a gist ? ( mimum code to reproduce the issue ) ?

davidemess, 2018-03-23:
simply this:
<script src="https://gist.github.com/davidemess/eeb2d8eee624c448cc4ac76c9bbdb70a.js"></script>
I would use Write instead of writeSigleNode because I would pass an Array of NodesToWrite.
I can write in the node with WriteSingleNode, but with Write I can't.

erossignon, 2018-03-24:
There is a typo in your code.
```javascript
 let nodeToWrite = {
    nodeId: "ns=4;s=opcContatore",
    attributeIds: opcua.AttributeIds.Value,
 }
```
```` attributeIds: opcua.AttributeIds.Value,```` should be ```attributeId: opcua.AttributeIds.Value,``` (singular)

see https://gist.github.com/erossignon/48ebf30d501b8d51472cac573e5160b2

You owe me a 🍺 or a 💲 😃

davidemess, 2018-03-26:
Ops. I'm sorry, I don't know where I find "Ids". Thank you very much!

closed 2018-03-26 by davidemess

bm-q, 2018-06-15:
@erossignon Hi Etienne, I've made the very same mistake and this led me to think that we both copied some wrong piece of code: I've searched a bit and found it, check it at 

http://node-opcua.github.io/api_doc/0.2.0/classes/ClientSession.html#methods

Thank you!

tommasocaporuscio, 2019-03-08:
@erossignon erossignon Hi Etienne I have the same problem my code:

  function _write_node_value(session,nodeId,value,attributeId,expectedStatusCodes,callback) {
       
    var nodesToWrite = [
        {
            nodeId: 'ns=4;s=OPC.byNewFinger',
            attributeId: opcua.AttributeIds.Value,
            indexRange: null,
            value: { /* dataValue*/
                serverTimestamp: new Date(2015, 5, 2),
                serverPicoseconds: 20,
                sourceTimestamp: new Date(2015, 5, 3),
                sourcePicoseconds: 30,
                value: { /* Variant */
                    
                     value: true
                }
          }
        }
    ];

    the_session.write(nodesToWrite, function(err, status, diag) {
  console.log(err);
  console.log(status);
  console.log(diag);
});

}

result is:
[ ConstantStatusCode {
    value: 2155020288,
    description:
     'The server not does support writing the combination of value',
    name: 'BadWriteNotSupported' } ]
[]

Solution?

mahboobelahi93, 2019-08-15:
@erossignon erossignon Hi Etienne I am using the following code for writing node value but it seems that  **await session.write(nodeToWrite);** does not eork correctly also i am attaching pic of error.
`        let val = await session.read(nodeIds[0])
        console.log(val.value.toString());

        let nodeToWrite = {
            nodeId: nodeIds[0],// 'ns=3;s="INPUT_1"'
            attributeId: opcua.AttributeIds.Value,
            indexRange: null,
            value: /* DataValue */ {
                //sourceTimestamp: new Date(),
                statusCode: opcua.StatusCodes.Good,// <==== 
                value: /* Variant */ {
                    dataType: opcua.DataType.Boolean,
                    value: true
                }
            },
        }
        let risultato = await session.write(nodeToWrite);
        console.log(risultato.toString());

        val = await session.read(nodeIds[0])
        console.log(val.value.toString())`

[image]

mikakaraila, 2025-02-18:
> Just delete the extra keys
> Just delete the extra keys in the value of nodesToWrite, and it should be
> like this:

Some servers don't like that client wants to write status code or time
stamps