node-opcua/node-opcua #465: on Linux, node-opcua cause the first path in NODE_PATH env not working.

boogiebug, 2018-03-23:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**
When running node-opcua on Linux with 'NODE_PATH' environment setting, the newly appended path is not working.

Inspecting into node-opcua/index.js, I found the 3rd line in the file as following:

```javascript
process.env.NODE_PATH=__dirname + ";" + process.env.NODE_PATH;
```

Since, the semi-colon is as the path separator on Windows platform only, this cause the problem when running node-opcua on Linux.

I'd suggest the following code to fix:

```javascript
process.env.NODE_PATH=__dirname + ( process.platform === 'win32' ? ';' : ':') + process.env.NODE_PATH; 
```

erossignon, 2018-03-24:
what version of node-opcua are you reffering to  ?

boogiebug, 2018-03-29:
Ooopss.. Sorry, just realized that I'm using the older version. There's no any problem with the current one.

closed 2018-03-29 by boogiebug