node-opcua/node-opcua #467: Failed to change event filter on EventNotifierObject

bruegth, 2018-03-26:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request

**I will make an pull request to fix that behavior, too.**

**Current behavior**

When changing EventFilter the method ModifyMonitoredItemsRequest will failed because of AssertionError:
```
EXCEPTION CAUGHT WHILE PROCESSING REQUEST !! ModifyMonitoredItemsRequest
{ /*ModifyMonitoredItemsRequest*/
 requestHeader                 /* RequestHeader                    */: {
   authenticationToken         /* NodeId                           */: ns=0;b=56e5537810df2a386d3a292d60171622
   timeStamp                   /* UtcTime                          */: 2018-03-23T12:37:15.736Z
   requestHandle               /* IntegerId                        */: 1011191               0xf6df7
   returnDiagnostics           /* UInt32                           */: 0               0x0
   auditEntryId                /* UAString                         */: null
   timeoutHint                 /* UInt32                           */: 10000               0x2710
   additionalHeader            /* ExtensionObject                  */: null
 }
 subscriptionId                /* IntegerId                        */: 414511               0x6532f
 timestampsToReturn            /* TimestampsToReturn               */: Both ( 2)
 itemsToModify                 /* MonitoredItemModifyRequest    [] */: [
   { /*0*/
     monitoredItemId           /* IntegerId                        */: 2               0x2
     requestedParameters       /* MonitoringParameters             */: {
       clientHandle            /* IntegerId                        */: 10               0xa
       samplingInterval        /* Duration                         */: 500
       filter                  /* ExtensionObject                  */: { /*EventFilter*/
 selectClauses                 /* SimpleAttributeOperand        [] */: [
   { /*0*/
     typeId                    /* NodeId                           */: ns=0;i=2041
     browsePath                /* QualifiedName                 [] */: [
       { /*0*/
         namespaceIndex        /* UInt16                           */: 0
         name                  /* String                           */: EventId
       }
     ]
     attributeId               /* IntegerId                        */: 13               0xd
     indexRange                /* NumericRange                     */: NumericRange:<Empty>
   },
   { /*1*/
     typeId                    /* NodeId                           */: ns=0;i=2041
     browsePath                /* QualifiedName                 [] */: [
       { /*0*/
         namespaceIndex        /* UInt16                           */: 0
         name                  /* String                           */: EventType
       }
     ]
     attributeId               /* IntegerId                        */: 13               0xd
     indexRange                /* NumericRange                     */: NumericRange:<Empty>
   },
  ..... ( x elements )
 ]
 whereClause                   /* ContentFilter                    */: {
   elements                    /* ContentFilterElement          [] */: null []
 }
};
       queueSize               /* Counter                          */: 4294967295
       discardOldest           /* Boolean                          */: true
     }
   }
 ]
};
 display_trace_from_this_project_only =
_.isNumber(node_minimumSamplingInterval,"expecting a number")

 messages  EXCEPTION CAUGHT WHILE PROCESSING REQUEST !!! ModifyMonitoredItemsRequest
_.isNumber(node_minimumSamplingInterval,"expecting a number")
AssertionError [ERR_ASSERTION]: _.isNumber(node_minimumSamplingInterval,"expecting a number")
```

**Expected behavior**

Can change event filters without AssertionError

**Minimal reproduction of the problem with instructions**

  1. Create event notifier object on nodejs server addresspace
  2. Connect via UAExpert client and try to modify event filter

**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )

erossignon, 2018-03-27:
In fact, as per specification, mimumSamplingInterval is an optional attribute of a UAVariable but is not a attribute of a UAObject/UAObjectType

 see OPCUA 1.03 5.6.2 part 3 
[image]

erossignon, 2018-03-27:
this being said , when a client attempt to modified the MonitoringParameter of a Event , it should handle this case gracefully without asserting

bruegth, 2018-03-29:
Sorry you are right and I missed to check the specification for that.

closed 2018-07-02 by erossignon