node-opcua/node-opcua #471: API documentation is totally wrong.

jonasclaes, 2018-04-01:
Hello,
I am submitting an issue for the API documentation.
I was reading up on how to read and write some variables, but the API is very, very wrong.
For example, the demo client project uses the opc session .readVariableValue(). I can't find this in the API documentation for v0.2.0.

Am I looking at the wrong API docs? Or are the docs just wrong?

I need this documentation for a business project that has to be done next month.

Kind regards,
Jonas Claes

msalafia, 2018-04-01:
It seems correct. The **readVariableValue()** method belongs to ClientSession as you can see from [here](http://node-opcua.github.io/api_doc/0.2.0/classes/ClientSession.html). 
If you are looking the demo client you are using the method **createSession()** that, as you can see [here](http://node-opcua.github.io/api_doc/0.2.0/files/packages_node-opcua-client_src_opcua_client.js.html#l584), returns the mentioned _ClientSession_.
Hope this can help you.

jonasclaes, 2018-04-02:
Hello,

Thanks for clearing this out for me. I now understand the documentation style.
Thanks a lot, you saved my deadline.

Kind regards,
Jonas Claes

closed 2018-04-02 by jonasclaes