node-opcua/node-opcua #474: reconnect - monitored items do not reconnect if connection was lost

biancode, 2018-04-15:
**I'm submitting a ...**  (check one with "x")

 - ( x) bug report => search github for a similar issue or PR before submitting

**Current behavior**

` Setting debug for  client_publish_engine           to  false
  opcuaIIoT:connector !!!!!!!!!!!!!!!!!!!!!!!!  CLIENT STARTING RECONNECTION !!!!!!!!!!!!!!!!!!! +25s
  opcuaIIoT:connector CONNECTION STARTING RECONNECTION: opc.tcp://desktop-cnf0dge.local:53530/OPCUA/SimulationServer +0ms
15 Apr 09:16:50 - [error] [OPCUA-IIoT-Read:8e406e2c.cbc77] Error: Client Not Open On Read
15 Apr 09:17:00 - [error] [OPCUA-IIoT-Read:8e406e2c.cbc77] Error: Client Not Open On Read
15 Apr 09:17:10 - [error] [OPCUA-IIoT-Read:8e406e2c.cbc77] Error: Client Not Open On Read
  opcuaIIoT:connector !!! CONNECTION FAILED FOR # 0  retrying  1.12  sec. !!! +26s
  opcuaIIoT:connector CONNECTION FAILED: opc.tcp://desktop-cnf0dge.local:53530/OPCUA/SimulationServer +0ms
  opcuaIIoT:connector undefined +1s
  opcuaIIoT:connector CONNECTION RE-ESTABLISHED: opc.tcp://desktop-cnf0dge.local:53530/OPCUA/SimulationServer +0ms
  opcuaIIoT:connector !!!!!!!!!!!!!!!!!!!!!!!!      CLIENT RECONNECTED     !!!!!!!!!!!!!!!!!!! +6ms
  opcuaIIoT:connector CONNECTION RECONNECTED: opc.tcp://desktop-cnf0dge.local:53530/OPCUA/SimulationServer +1ms
  opcuaIIoT:connector Request For New Session From after_reconnection +0ms
  opcuaIIoT:connector Working Session On Start Request +0ms
  opcuaIIoT:connector Session Closed With StatusCode 2149974016 +218ms
  opcuaIIoT:connector Session Closed +0ms
ERROR !!! , please check here !!!! callback may be called twice !!  { AssertionError [ERR_ASSERTION]: this.isSuspended !== !!flag,"invalid state"
    at ClientSidePublishEngine.suspend (node_modules/node-opcua-client/src/client_publish_engine.js:52:5)
    at start_publishing_as_normal (node_modules/node-opcua-client/src/opcua_client.js:912:60)
    at node_modules/async/dist/async.js:3866:24
    at replenish (node_modules/async/dist/async.js:998:17)
    at iterateeCallback (node_modules/async/dist/async.js:983:17)
    at node_modules/async/dist/async.js:958:16
    at node_modules/async/dist/async.js:3871:13
    at node_modules/node-opcua-client/src/opcua_client.js:734:9
    at node_modules/async/dist/async.js:473:16
    at replenish (node_modules/async/dist/async.js:993:25)
    at node_modules/async/dist/async.js:1002:9
    at eachOfLimit (node_modules/async/dist/async.js:1027:24)
    at node_modules/async/dist/async.js:1032:16
    at Object.eachOf (node_modules/async/dist/async.js:1103:5)
    at ClientSidePublishEngine.republish (node_modules/node-opcua-client/src/client_publish_engine.js:487:11)
    at OPCUAClient._ask_for_subscription_republish (node_modules/node-opcua-client/src/opcua_client.js:730:32)
  generatedMessage: false,
  name: 'AssertionError [ERR_ASSERTION]',
  code: 'ERR_ASSERTION',
  actual: undefined,
  expected: undefined,
  operator: undefined }
MessageBuilder : ERROR DETECTED IN event handler
Error: Callback was already called.
    at node_modules/async/dist/async.js:955:32
    at node_modules/async/dist/async.js:3871:13
    at node_modules/node-opcua-client/src/opcua_client.js:876:33
    at node_modules/node-opcua-client/src/opcua_client.js:527:24
    at node_modules/node-opcua-client/src/client_session.js:1149:20
    at modified_callback (node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1007:13)
    at process_request_callback (node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:78:5)
    at ClientSecureChannelLayer._on_message_received (node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:129:5)
    at MessageBuilder.emit (events.js:127:13)
    at MessageBuilder._decode_message_body (node_modules/node-opcua-secure-channel/src/message_builder.js:459:22)
    at MessageBuilderBase._feed_messageChunk (node_modules/node-opcua-transport/src/message_builder_base.js:196:18)
    at PacketAssembler.<anonymous> (node_modules/node-opcua-transport/src/message_builder_base.js:53:14)
    at PacketAssembler.emit (events.js:127:13)
    at PacketAssembler.feed (node_modules/node-opcua-packet-assembler/src/packet_assembler.js:107:14)
    at MessageBuilderBase.feed (node_modules/node-opcua-transport/src/message_builder_base.js:159:30)
    at ClientSecureChannelLayer._on_receive_message_chunk (node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:869:25)
Error: Callback was already called.
    at node_modules/async/dist/async.js:955:32
    at node_modules/async/dist/async.js:3871:13
    at node_modules/node-opcua-client/src/opcua_client.js:876:33
    at node_modules/node-opcua-client/src/opcua_client.js:527:24
    at node_modules/node-opcua-client/src/client_session.js:1149:20
    at modified_callback (node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1007:13)
    at process_request_callback (node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:78:5)
    at ClientSecureChannelLayer._on_message_received (node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:129:5)
    at MessageBuilder.emit (events.js:127:13)
    at MessageBuilder._decode_message_body (node_modules/node-opcua-secure-channel/src/message_builder.js:459:22)
    at MessageBuilderBase._feed_messageChunk (node_modules/node-opcua-transport/src/message_builder_base.js:196:18)
    at PacketAssembler.<anonymous> (node_modules/node-opcua-transport/src/message_builder_base.js:53:14)
    at PacketAssembler.emit (events.js:127:13)
    at PacketAssembler.feed (node_modules/node-opcua-packet-assembler/src/packet_assembler.js:107:14)
    at MessageBuilderBase.feed (node_modules/node-opcua-transport/src/message_builder_base.js:159:30)
    at ClientSecureChannelLayer._on_receive_message_chunk (node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:869:25)
  opcuaIIoT:connector Request For New Session From Handle Session Close +535ms
  opcuaIIoT:connector Start New Session +0ms
  opcuaIIoT:connector Session Created On opc.tcp://desktop-cnf0dge.local:53530/OPCUA/SimulationServer For Handle Session Close +189ms
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! 
 warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel `

**Expected behavior**

should reconnect and re-subscribe as before on monitored itmes

**Minimal reproduction of the problem with instructions**

  1. open a client connection
  2. stop the server for the client connection
  3. restart the server for the client
  4. it crashes on reconnecting
  
**What is the motivation / use case for changing the behavior?**

it is a core behavior to get reconnected on lost OPC UA communication

**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( ) I have installed node-opcua as a package ( using npm install )
  * ( x) I am using an application that uses node-opcua 
  
       - ( x) node-red
       - ( ) other : please specify 
        
  * Device: macbook pro
  * OS version: _____
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( x) macOs   : version : 10.13
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
[image]

     
    
* **node-opcua version:** ``

 node-opcua 0.2.3
* **Node:** 
   `node --version` = 9.7.1

biancode, 2018-04-15:
hint: the reconnect creates two sessions in the simulation server

biancode, 2018-04-15:
hint: v0.2.2 of node-opcua works very well in that use case

erossignon, 2018-04-15:
@biancode this bug has probably been fix with the latest commit (d5d9e51bc219de1ab242640714fbc40dc946d72b)  ... please check with master as your scenario will help us validating recent improvement (not released yet) in reconnection.

erossignon, 2018-04-15:
Here is a standalone test file that you could use , it has been adpated to run with the ProsysDemo server :2e9e7519fc16720f3eeb1cef9b6bfeac

closed 2018-09-23 by erossignon

samminen, 2019-01-28:
I have same problem now. Whenever OPC UA Server comes up, it create all node GUIDs dynamically (no static GUIDs for all nodes). So, on reconnect, I need to start creating session and start node crawl to get new GUIDs and start subscription again and monitor for values. How can we initiate start creating session and node crawl on reconnect (I am using async.series() callback mechanism)  ?