node-opcua/node-opcua #478: InductiveAutomation OPC-UA quick client fails to read node-opcua server variables

DimitrijeManic, 2018-04-27:
**I'm submitting a ...**  (check one with "x")

 - (X) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

Inductive Automation (Ignition) OPC-UA quick client displays all tags as expected but fails when attempting to read or subscribe to a variable. 
The error displayed in the quick client when attempting to read a variable : 
```
Status: Bad
Value: null
Quality: [Bad] Bad; unspecified.
``` 
**Expected behavior**

Quick client should properly display the value
**Minimal reproduction of the problem with instructions**
1. Run the node-opcua simple_server.js example
2. Connect to the server using Inductive Automation 
3. Attempt to read any server variable using the quick client
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

Ubuntu 16.04.4 LTS
package.json : 
```
node-opcua-samples@0.3.0
node-opcua: "^0.3.0"
```
  * (X) I have installed node-opcua from source ( using git clone)
  * (X) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - (X) other : Inductive Automation
        
  * Device: _____
  * OS version: Ubuntu 16.04.4 LTS
    
    - ( ) Windows : version : _________
    - (X) Linux   : version : Ubuntu 16.04.4 LTS
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: Ignition by Inductive Automation
     - Version: Ignition-7.9.7
     - Manufacturer/Software vendor: Inductive Automation
     - link : https://inductiveautomation.com     
    
* **node-opcua version:** ``

* **Node:** 
   `node --version` = v8.9.3
[image]

DimitrijeManic, 2018-05-02:
This is the dump produced from executing the following steps;
1. start OPCUA server
2. Add OPCUA server to Ignitions OPCUA server list
3. Open the folder structure in Ignition
4. Attempt to read value from "FreeMemory"

```
dmanic:~/development/connect/Connect(wip/opcua)$ node opcua/server.js 
 Setting debug for  ua_variable                     to  true
 Setting debug for  ua_variable_type                to  true
 Setting debug for  condition                       to  true
 Setting debug for  argument_list                   to  true
 Setting debug for  opcua_server                    to  true
 Setting debug for  server_subscription             to  true
 Setting debug for  monitored_item                  to  true
 Setting debug for  server_publish_engine           to  true
 Setting debug for  server_session                  to  true
 Setting debug for  server_engine                   to  true
 Setting debug for  server_end_point                to  true
 Setting debug for  server_secure_channel_layer     to  true
server_engine                 :491   Loading  /home/dmanic/development/connect/Connect/node_modules/node-opcua-nodesets/nodesets/Opc.Ua.NodeSet2.xml ...
server_engine                 :502   Loading  /home/dmanic/development/connect/Connect/node_modules/node-opcua-nodesets/nodesets/Opc.Ua.NodeSet2.xml  done :  793  ms
 ------------------------------ binging  ServerDiagnosticsSummary ns=0;i=2275
ua_variable                   :441    Warning:  UAVariable#readValue ServerDiagnosticsSummary (ns=0;i=2275) exists but dataValue has not been defined
xxx  dataType ns=0;i=7 ServerViewCount                     valueRank -1  UInt32                    basicType =  UInt32               ns=0;i=2276 Good (0x00000)
xxx  dataType ns=0;i=7 CurrentSessionCount                 valueRank -1  UInt32                    basicType =  UInt32               ns=0;i=2277 Good (0x00000)
xxx  dataType ns=0;i=7 CumulatedSessionCount               valueRank -1  UInt32                    basicType =  UInt32               ns=0;i=2278 Good (0x00000)
xxx  dataType ns=0;i=7 SecurityRejectedSessionCount        valueRank -1  UInt32                    basicType =  UInt32               ns=0;i=2279 Good (0x00000)
xxx  dataType ns=0;i=7 RejectedSessionCount                valueRank -1  UInt32                    basicType =  UInt32               ns=0;i=3705 Good (0x00000)
xxx  dataType ns=0;i=7 SessionTimeoutCount                 valueRank -1  UInt32                    basicType =  UInt32               ns=0;i=2281 Good (0x00000)
xxx  dataType ns=0;i=7 SessionAbortCount                   valueRank -1  UInt32                    basicType =  UInt32               ns=0;i=2282 Good (0x00000)
xxx  dataType ns=0;i=7 CurrentSubscriptionCount            valueRank -1  UInt32                    basicType =  UInt32               ns=0;i=2285 Good (0x00000)
xxx  dataType ns=0;i=7 CumulatedSubscriptionCount          valueRank -1  UInt32                    basicType =  UInt32               ns=0;i=2286 Good (0x00000)
xxx  dataType ns=0;i=7 PublishingIntervalCount             valueRank -1  UInt32                    basicType =  UInt32               ns=0;i=2284 Good (0x00000)
xxx  dataType ns=0;i=7 SecurityRejectedRequestsCount       valueRank -1  UInt32                    basicType =  UInt32               ns=0;i=2287 Good (0x00000)
xxx  dataType ns=0;i=7 RejectedRequestsCount               valueRank -1  UInt32                    basicType =  UInt32               ns=0;i=2288 Good (0x00000)
 ------------------------------ binging  ServerStatus ns=0;i=2256
ua_variable                   :441    Warning:  UAVariable#readValue ServerStatus (ns=0;i=2256) exists but dataValue has not been defined
xxx  dataType ns=0;i=2 StartTime                           valueRank -1  UtcTime                   basicType =  DateTime             ns=0;i=2257 Good (0x00000)
xxx  dataType ns=0;i=2 CurrentTime                         valueRank -1  UtcTime                   basicType =  DateTime             ns=0;i=2258 Good (0x00000)
xxx  dataType ns=0;i=8 State                               valueRank -1  ServerState               basicType =  Int32                ns=0;i=2259 Good (0x00000)
xxx  dataType ns=0;i=3 BuildInfo                           valueRank -1  BuildInfo                 basicType =  ExtensionObject      ns=0;i=2260 Good (0x00000)
 ------------------------------ binging  BuildInfo ns=0;i=2260
xxx  dataType ns=0;i=1 ProductUri                          valueRank -1  String                    basicType =  String               ns=0;i=2262 Good (0x00000)
xxx  dataType ns=0;i=1 ManufacturerName                    valueRank -1  String                    basicType =  String               ns=0;i=2263 Good (0x00000)
xxx  dataType ns=0;i=1 ProductName                         valueRank -1  String                    basicType =  String               ns=0;i=2261 Good (0x00000)
xxx  dataType ns=0;i=1 SoftwareVersion                     valueRank -1  String                    basicType =  String               ns=0;i=2264 Good (0x00000)
xxx  dataType ns=0;i=1 BuildNumber                         valueRank -1  String                    basicType =  String               ns=0;i=2265 Good (0x00000)
xxx  dataType ns=0;i=2 BuildDate                           valueRank -1  UtcTime                   basicType =  DateTime             ns=0;i=2266 Good (0x00000)
[... cut ...]
      "value": {
        "dataType": "Double",
        "arrayType": "Scalar",
        "value": 12.226142828340407
      },
      "statusCode": {
        "value": 0,
        "description": "No Error",
        "name": "Good"
      },
      "sourceTimestamp": "2018-05-02T14:45:17.985Z",
      "sourcePicoseconds": 6,
      "serverTimestamp": "2018-05-02T14:45:17.985Z",
      "serverPicoseconds": 6
    }
  ],
  "diagnosticInfos": []
}
```

Could it be that Ignition is expecting a different message format?

Mhorta, 2018-05-08:
I have the same problem, the ignition log says:

org.eclipse.milo.opcua.stack.core.UaSerializationException: max array length exceeded (length=66003, max=65535)

erossignon, 2018-05-09:
Thank you @Mhorta , it looks like the version of InductiveAutomation's Ignition client is affected by the "picosecond" bug in Milo.
(see https://github.com/eclipse/milo/issues/215)
Please contact InductiveAutomation customer service to signal the bug and get an update.

Mhorta, 2018-05-10:
I did and open a topic on the forum. They answered me and created an update for the opcua module with the new version of milo so everything is working fine now.

DimitrijeManic, 2018-05-10:
My InductiveAutomation logs states
```
java.lang.IndexOutOfBoundsException: readerIndex(63) + length(1) exceeds writerIndex(63): CompositeByteBuf(ridx: 63, widx: 63, cap: 63, components=1)

at io.netty.buffer.AbstractByteBuf.checkReadableBytes0(AbstractByteBuf.java:1178)

at io.netty.buffer.AbstractByteBuf.readByte(AbstractByteBuf.java:587)

at io.netty.buffer.SwappedByteBuf.readByte(SwappedByteBuf.java:435)

at org.eclipse.milo.opcua.stack.core.serialization.OpcUaBinaryStreamDecoder.readDiagnosticInfo(OpcUaBinaryStreamDecoder.java:256)

at org.eclipse.milo.opcua.stack.core.serialization.OpcUaBinaryStreamDecoder.readDiagnosticInfo(OpcUaBinaryStreamDecoder.java:649)

at org.eclipse.milo.opcua.stack.core.serialization.OpcUaBinaryStreamDecoder.readArray(OpcUaBinaryStreamDecoder.java:672)

at org.eclipse.milo.opcua.stack.core.types.structured.ReadResponse$Codec.decode(ReadResponse.java:88)

at org.eclipse.milo.opcua.stack.core.types.structured.ReadResponse$Codec.decode(ReadResponse.java:77)

at org.eclipse.milo.opcua.stack.core.serialization.codecs.BuiltinDataTypeCodec.decode(BuiltinDataTypeCodec.java:25)

at org.eclipse.milo.opcua.stack.core.serialization.codecs.BuiltinDataTypeCodec.decode(BuiltinDataTypeCodec.java:21)

at org.eclipse.milo.opcua.stack.core.serialization.codecs.GenericDataTypeCodec$GenericBinaryDataTypeCodec.decode(GenericDataTypeCodec.java:49)

at org.eclipse.milo.opcua.stack.core.serialization.OpcUaBinaryStreamDecoder.readMessage(OpcUaBinaryStreamDecoder.java:772)

at org.eclipse.milo.opcua.stack.client.handlers.UaTcpClientMessageHandler$5.onMessageDecoded(UaTcpClientMessageHandler.java:665)

at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder$AbstractDecoder.decode(ChunkDecoder.java:194)

at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decode(ChunkDecoder.java:83)

at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decodeSymmetric(ChunkDecoder.java:71)

at org.eclipse.milo.opcua.stack.client.handlers.UaTcpClientMessageHandler.lambda$onSecureMessage$10(UaTcpClientMessageHandler.java:629)

at org.eclipse.milo.opcua.stack.core.channel.SerializationQueue.lambda$decode$1(SerializationQueue.java:64)

at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$PollAndExecute.run(ExecutionQueue.java:107)

at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)

at java.util.concurrent.FutureTask.run(FutureTask.java:266)

at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)

at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)

```

This is something on their side I assume?

Mhorta, 2018-05-10:
Yes it is. I opened this threat and they sent me an update that solves this problem, you have the link on the post that i marked as solution.
https://forum.inductiveautomation.com/t/node-opc-ua-does-not-work/17953

DimitrijeManic, 2018-05-11:
Installing the module as @Mhorta suggested fixed the issue, thanks for taking the time to look into this!

closed 2018-05-11 by DimitrijeManic

erossignon, 2018-05-13:
@kevinherron  from milo kindly confirmed that this issue was fixed in Milo 0.2.1 which has been integrated for the InductiveAutomation OPC-UA quick client version 7.9.8 release.