node-opcua/node-opcua #491: Not an issue but a question ! Using node-opcua with a GSM and a RPI

tototrix, 2018-06-11:
Hello,

I apologize for using the issues tab but i don't find the right place for asking a question.

Is it possible to use node-opcua with a GPRS modem on a Raspberry ? If so, have you advices or links to start with ?

Thank you in advance.

erossignon, 2018-06-13:
If there any way to communicate with your GPRS modem in nodejs , then yes ...
this package may help https://www.npmjs.com/package/simcom

tototrix, 2018-06-13:
Thank you for your answer. 

In the "simple server example" when you define the server as below, the socket seems to be automaticaly configured and managed by either the library or by the BSP. It's obviously perfect for ethernet or Wifi:

[image]

==> Therfore i don't see how to link the node-opcua library to an other medium of communication (like à GPRS)

erossignon, 2018-06-13:
OK, I understand now that you want to use GPRS modem as a Transport Layer for the OPCUA communication. At this time the OPCUA specifications only provide TCP/IP transport. So if  your GPRS module can be seen as a Ethernet card, this should be possible. 
However, nothing exists in the OPCUA spec to describe a communication over raw AT command of a GRPS module.
The Transport layer is pretty well abstracted and one could decide to extend node-opcua to implement a dedicated transport if necessary . (big task though)

closed 2019-08-03 by stale[bot]