node-opcua/node-opcua #493: OPCUA server.engine.addressSpace.getPrivateNamespace() method not found

innoaloe, 2018-06-15:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

Following the "Simple Server" tutorial (https://github.com/node-opcua/node-opcua/blob/master/documentation/creating_a_server.md), I'm stuck at the the point to get namespace definition from the OPCUAServer engine.
It says the getPrivateNamespace() method which is required there is not a method.
I'm coding it using Visual Studio Code, and indeed the Intellisense cannot find the said method when I'm writing server.engine.addressSpace.getPrivateNamespace() outside of function definitions.

**Expected behavior**

As the tutorial suggest, this method should have existed somewhere in the API. Maybe the tutorial version has deprecated? Or am I missing something else here...

**Minimal reproduction of the problem with instructions**

  1. Just following the entire "Simple Server" tutorial. Copy-pasting the whole code to it's respective sections produces same result.
  2. ________________
  3. ________________
  ...
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

OPCUAServer,

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: Personal Computer
  * OS version: 64-bit
    
    - (x) Windows : version : Windows 10
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:** ``

Should be 0.2.0 I believe, since I use npm install node-opcua --save method as suggested from the tutorial. One weird thing though, during npm install, it says 1.0.1 no longer maintained.

* **Node:** 
   `node --version` = 8.11.3

erossignon, 2018-06-16:
In fact, the tutorial is too much ahead of the current node-opcua version (0.4.1).

getPrivateNamespace() has just been introduced recently,  you'll need to clone node-opcua and use the most recent version on master. 

We need to do something to make sure that our link points to the correct documentation.

erossignon, 2018-06-16:
I have updated http://node-opcua.github.io/ so that examples relates to current published version and not master.

closed 2018-06-16 by erossignon

innoaloe, 2018-06-16:
@erossignon Thank you for the response. Will try on the tutorial for the released version. And thank you to develop this great API.