node-opcua/node-opcua #495: Get BadUnexpectedError trying client.connect().

nnnntosh, 2018-06-21:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

When connecting to CogentDataHub-server with node-opcua-client, the server returns BadUnexpectedError (0x80010000) with GetEndpointsRequest.
The same problem occurs even when using node-opcua-samples / bin / simple_clients.js.
Also, node-opcua 0.3.0 successfully connected.

I analyzed error case and normal case communication with Wireshark. When BadUnexpectedError occurs, it is found that endpointUrl is empty.

Run on node-opcua 0.4.1 (error)
[image]

Run on node-opcua 0.3.0 (success)
[image]

I think that comment-out of the code below may be affecting.

https://github.com/node-opcua/node-opcua/blob/1096888c31dfaedd6c6105b57a75176c429f13cb/packages/node-opcua-client/src/client_base.js#L726

[comparing code]

[comparing code]: https://github.com/node-opcua/node-opcua/compare/92628f0a9182478085a2e4a928a8b3f051359363...1096888c31dfaedd6c6105b57a75176c429f13cb#diff-f3fcbb4405d66e4b4fbc75ad524b8401

**Minimal reproduction of the problem with instructions**

  1. clinet.connect(endpointUrl)
  2. ________________
  3. ________________
  ...
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: Raspberry Pi 3
  * OS version: Raspbian GNU/Linux 8 (jessie)
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: CogentDataHub
     - link : https://cogentdatahub.com/features/protocols/opc-ua/
     
    
* **node-opcua version:** ``

0.4.1

* **Node:** 
   `node --version` =v8.11.3

erossignon, 2018-06-26:
what is the version of CogentDataHub that you are using ?
I have reproduce the error with CogentDataHub version 8.05.249.
It seems to be a issue with CogentDataHub that doesn't allow endpoint to be null in the GetEndpointsRequest

nnnntosh, 2018-06-28:
The version of CogentDataHub is V8.0.3.224.
As you say, this problem may be a problem with CogentDataHub.
Are you planning on node-opcua-client to workaround this problem?

erossignon, 2018-06-28:
yes, the workaround will be in next release

nnnntosh, 2018-06-29:
Thank you for your support.

closed 2018-08-23 by erossignon