node-opcua/node-opcua #497: TypeScript error TS1016: A required parameter cannot follow an optional parameter.

jared-DM, 2018-06-29:
**I'm submitting a ...**  (check one with "x")

 - (X ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

I'm getting TypeScript errors when using the latest version:

```
node_modules/node-opcua/node-opcua.d.ts:265:46 - error TS1016: A required parameter cannot follow an optional parameter.

265     findServers(options?: findServerOptions, callback: ResponseCallback<Array<ApplicationDescription>>):void;
                                        
node_modules/node-opcua/node-opcua.d.ts:266:64 - error TS1016: A required parameter cannot follow an optional parameter.

266     findServersOnNetwork(options?: findServerOnNetworkOptions, callback: ResponseCallback<Array<ServerOnNetwork>>):void
```

**Expected behavior**

No TypeScript errors

**Minimal reproduction of the problem with instructions**

 Just run tsc on a typescript project using node-opcua

**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * (X) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: Ubuntu 16.04.4 LTS
  * OS version: 16.04.4
    
    - ( ) Windows : version : _________
    - (X ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
    
* **node-opcua version:** ``
0.4.1

* **Node:** 
   `node --version` = 10.5.0

closed 2018-08-23 by erossignon

ppaul, 2019-06-21:
this is not fixed for me; I'm puzzled to find that it has a link to OS. My is Linux Mint 19

erossignon, 2019-06-21:
@ppaul , if you use typescript we suggest that you switch to node-opcua 2.0.0 that has been fully refactored to typescript.

ppaul, 2019-06-21:
sorry, I 've made a confusion, I was using something else, just the error was the same