node-opcua/node-opcua #501: Error: QueryFirst is not supported by Server

Chrissie, 2018-07-09:
**I'm submitting a ...** 

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

I can't get the node-opcua client running without encountering this problem. As a opcua client, I get this error both with the node-opcua sample server and with a 'real' OPC UA server on a Siemens PLC.   

The error message I get on the terminal is as follows:

>  -------------------------------------------------------------------- Alarms & Conditions --------------------
>  -----------------------------------------------------------------------------------------------------------------
> QueryFirst is not supported by Server
> ERROR !!! , please check here !!!! callback may be called twice !!  TypeError: opcua.readHistoryServerCapabilities is not a function
>     at _resd_historyServerCapabilities (C:\Users\MyUserFolder\node-opcua\packages\node-opcua-samples\bin\simple_client.js:679:15)
>     at C:\Users\MyUserFolder\node-opcua\packages\node-opcua-samples\node_modules\async\dist\async.js:3880:24
>     at replenish (C:\Users\MyUserFolder\node-opcua\packages\node-opcua-samples\node_modules\async\dist\async.js:1011:17)
>     at iterateeCallback (C:\Users\MyUserFolder\node-opcua\packages\node-opcua-samples\node_modules\async\dist\async.js:995:17)
>     at C:\Users\MyUserFolder\node-opcua\packages\node-opcua-samples\node_modules\async\dist\async.js:969:16
>     at C:\Users\MyUserFolder\node-opcua\packages\node-opcua-samples\node_modules\async\dist\async.js:3885:13
>     at C:\Users\MyUserFolder\node-opcua\packages\node-opcua-samples\bin\simple_client.js:672:13
>     at C:\Users\MyUserFolder\node-opcua\packages\node-opcua-client\src\client_session.js:1568:20
>     at C:\Users\MyUserFolder\node-opcua\packages\node-opcua-client\src\client_session.js:1568:20
>     at C:\Users\MyUserFolder\node-opcua\packages\node-opcua-client\src\client_session.js:1168:20
>     at modified_callback (C:\Users\MyUserFolder\node-opcua\packages\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:1005:28)
> QueryFirst is not supported by Server
> MessageBuilder : ERROR DETECTED IN event handler
> Error: Callback was already called.
>     at C:\Users\MyUserFolder\node-opcua\packages\node-opcua-samples\node_modules\async\dist\async.js:966:32
>     at C:\Users\MyUserFolder\node-opcua\packages\node-opcua-samples\node_modules\async\dist\async.js:3885:13
>     at C:\Users\MyUserFolder\node-opcua\packages\node-opcua-samples\bin\simple_client.js:672:13
>     at C:\Users\MyUserFolder\node-opcua\packages\node-opcua-client\src\client_session.js:1568:20
>     at C:\Users\MyUserFolder\node-opcua\packages\node-opcua-client\src\client_session.js:1168:20
>     at modified_callback (C:\Users\MyUserFolder\node-opcua\packages\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:1009:13)
>     at process_request_callback (C:\Users\MyUserFolder\node-opcua\packages\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:78:5)
>     at ClientSecureChannelLayer._on_message_received (C:\Users\MyUserFolder\node-opcua\packages\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:129:5)
>     at emitMany (events.js:147:13)
>     at MessageBuilder.emit (events.js:224:7)
> Error: Callback was already called.
>     at C:\Users\MyUserFolder\node-opcua\packages\node-opcua-samples\node_modules\async\dist\async.js:966:32
>     at C:\Users\MyUserFolder\node-opcua\packages\node-opcua-samples\node_modules\async\dist\async.js:3885:13
>     at C:\Users\MyUserFolder\node-opcua\packages\node-opcua-samples\bin\simple_client.js:672:13
>     at C:\Users\MyUserFolder\node-opcua\packages\node-opcua-client\src\client_session.js:1568:20
>     at C:\Users\MyUserFolder\node-opcua\packages\node-opcua-client\src\client_session.js:1168:20
>     at modified_callback (C:\Users\MyUserFolder\node-opcua\packages\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:1009:13)
>     at process_request_callback (C:\Users\MyUserFolder\node-opcua\packages\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:78:5)
>     at ClientSecureChannelLayer._on_message_received (C:\Users\MyUserFolder\node-opcua\packages\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:129:5)
>     at emitMany (events.js:147:13)
>     at MessageBuilder.emit (events.js:224:7)

The node-opcua server returns:

> UNSUPPORTED REQUEST !! QueryFirstRequest
>  messages  UNSUPPORTED REQUEST !! QueryFirstRequest
> 08:48:23:463 220 ServiceFault  status =  BadNotImplemented (0x80400000)

**Expected behavior**

A client-server connection without crashing?

**Minimal reproduction of the problem with instructions**

  1. Install node-opcua from source (as described in the readme.md)
  2. Run the demo server from source: 

> node packages/node-opcua-samples/bin/simple_server

The server seems to start correctly. Full server output on the terminal here: https://pastebin.com/y3ndYqMD

  3. Run the demo client from source: 
> node packages/node-opcua-samples/bin/simple_client.js -e "opc.tcp://MY-PC:26543"

The client seems to start correctly, until I get the error. Full client output on the terminal here: https://pastebin.com/UGjm1TeZ  
  
**What is the motivation / use case for changing the behavior?**

I want to use node-opcua in my project, but can't even get the basics to work.

**Please tell us about your environment:**

I reproduced the errors both from source and as a npm package, getting the exact same errors. The same error also appears when using the Siemens S7 OPC UA Server instead of the node-opcua sample server.
This issue is about the errors with node-opcua installed from source.

  * (x) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: PC
  * OS version: Windows 10
    
    - (x) Windows : version : 10 Build 17134
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: NodeOPCUA-SERVER
     - Version: 0.4.1 build 1234
     - Manufacturer/Software vendor: node-opcua
     - link : https://
     
    
* **node-opcua version:** 

Latest git version:  622803f

* **Node:** 
   `node --version` = v8.11.3

Chrissie, 2018-07-10:
Removing the code from commit 93da512f0f92afe613c9e9d7f00d4a443ca4defd seems to fix the main error. I am doing some further investigation at this point. Why was there an unused function in the code?

erossignon, 2018-07-10:
> ERROR !!! , please check here !!!! callback may be called twice !!  TypeError: opcua.readHistoryServerCapabilities is not a function

The error comes from a typo in readHistorySeverCapabilities ... let me update the title

closed 2018-07-10 by erossignon