node-opcua/node-opcua #503: Cannot read:C:\Users\siemens\AppData\Roaming\npm\node_modules\node-opcua-client\certificates\client_selfsigned_cert_1024.pem file

Dhiaaeddinanabtawi, 2018-07-12:
Hi,

I am new to Node-RED and right now I have installed it and everything works fine. After creating my first flow, every time I start node red, I get an error saying:

```
Cannot read:C:\Users\siemens\AppData\Roaming\npm\node_modules\node-opcua-client\certificates\client_selfsigned_cert_1024.pem file
cannot locate certificate file C:\Users\siemens\AppData\Roaming\npm\node_modules\node-opcua-client\certificates\client_selfsigned_cert_1024.pem
```
Well, I figured out that in my node-modules folder I don't have any folder called `node-opcua-client` at all. Is there any explanation of what this error actually mean and how to solve it?
I am using Node-red on a windows 7 virtual machine with the following:
Npm: 5.6.0
Node: v8.11.3
Node-RED: v0.18.7
I know there is a similar issue, but nothing helped me so far.

mikakaraila, 2018-07-12:
You have installed this with -g global, not under .node-red but check if you have it in local not roaming profile. Hope this helps...

Dhiaaeddinanabtawi, 2018-07-12:
Maybe it is a dum question, but when I search for it, I find them under the mentioned folder, but when I open that folder, I see nothing?
How to solve this problem?

closed 2019-08-03 by stale[bot]