node-opcua/node-opcua #504: Connection cannot be established from 3rd party embedded client to node-opcua-server

nospam2000, 2018-07-17:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

A 3rd party embedded device OPC UA client drops the TCP connection right after it receives the OPC UA ACK answer to its HEL request from node-opcua-server.
The ReceiveBufferSize and the SendBufferSize of the ACK response are larger than the values which are requested by the client (8196 vs. 8192 according to a Wireshark capture).

### The HEL request from the 3rd party client:
OpcUa Binary Protocol
    Message Type: HEL
    Chunk Type: F
    Message Size: 60
    Version: 0
    **ReceiveBufferSize: 8192
    SendBufferSize: 8192**
    MaxMessageSize: 2097152
    MaxChunkCount: 256
    EndPointUrl: opc.tcp://141.73.114.71:8080

### The Ack response from the node-opcua-server:
OpcUa Binary Protocol
    Message Type: ACK
    Chunk Type: F
    Message Size: 28
    Version: 0
    **ReceiveBufferSize: 8196
    SendBufferSize: 8196**
    MaxMessageSize: 2097152
    MaxChunkCount: 256

**Expected behavior**

The ReceiveBufferSize and the SendBufferSize of the response should be smaller than or equal to the values which are requested by the client (8192 instead of 8196).

**Minimal reproduction of the problem with instructions**

  1. use a client with a ReceiveBufferSize and SendBufferSize of 8192 byte
  2. connect to a node-opcua-server
  3. verify the values of ReceiveBufferSize and SendBufferSize in the ACK response which is sent to the client. They have to be less than or equal to the values requested by the client.

  
**What is the motivation / use case for changing the behavior?**

I wish to be able to connect from the 3rd party client to the node-opcua-server.

**Please tell us about your environment:**

The server operating system is Windows 10 build 1803
"node-opcua-server"   version "0.4.1" (from npmjs.org)

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - (x ) Windows : version : Windows 10 build 1803
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://

* **node-opcua version:** ``

    "node-opcua": {
      "version": "0.0.65",
      "resolved": "https://registry.npmjs.org/node-opcua/-/node-opcua-0.0.65.tgz",
      "integrity": "sha512-ITLJ2dnmOjerS53SPR2O4jgF4WA9WItEKbCL9Xd02y/SvgmZE8aPGCtcTS6D8ynLJi9FCzgutgESG7WR9RdHbg==",

    "node-opcua-server": {
      "version": "0.4.1",
      "resolved": "https://registry.npmjs.org/node-opcua-server/-/node-opcua-server-0.4.1.tgz",
      "integrity": "sha512-tN+AWDeJ71AzWdqJK8JuHiAaz3eg5RE6aOfAtsrR2EP1fwIXxWsjtX+YRf9PPvBiUSubNYc+BrNCDOn1oRwHcA==",

* **Node:** 
   `node --version` = v8.11.3

nospam2000, 2018-07-17:
In the code there is a comment, that the smallest message/chunk size should be 8196.
In former version of the OPC UA specification there was a bug because it state 8196, but now it has been corrected to either 8192 for secure connections and no limit for unsecured communication.

See errata _"Definition of min send and receive buffer size wrong"_ in the errata 1.03.3 for OPCUA Part 6, chapters 6.7.2 and 7.1.2, Mantis Reference 0003447.

### Problem Statement
_The specification specifies a minimum message size of 8196 for UA Secure Conversation and states that the message size is greater than 8192 for UA TCP. The layers are independent so different values are not wrong but they are confusing._

### Solution
_Remove the limits from the OPC UA TCP description.
Change the limit to >= 8192 for OPC UA Secure Conversation._

In the OPC UA Specification Part 6, Release 1.04 from November 22, 2017 it now states a minimum size of 8192 for Secure Conversation.
6.7 OPC UA Secure Conversation
6.7.1 Overview
OPC UA Secure Conversation (UASC) allows secure communication using binary encoded
Messages.
UASC is designed to operate with different TransportProtocols that may have limited buffer
sizes. For this reason, OPC UA Secure Conversation will break OPC UA Messages into several
pieces (called ‘MessageChunks’) that are smaller than the buffer size allowed by the
TransportProtocol. **UASC requires a TransportProtocol buffer size that is at least 8 192 bytes.**

And in chapter 7.1.2.3 Hello Message and 7.1.2.4 Acknowledgement Message i**s now says nothing about a minimum size**
### From  7.1.2.4 Acknowledgement Message
**ReceiveBufferSize**: The largest MessageChunk that the sender can receive.
This value shall not be larger than what the Client requested in the Hello Message
**SendBufferSize**: The largest MessageChunk that the sender will send.
This value shall not be larger than what the Client requested in the Hello Message

nospam2000, 2018-07-17:
in the code the biggest problem is in node-opcua-transport\src\server_tcp_transport.js :
```
ServerTCP_transport.prototype._send_ACK_response = function (helloMessage) {
[...]
    self.receiveBufferSize = clamp_value(helloMessage.receiveBufferSize, 8192, 512 * 1024);
    self.sendBufferSize    = clamp_value(helloMessage.sendBufferSize,    8192, 512 * 1024);
```
Here I replaced 8196 by 8192. Actually there should be no limit for non-secure communication.

There are some other places which use the value 8196 for minimum sizes in the code, they should also be adapted.

erossignon, 2018-10-22:
> Actually there should be no lower limit for non-secure communication. The server must never increase the value it gets from the client.

In fact 8192 bytes is the smallest buffer size allowed, I believe that the communication shall be droppedb by the server (error) if the client suggests a buffer size lesser than 8192, as the value shall not be increased.

nospam2000, 2018-10-22:
@erossignon "In fact 8192 bytes is the smallest buffer size allowed":
Can you please tell me where you found this information in OPC UA release **1.04** for **non-secured** communication?

For secured communication I agree.

erossignon, 2018-10-23:
the 8192 min buffer size appears in a paragraph that relates to the SecureConnection.

[image]

However: 
* SecureConnection is used for all type of type of Binary Connection
* there must be a minimum Buffer size regardless of whether the connection is secure or not. 
* Buffer size negotiation takes place at the transport level (Hello/Acknowledge transaction). But Secure Strategy takes place later during the OpenSecureChannel , once the transport layer and buffer size are already negociated.
* the minimum Buffer size must accommodate a secure or a unsecure connection. Secure  connection need space for signature and possible inflation of encrypted message, It must also take into account message headers. Therefore the value of minimum buffer size is imposed by secure connection but applies also to unsecure connection.

Investigating other opensource solution demonstrate that the min buffer size is applied as well

https://github.com/OPCFoundation/UA-AnsiC/blob/8c251e8fc5ed137bdb6d9fee4b03ccb6967628f8/Stack/transport/tcp/opcua_tcpconnection.c#L372
https://github.com/OPCFoundation/UA-AnsiC/blob/8c251e8fc5ed137bdb6d9fee4b03ccb6967628f8/Stack/transport/tcp/opcua_tcplistener.c#L1088

https://github.com/OPCFoundation/UA-.NETStandard/blob/0d08d4018012bf0fb5ee5e535a0841205fcad370/Stack/Opc.Ua.Core/Stack/Tcp/TcpMessageType.cs#L153

https://github.com/OPCFoundation/UA-.NETStandard/blob/0d08d4018012bf0fb5ee5e535a0841205fcad370/Stack/Opc.Ua.Core/Stack/Tcp/UaSCBinaryChannel.cs#L106

https://github.com/OPCFoundation/UA-.NETStandard/tree/master/Stack/Opc.Ua.Core/Stack/Tcp

Therefore it leads me to think that the  8192 min buff size applies to all type of binary connection.
This being said, I would be happy to be proven wrong,  as it is important to get it as per the specification.

nospam2000, 2018-10-23:
I will try to catch a specialist in the next days.

nospam2000, 2018-10-24:
> Therefore it leads me to think that the 8192 min buff size applies to all type of binary connection.

@erossignon, yes in the code you referenced from [opcua_tcplistener.c](https://github.com/OPCFoundation/UA-AnsiC/blob/8c251e8fc5ed137bdb6d9fee4b03ccb6967628f8/Stack/transport/tcp/opcua_tcplistener.c#L1088) is pretty clear that the limit is **always** applied :

```
  /* This value shall be greater or equal than 8 192 bytes (see 1.03 Errata) */
    if(pConnection->SendBufferSize < 8192 || pConnection->ReceiveBufferSize < 8192)
    {
        uStatus = OpcUa_TcpListener_CloseConnection(a_pListener,
            pConnection, OpcUa_BadConnectionRejected);
        OpcUa_ReturnStatusCode;
    }
```
and some lines above they do the clipping for an **upper** limit:
```
    pConnection->SendBufferSize     = (pConnection->SendBufferSize    > (OpcUa_UInt32)OpcUa_ProxyStub_g_Configuration.iTcpListener_DefaultChunkSize)?(OpcUa_UInt32)OpcUa_ProxyStub_g_Configuration.iTcpListener_DefaultChunkSize:pConnection->SendBufferSize;
    pConnection->ReceiveBufferSize  = (pConnection->ReceiveBufferSize > (OpcUa_UInt32)OpcUa_ProxyStub_g_Configuration.iTcpListener_DefaultChunkSize)?(OpcUa_UInt32)OpcUa_ProxyStub_g_Configuration.iTcpListener_DefaultChunkSize:pConnection->ReceiveBufferSize;
```
 
> What is your analysis 

I came to the same conclusion.

  Michael

coconut147, 2018-10-25:
_Therefore it leads me to think that the 8192 min buff size applies to all type of binary connection._

Yes - you are right. There is no difference in nonseure and secure communications. The HEL / ACK pair is NEVER encrypted when using TCP transport stream, only the secure channel "decides" if there is an encryption or not.

Please also note that the "HEL" message defined in Part 6 was changed in Specification 1.04 - the minimum Buffer size has been removed completely:

[image]

erossignon, 2018-12-04:
fixed in 0.5.0

closed 2018-12-04 by erossignon