node-opcua/node-opcua #505: Multiple Subscription to Multiple Items Fails After Reconnecting to OPCServer

barisvelioglu, 2018-07-18:
- (x) bug report 
 - ( ) feature request
 - ( ) support request

**Current behavior**
If I use multiple subscription for multiple monitoredItem, it starts listening without any problem but when I close the OPCServer it crashes.

```
ns=2;s=AdvancedTags.OPC1_Weld_v362_PLC12.MIS_PLC12.6X.6x60.ST_Alarm1 361
ns=2;s=AdvancedTags.OPC1_Weld_v362_PLC12.MIS_PLC12.6X.6x60.ST_Alarm2 361
backoff  attempt # 0  retrying in  1.04  seconds
backoff  attempt # 1  retrying in  2.049  seconds
backoff  attempt # 2  retrying in  4.219  seconds
 !!!!!!!!!!!!!!!!!!!!!!!!  CONNECTION RE-ESTABLISHED !!!!!!!!!!!!!!!!!!!
-----------------------------------------------------------
Error
    at assert (C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-assert\dist\index.js:6:21)
    at ClientSidePublishEngine.getSubscription (C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-client\src\client_publish_engine.js:337:5)
    at recreate_subscription (C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-client\src\reconnection.js:199:69)
    at C:\Users\02484376\Desktop\ipenew\node_modules\async\dist\async.js:1135:9
    at eachOfArrayLike (C:\Users\02484376\Desktop\ipenew\node_modules\async\dist\async.js:1069:9)
    at eachOf (C:\Users\02484376\Desktop\ipenew\node_modules\async\dist\async.js:1117:5)
    at _asyncMap (C:\Users\02484376\Desktop\ipenew\node_modules\async\dist\async.js:1133:5)
    at Object.map (C:\Users\02484376\Desktop\ipenew\node_modules\async\dist\async.js:1122:16)
    at C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-client\src\reconnection.js:197:23
    at C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-client\src\client_session.js:866:9
-----------------------------------------------------------
ERROR !!! , please check here !!!! callback may be called twice !!  Error
    at assert (C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-assert\dist\index.js:6:21)
    at ClientSidePublishEngine.getSubscription (C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-client\src\client_publish_engine.js:337:5)
    at recreate_subscription (C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-client\src\reconnection.js:199:69)
    at C:\Users\02484376\Desktop\ipenew\node_modules\async\dist\async.js:1135:9
    at eachOfArrayLike (C:\Users\02484376\Desktop\ipenew\node_modules\async\dist\async.js:1069:9)
    at eachOf (C:\Users\02484376\Desktop\ipenew\node_modules\async\dist\async.js:1117:5)
    at _asyncMap (C:\Users\02484376\Desktop\ipenew\node_modules\async\dist\async.js:1133:5)
    at Object.map (C:\Users\02484376\Desktop\ipenew\node_modules\async\dist\async.js:1122:16)
    at C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-client\src\reconnection.js:197:23
    at C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-client\src\client_session.js:866:9
ERROR !!! , please check here !!!! callback may be called twice !!  TypeError: Cannot read property 'createMonitoredItems' of null
    at C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-client\src\client_subscription.js:714:34
    at C:\Users\02484376\Desktop\ipenew\node_modules\async\dist\async.js:3880:24
    at replenish (C:\Users\02484376\Desktop\ipenew\node_modules\async\dist\async.js:1011:17)
    at iterateeCallback (C:\Users\02484376\Desktop\ipenew\node_modules\async\dist\async.js:995:17)
    at C:\Users\02484376\Desktop\ipenew\node_modules\async\dist\async.js:969:16
    at C:\Users\02484376\Desktop\ipenew\node_modules\async\dist\async.js:3885:13
    at C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-client\src\client_subscription.js:172:17
    at C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-client\src\client_session.js:906:9
    at C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-client\src\client_session.js:1165:9
    at modified_callback (C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:1005:28)
MessageBuilder : ERROR DETECTED IN event handler
Error: Callback was already called.
    at C:\Users\02484376\Desktop\ipenew\node_modules\async\dist\async.js:966:32
    at C:\Users\02484376\Desktop\ipenew\node_modules\async\dist\async.js:3885:13
    at C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-client\src\client_subscription.js:149:24
    at C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-client\src\client_session.js:903:20
    at C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-client\src\client_session.js:1156:20
    at modified_callback (C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:1009:13)
    at process_request_callback (C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:78:5)
    at ClientSecureChannelLayer._on_message_received (C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:129:5)
    at emitMany (events.js:147:13)
    at MessageBuilder.emit (events.js:224:7)
Error: Callback was already called.
    at C:\Users\02484376\Desktop\ipenew\node_modules\async\dist\async.js:966:32
    at C:\Users\02484376\Desktop\ipenew\node_modules\async\dist\async.js:3885:13
    at C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-client\src\client_subscription.js:149:24
    at C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-client\src\client_session.js:903:20
    at C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-client\src\client_session.js:1156:20
    at modified_callback (C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:1009:13)
    at process_request_callback (C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:78:5)
    at ClientSecureChannelLayer._on_message_received (C:\Users\02484376\Desktop\ipenew\node_modules\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:129:5)
[... 160 lines cut ...]
  
       - ( ) node-red
       - (x) https://www.kepware.com/en-us/products/kepserverex/
        
  * Device: Dell Notebook
  * OS version: Windows
    
    - (x) Windows : version : 10
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:** ``

* **Node:** 
   `node --version` = 8.9.4

anthrogan, 2018-07-23:
Hi @barisvelioglu 

The library does not appear to hold the session on loss of connection which kind of makes sense. So on reconnect it simply connects then does nothing, you need to recreate the session so the subscription can start again: 

```
client.on("connection_reestablished", function () {
      console.log(" !!!!!!!!!!!!!!!!!!!!!!!!  CONNECTION RE-ESTABLISHED - RECREATE SESH !!!!!!!!!!!!!!!!!!!".bgWhite.red);
      client.createSession(function (err, session) {
        if (!err) {
          the_session = session
        }
        testMonitor("ns=2;s=AdvancedTags.OPC1_Weld_v362_PLC12.MIS_PLC12.6X.6x60.ST_Alarm2");
      })
    });
```

I'll be honest I don't fully understand the inner workings on node-upcua so i could be wrong here but, I belive `CreateSession `is only used as options for the `ClientSubscription`.
`constructor(session: ClientSession, options: ClientSubscriptionOptions)`

 If you want to be entirely sure you could use `closeSession` first before creating one on reconnect?

I'm not sure what this does on OPC side though, we use Kepware and it doesn't appear to cause issue (e.g. the client count does not go up on reconnect)

Also, good luck with monitoring 200k tags!

barisvelioglu, 2018-07-23:
Hi @anthrogan,

I didnt get it because when I use only one subscription, it works with the same session without recreating it. When I use multi subscription, then it doesnt work.

anthrogan, 2018-07-23:
My only assumption is the subscription to the session is in the connection function in your working code, so it gets called as it connects/reconnects. 

I know its vague, but we've hit this issue and we got round it by adding the session to the reconnection so thought I'd share 😄 ! The node opc library itself is massive and if you're not familiar with typescript its difficult to read. I lost a day finding out where dataValue came from 😕

@erossignon might be able to give you a more definitive answer!

barisvelioglu, 2018-07-23:
Hahaha 👍 Thanks for the answer.

You said you also use KepwareServer. Whats your PublishingInterval and how many tags u are subscribing ? We have some problems that some tags doesnt send data after a while (a few hours later and more) After we restarting the app, we are getting again.

anthrogan, 2018-07-24:
OPC specifically, we sub to about 220 tags, we have tested up to 2000 but we've worked hard to make sure the comms are efficient as we don't want the apps to crash so we put lots of work PLC side where possible. We have the sample interval currently at 100, but will likely go to 1000. In the PoC stage we had it set to 10 and it had no issues. Our OPC server interval is 1000 anyway.
We also communicate with Siemens PLCs but direct (Using S7Client) so in total only about 500 tags send us data to our front end application. (Front end app is a VueJS PWA, we deliver data over MQTT/ws to it, so basically we see PLC changes in real time on any client. For free!)

I've not seen kepware have any issues like that tbh, the tags are super stable. Infact its works better than I expected, we check for the status code of the tag and deal with it like so:
```
// Check tag status is ok
      if (dataValue.statusCode.name === 'Good') {
        value = dataValue.value.value;
      } else {
        console.warn(`${emoji.get('no_entry')} [Error] - [OPC] - Status Code ${dataValue.statusCode.name} for ${tagName.facility} ${tagName.tag}`.red);
        value = -1;
        // -1 status code demonstrates error state
      }
```
this stops the errors overflowing and eventually rolling the application (We run the services with pm2) but the above code works great, we see maintenance stops etc the tags drop out but reconnect again fine.

We run KepServer 5.19 alongside WinCC 7.4 (Which also communicates to it) 

I'd recommend checking the statusCode name for the tag state, it will give you a idea of where the issue is. If the error is with Kepware it will likley show you a error message in its diagnostic window at the bottom.

What are your PLCs? I've had experience with Siemens and Omron, the finns comms are really difficult tbh and its lots easier to communicate over OPC, were as siemens is easier to go direct access. You just have to write your own compare logic so you're not just sending data as you poll!

opcua-user, 2019-05-16:
I think there is an issue in file client_publish_engine.js in node-opcua-client. 

The second parameter of map is index and the second parameter of parseInt is the base of the number. That do not work for more than one value in self.subscriptionMap.

The code 
``` javascript
ClientSidePublishEngine.prototype.getSubscriptionIds = function() {
    const self = this;
    return Object.keys(self.subscriptionMap).map(parseInt);
};
```
should be replaced by the following code.
``` javascript
ClientSidePublishEngine.prototype.getSubscriptionIds = function() {
    const self = this;
    return Object.keys(self.subscriptionMap).map((id)=>{
          return parseInt(id);
    });
};
```

erossignon, 2019-11-14:
0da9d0742b9707202cdfa7d76e07f20e00a216ce addressed the issue.

closed 2019-11-14 by erossignon