node-opcua/node-opcua #506: UA-Client can't connect with client-cert authentication.

nnnntosh, 2018-07-19:
**I'm submitting a ...**  (check one with "x")

 - ( ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - (x) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

I'm trying to connect to OPC Foundation Sample-Server using client-cert (X509) authentication, but I get an error.
The server returns BadUserAccessDenied.
The server is set to allow only certificates.

Does node-opcua-client currently support client-cert (X509) authentication?

**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
      
  * Device: raspberry-pi
  * OS version: Raspbian GNU/Linux 8 (jessie)
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: OPC Foundation Sample-Server 
     
    
* **node-opcua version: 0.4.1

* **Node:** 
   `node --version` = v8.11.3

nnnntosh, 2018-07-20:
I read the client side source code.
AnonymousIdentityToken class and UserNameIdentityToken class are used, but X509IdentityToken class was not used.
Is client-cert not supported?

nnnntosh, 2018-07-20:
I'm sorry.
There was a statement that it is unsupported on the top page.

erossignon, 2018-07-30:
Yes, client-cert (X509) authentication is not supported at this time... any volunteers ?

jared-DM, 2018-08-09:
This is unfortunate that this isn't supported when implementing clients using node-opua.  What do you think the level of effort would be for this work?

closed 2019-12-01 by stale[bot]