node-opcua/node-opcua #512: Cannot connect to sample_server with Prosys OPC UA Client

StephanLeschke, 2018-08-26:
**I'm submitting a ...**  (check one with "x")

 - (X ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**
Cannot connect to sample_server with Prosys OPC UA Client.
Message is: Could not connect to the server: Error=null
Problem only occurs with node-opcua@0.4.0 and above.
With node-opcua@0.3.0 connection is possible. 

**Expected behavior**
Connection should work with latest version of node-opcua

**Minimal reproduction of the problem with instructions**
  1. Install node-opcua@v0.4.2 or v0.4.1 or v0.4.0
  2. Run the following code:

`var opcua = require("node-opcua");

var server = new opcua.OPCUAServer({
    port: 4334,
});

server.start(function() {
    console.log("Server is now listening ... ( press CTRL+C to stop)");
});`

  3. Try to connect with Prosys OPC UA Client v3.1.4-293 running on Linux
  4. Alternatively try to connect with UaExpert.

**What is the motivation / use case for changing the behavior?**
I wanted to follow the 'Simple Server' tutorial and failed to connect with any client I could get hold off, except the 'Sample Client'.
Strangely enough I was able to connect to the sample_server used in the HTML dashboard tutorial.
It took me some time to figure out that this example was using a very old version of node-opcua (v0.0.65) 

**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * (X ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: Notebook PC
  * OS version: 
    
    - ( X) Windows : version : 7
    - ( X) Linux   : version : Mint kernel 4.15.0-32
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: Prosys OPC UA Client
     - Version: v3.1.4-293
     - Manufacturer/Software vendor: Prosys OPC Ltd.
     - link : https://www.prosysopc.com
     
    
* **node-opcua version:** ``
v0.4.2
* **Node:** 
   `node --version` = v8.11.4

erossignon, 2018-08-29:
Testing today with Prosys OPCUA CLient v3.1.4.293 and node-opcua@0.4.4 on Windows : 
[image]

StephanLeschke, 2018-08-29:
Retested on my side with node-opcua@0.4.4 on Windows.
Connection was possible with Prosys OPC UA Client and with UaExpert.
Many thanks.

closed 2018-08-29 by StephanLeschke