node-opcua/node-opcua #514: node-opcua-sampleserver app throws "Invalid namespace index ( out of bound)" error with version 0.4.2, 0.4.3

jared-DM, 2018-08-27:
**I'm submitting a ...**  (check one with "x")

 - ( X) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**
The node-opcua-sampleserver app throws "Invalid namespace index ( out of bound)" error with version 0.4.2, 0.4.3:

**Console debug log output:**

```
initialized
Warning !  AddressSpace#addFolder  is now deprecated
          use addressSpace.getOwnNamespace().addFolder(..) instead
Warning !  AddressSpace#addVariable  is now deprecated
          use addressSpace.getOwnNamespace().addVariable(..) instead
-----------------------------------------------------------
Error: invalid namespace index ( out of bound)
    at assert (/mnt/c/Users/testuser/build/temp/node-opcua-sampleserver/node_modules/node-opcua-assert/dist/index.js:6:21)
    at AddressSpace.getNamespace (/mnt/c/Users/testuser/build/temp/node-opcua-sampleserver/node_modules/node-opcua-address-space/src/address_space.js:116:9)
    at AddressSpace._resolveRequestedNamespace (/mnt/c/Users/testuser/build/temp/node-opcua-sampleserver/node_modules/node-opcua-address-space/src/address_space.js:964:17)
    at AddressSpace.addVariable (/mnt/c/Users/testuser/build/temp/node-opcua-sampleserver/node_modules/node-opcua-address-space/src/address_space.js:974:17)
    at AddressSpace.Constructor.(anonymous function) [as addVariable] (/mnt/c/Users/testuser/build/temp/node-opcua-sampleserver/node_modules/node-opcua-utils/dist/set_deprecated.js:15:27)
    at construct_my_address_space (/mnt/c/Users/testuser/build/temp/node-opcua-sampleserver/server.js:50:41)
    at /mnt/c/Users/testuser/build/temp/node-opcua-sampleserver/server.js:24:5
    at Immediate._onImmediate (/mnt/c/Users/testuser/build/temp/node-opcua-sampleserver/node_modules/node-opcua-server/src/opcua_server.js:625:9)
    at runCallback (timers.js:693:18)
    at tryOnImmediate (timers.js:664:5)
-----------------------------------------------------------
/mnt/c/Users/testuser/build/temp/node-opcua-sampleserver/node_modules/node-opcua-assert/dist/index.js:10
        throw err;
        ^

Error: invalid namespace index ( out of bound)
    at assert (/mnt/c/Users/testuser/build/temp/node-opcua-sampleserver/node_modules/node-opcua-assert/dist/index.js:6:21)
    at AddressSpace.getNamespace (/mnt/c/Users/testuser/build/temp/node-opcua-sampleserver/node_modules/node-opcua-address-space/src/address_space.js:116:9)
    at AddressSpace._resolveRequestedNamespace (/mnt/c/Users/testuser/build/temp/node-opcua-sampleserver/node_modules/node-opcua-address-space/src/address_space.js:964:17)
    at AddressSpace.addVariable (/mnt/c/Users/testuser/build/temp/node-opcua-sampleserver/node_modules/node-opcua-address-space/src/address_space.js:974:17)
    at AddressSpace.Constructor.(anonymous function) [as addVariable] (/mnt/c/Users/testuser/build/temp/node-opcua-sampleserver/node_modules/node-opcua-utils/dist/set_deprecated.js:15:27)
    at construct_my_address_space (/mnt/c/Users/testuser/build/temp/node-opcua-sampleserver/server.js:50:41)
    at /mnt/c/Users/testuser/build/temp/node-opcua-sampleserver/server.js:24:5
    at Immediate._onImmediate (/mnt/c/Users/testuser/build/temp/node-opcua-sampleserver/node_modules/node-opcua-server/src/opcua_server.js:625:9)
    at runCallback (timers.js:693:18)
    at tryOnImmediate (timers.js:664:5)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-opcua-sampleserver@0.0.56 start: `node server.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-opcua-sampleserver@0.0.56 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/testuser/.npm/_logs/2018-08-27T13_04_26_217Z-debug.log
```

**Minimal reproduction of the problem with instructions**
Update the package.json file for the node-opcua-sampleserver project:

  "dependencies": {
    "node-opcua": "~0.4.3"
  },
 
**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * (X ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
    
* **node-opcua version:** ``
0.4.2, 0.4.3

erossignon, 2018-08-28:
Please provide the code snippet that is used for creating the variable. I suspect that the server is trying to create a variable/object in a namespace that has not been created.

jared-DM, 2018-08-28:
This can be duplicated using the node-opcua-sampleserver sample code:

https://github.com/node-opcua/node-opcua-sampleserver/blob/master/server.js

erossignon, 2018-08-29:
Yes, doc is not up to date ... we shall fix it .

jared-DM, 2018-08-29:
Thanks for the clarification.

closed 2018-12-04 by erossignon