node-opcua/node-opcua #515: UaExpert tool can't connect to server app

jared-DM, 2018-08-27:
**I'm submitting a ...**  (check one with "x")

 - (X ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**
Since updating to 0.4.2 or 0.4.3 the UaExpert client tool is no longer able to connect to our server application.  The UaExport app shows "Could not connect to server: "BadTimeout"

With DEBUB set to "ALL" I get the following console output (with server name changed):
```

server_end_point              :241   OPCUAServerEndPoint#_on_client_connection true
server_end_point              :130   MAX CONNECTIONS =  11
server_end_point              :149   server connected  with : ::ffff:127.0.0.1:56942
server_end_point              :128   CONCURRENT CONNECTION =  1
server_end_point              :257    nbConnections  0  self._server.maxConnections 11 10
server_end_point              :264   OPCUAServerEndPoint._on_client_connection successful => New Channel
server_tcp_transport          :160   SERVER received HEL
server_tcp_transport          :161   SERVER received 00000000: 48 45 4c 46 3b 00 00 00 00 00 00 00 00 00 01 00 00 00 01 00 00 00 00 01 88 13 00 00 1b 00 00 00    HELF;...........................
00000020: 6f 70 63 2e 74 63 44 3a 44 44 44 44 44 44 44 44 44 44 44 44 44 44 3a 34 38 34 30                   opc.tcp://servername:4840

server_tcp_transport          :122   server send: ACK
server_tcp_transport          :123   server send: 00000000: 41 43 4b 46 1c 00 00 00 00 00 00 00 00 00 01 00 00 00 01 00 00 00 00 01 88 13 00 00                ACKF........................

server_tcp_transport          :124   acknowledgeMessage= AcknowledgeMessage {
  protocolVersion: 0,
  receiveBufferSize: 65536,
  sendBufferSize: 65536,
  maxMessageSize: 16777216,
  maxChunkCount: 5000 }
server_secure_channel_layer   :369   ServerSecureChannelLayer : Transport layer has been initialized
server_secure_channel_layer   :370   ... now waiting for OpenSecureChannelRequest...
message_builder               :390    Sequence Header SequenceHeader { sequenceNumber: 51, requestId: 1 }
message_builder               :119    Sequence Number =  51
message_builder               :450   message size = 133  body size = 54
server_secure_channel_layer   :446   { /*SequenceHeader*/
 sequenceNumber                /* UInt32                           */: 51               0x33
 requestId                     /* UInt32                           */: 1               0x1
};
server_secure_channel_layer   :447   { /*AsymmetricAlgorithmSecurityHeader*/
 securityPolicyUri             /* String                           */: http://opcfoundation.org/UA/SecurityPolicy#None
 senderCertificate             /* ByteString                       */: null
 receiverCertificateThumbprint /* ByteString                       */: null
};
server_secure_channel_layer   :289   SecurityToken 1
server_end_point              :277   Channel#init done null
server_end_point              :281   server receiving a client connection
server_end_point              :629   _registerChannel =  channel.hashKey =  1
```

**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * (X ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: ubuntu
  * OS version: 18.0.1
    
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_UxExpert

erossignon, 2018-08-28:
We found this issue as well, it should be fixed in node-opcua@0.4.4 => can you confirmed ?

jared-DM, 2018-08-28:
Yes, this is fixed with version 0.4.4. Thanks!!!

closed 2018-08-29 by erossignon