node-opcua/node-opcua #518: Issues with the API documentation

SystemParadox, 2018-09-04:
Hi. This module looks really helpful, but despite having a reasonable background knowledge of OPC I am finding it very difficult to use. At the moment it feels completely opaque - it's really hard to find out how to do even the simplest task. I think a lot of this is because of issues with the API documentation:

1. There are quite a few broken links. The generator seems to have included the first paragraph as part of the title for some pages, which creates a huge broken URL. See BaseNode, ClientSidePublishEngine, MandatoryChildOrRequestedOptionalFilter, OPCUAServerEndPoint, ServerSidePublishEngine, UACertificateExpirationAlarm, UADiscreteAlarm, UATripAlarm.
2. The return type for most methods is just "Promise", which is really unhelpful. A promise for what?
3. The search only searches for prefixes. If I search for "client", surely I should get OPCUAClient in the results?
4. There is no way to search for methods or properties, only classes.
5. It would be really helpful if each class started with an example of how you would usually get an instance of it, since the vast majority of the classes are not meant to be instantiated directly.

Thanks.

spike0xff, 2018-09-15:
well that's pretty funny, because I thought my frustration with (trying to use) this module was maybe due to my lack of knowledge of OPC-UA.

SystemParadox, 2018-09-16:
OPC-UA is just a protocol so in my opinion you should be able to understand how to use this library just from its own documentation, without having to refer to the OPC-UA spec.

As it stands, lack of intimate knowledge of OPC-UA is also a serious barrier to using this module. There are fundamental concepts which are just not explained, or reply upon prior knowledge of the predefined OPC-UA constants. For example, a folder is just a node with a type of FolderType, but you wouldn't know that from reading this documentation.

FolderType is just a particular nodeId as defined by the specification, but I can't even find it right now. I shall also take this moment to rant at the OPC foundation for not making the specification easier to download!!

To be honest, this looks pretty much like a direct port of a C++ or Java API. If you're familiar with an API for a different language you might be ok, but otherwise it's really hard to figure out.

erossignon, 2018-10-12:
there will be an overhaul in the way documentation is generated in the next node-opcua release (see @next branch)...

sanderd17, 2018-11-11:
IMO, the examples are even more important, and currently lacking a lot of clarity.

The example still uses callbacks with the async module. While most of the library now supports promises and thus native async functionality. 

Also, the examples give no explanation on what the different config options mean. F.e. what is `requestedLifetimeCount`? It's used in the example, so I expect that it's a decision I should make (based on the server I connect to, or the behaviour I want on the client). But no explanation is given, and the docs just say it's a number.

The same is true for the "changed" handler. It expects a function that gets a `dataValue`. The docs just say it's an `Object`. And in the example, apparently I can get the value by calling `dataValue.value.value`. There's probably a design reason why the value isn't given in the first level of the object (or even directly as function argument). But with the current situation, I have no idea what else I can get from the `dataValue` (f.e. the timestamps or quality) or why I need to get the value of it two levels deep.
```js
monitoredItem.on("changed",function(dataValue){
      console.log(" % free mem = ",dataValue.value.value);
});
```

SystemParadox, 2019-11-11:
This is still in need of attention. Please do not close.

erossignon, 2020-01-30:
> IMO, the examples are even more important, and currently lacking a lot of clarity.

I am glad to announce "Node-OPCUA by example" book is available now.
please visit https://leanpub.com/node-opcuabyexample

closed 2020-01-30 by erossignon