node-opcua/node-opcua #519: Nodecrawler is not browsing some nodes

rrdlpl, 2018-09-04:
**I'm submitting a bug**  (check one with "x")
 - ( x) bug report => search github for a similar issue or PR before submitting

**Current behavior**
I'm trying to browse a node with the NodeCrawler, but it doesn't return the childNodes. OPCUA - Expert does.

**Expected behavior**
Should return the nodes.

**Minimal reproduction of the problem with instructions**
This is how my code looks like >

```

...
   this.crawl(session, 'ns=6;s=::Program:mpOeeCoreUIConnect');
...
    public crawl(session: opcua.ClientSession, nodeId: string | undefined | null) {
        const crawler = new NodeCrawler(session);

        console.log('Now crawling object folder ...please wait...');
        crawler.read(nodeId, (err, obj) => {
            if (!err) {
                // console.log(obj);
                treeify.asLines(obj, true, true, (line) => {
                    console.log(line);
                });
                return;
            }
            console.log(err);
        });
    }
```

The output:
```

now crawling object folder ...please wait...
├─ browseName: mpOeeCoreUIConnect
├─ nodeId: ns=6;s=::Program:mpOeeCoreUIConnect
├─ nodeClass: Variable
├─ dataType: ns=0;i=22
├─ dataTypeName: undefined
└─ dataValue: [object Object]
``` 

What I'm expecting to get:

/home/r/Pictures/Screenshot from 2018-09-04 12-55-10.png

**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: X20CP0484
  * OS version: Debian stretch
    
    - ( ) Windows : version : _________
    - (x ) Linux   : version : Debian stretch
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: X20CP0484
     - Manufacturer/Software vendor: B und R Automation
     - link : https://www.br-automation.com/product/X20CP0484
     
    
* **node-opcua version:** ``
0.4.5

* **Node:** 
   `node --version` = 8.11.3

erossignon, 2018-12-03:
This is probably related to current lack of support for continuation points

Nandish-Reddy, 2019-05-30:
Hi Etienne,

Thanks for all the efforts you have put on node-opcua. Is there  a plan to support continuation points.
I am new to node  and hence finding alternatives for browsing all the nodes recursively.

Error: NodeCrawler doesn't support continuation point yet
    at assert (/home/Nandish/node_opcua_client/node_modules/node-opcua-assert/dist/index.js:6:21)
    at NodeCrawler._process_single_browseResult (/home/Nandish/node_opcua_client/node_modules/node-opcua-client-crawler/src/node_crawler.js:412:5)
    at NodeCrawler._process_browse_response (/home/Nandish/node_opcua_client/node_modules/node-opcua-client-crawler/src/node_crawler.js:524:14)
    at Immediate._onImmediate (/home/Nandish/node_opcua_client/node_modules/node-opcua-client-crawler/src/node_crawler.js:80:23)
    at runCallback (timers.js:705:18)
    at tryOnImmediate (timers.js:676:5)
    at processImmediate (timers.js:658:5)

erossignon, 2019-05-30:
Support for continuation point is available in new version @2.0.0 which has been released today.

Nandish-Reddy, 2019-05-31:
Thanks, I have a OPCUA server running in my PLC. Is it browsing memory /output region of the project created in PLC recursively? I don't see the nodeid's of these regions.

Nandish-Reddy, 2019-08-05:
Hi Etienne,

Is there any limit on the array data type that this crawler api can read?

I am actually facing issue in reading array data types with array size equal to 100 , size equal to 1000  . Only those sizes. while i browse a node with arr[0..99 ] or [0..999] these elements will have datatype set to undefined. 
Any help / suggestion here is greatly appreciated.

Thanks a lot,
Nandish

Nandish-Reddy, 2019-08-07:
An interesting observation,

while i browse a node arr[0..199] using crawler, it browses from 100th element, 101,102 ...199.

If i use opcua commander too, it browses only till 99. i.e. from 0,1,2 .. 99.

If i use UAExpert client it browses everything from 0.. 199.

erossignon, 2019-08-23:
duplicates #655

closed 2019-08-23 by erossignon