node-opcua/node-opcua #521: TypeError: server.engine.createFolder is not a function

Zaniyar, 2018-09-14:
( x ) bug report => search github for a similar issue or PR before submitting
  * ( x ) I have installed node-opcua from source ( using git clone)
  * ( x ) I have installed node-opcua as a package ( using npm install )
  - ( x ) Windows
 - ( x ) MacOs 
* **Node:** 
   `node --version` = 8.11.4

Error: 
`TypeError: server.engine.createFolder is not a function`

Code: 
 `server.engine.createFolder("RootFolder", {browseName: "Device"});`

I see a lot of examples of creating folders, but couldn't make it run. The createFolder() function is missing; maybe there is now a new way to create folders?

erossignon, 2018-09-14:
try ```addFolder``` instead , former createFolder has been renamed ```addFolder``` to be consistent with  other methods such as `addObject`, `addVariable` etc .. a very long time ago.

closed 2018-10-10 by erossignon

hunter7793, 2019-06-20:
i have a problem described above, so i tryied to using "addFolder" instead "createFolder", but error still there: server.engine.addFolder is not a function

erossignon, 2019-06-20:
the  ```addFolder``` method has been moved a while ago (in version 0.5.6) to the Namespace object.

```
const addressSpace = server.engine.addressSpace;
const namespace = addressSpace.getOwnNamespace();
const folder = namespace.addFolder(...);`
```