node-opcua/node-opcua #522: OPC UA connection error: ClientSecureChannelLayer => Socket is closed

adamgranted, 2018-09-16:
- ( x) bug report => search github for a similar issue or PR before submitting

**Current behavior**
When using node-red during development, multiple deploys will eventually cause the OPCUA socket to disconnect. I have to restart node-red when the socket closes and it will return to normal behavior.

**Expected behavior**
I would expect the socket to stay open regardless of number of re-deploys in node-red

**Minimal reproduction of the problem with instructions**
1. Have OPCUA Browser running in node red
2. Have multiple clients with auto inject for subscriptions
3. Keep redeploying 
4. Socket closes
5. Node red restart required
  
**What is the motivation / use case for changing the behavior?**
I may loose the socket connection once the system is installed permanently

**Please tell us about your environment:**
Using a Wago 750-8100 PLC with OPCUA features enabled. The PLC program is just passing along the variables of all the inputs over OPCUA. Raspberry PI 3+ is running node red, node red is running the OPCUA package. All inputs and outputs work perfectly fine until I deploy enough - since I am testing a lot right now I have been deploying often which seems to consistently cause the socket to close - somewhere around every 15 - 25 deployments.

  * (x ) I am using an application that uses node-opcua 
  
       - ( x) node-red
       - ( ) other : please specify 
        
  * Device: Macbook Air
  * OS version: OSX 10.13.6

   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: Wago PLC
     - Version: 750-8100
     
    
* **node-opcua version:**
0.2.32

* **Node:** 
   `node --version` = 8.11.4

>  Server State =  Running
 Server State =  Running
15 Sep 21:07:04 - [info] Stopping flows
15 Sep 21:07:04 - [info] Stopped flows
15 Sep 21:07:04 - [info] Starting flows
15 Sep 21:07:04 - [info] [OpcUa-Browser:9a37096.8e98af8] start browse client on opc.tcp://192.168.1.157:4840
15 Sep 21:07:04 - [info] Started flows
15 Sep 21:07:05 - [info] [OpcUa-Browser:9a37096.8e98af8] start browse session on opc.tcp://192.168.1.157:4840
15 Sep 21:07:05 - [warn] [OpcUa-Browser:9a37096.8e98af8] browseTopic:ns=0;i=85
BrowseResult {
  statusCode: { [Number: 0] value: 0, description: 'No Error', name: 'Good' },
  continuationPoint: null,
  references:
   [ ReferenceDescription {
       referenceTypeId: [Object],
       isForward: true,
       nodeId: [Object],
       browseName: [Object],
       displayName: [Object],
       nodeClass: [Object],
       typeDefinition: [Object] },
     ReferenceDescription {
       referenceTypeId: [Object],
       isForward: true,
       nodeId: [Object],
       browseName: [Object],
       displayName: [Object],
       nodeClass: [Object],
       typeDefinition: [Object] },
     ReferenceDescription {
       referenceTypeId: [Object],
       isForward: true,
       nodeId: [Object],
       browseName: [Object],
       displayName: [Object],
       nodeClass: [Object],
       typeDefinition: [Object] },
     ReferenceDescription {
       referenceTypeId: [Object],
       isForward: false,
       nodeId: [Object],
       browseName: [Object],
       displayName: [Object],
       nodeClass: [Object],
       typeDefinition: [Object] } ] }
15 Sep 21:07:05 - [warn] [OpcUa-Browser:9a37096.8e98af8] sending items 4
15 Sep 21:07:05 - [warn] [OpcUa-Browser:9a37096.8e98af8] close browse session
15 Sep 21:07:06 - [error] [OpcUa-Client:9917616f.a0cb]  OPC UA connection error: ClientSecureChannelLayer => Socket is closed !

adamgranted, 2018-09-16:
To add to this it seems that the more client connections I open to the PLC the more unstable the connection gets? I am going to try and wrap the data up in multiple arrays from the PLC before subscribing to it so its sending more data per subscription. Please advise? I currently have a total of 17 client connections and it has become incredibly unstable at this point. If the OPC server cannot handle I may have to look into a different solution.

I will also try staggering the connections with delays.

Here's the newest set of errors from the console:

> 16 Sep 12:28:49 - [info] Starting flows
16 Sep 12:28:49 - [info] Started flows
16 Sep 12:28:49 - [error] [OpcUa-Client:416fb27b.489eec]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
16 Sep 12:28:49 - [error] [OpcUa-Client:b00e84a3.bff9a8]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
16 Sep 12:28:49 - [error] [OpcUa-Client:46d68f46.59ae4]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
16 Sep 12:28:49 - [error] [OpcUa-Client:8b8a57d6.d128b8]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
16 Sep 12:28:49 - [error] [OpcUa-Client:235929e0.8c39e6]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
16 Sep 12:28:49 - [error] [OpcUa-Client:daf571fd.13369]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
16 Sep 12:28:51 - [error] [OpcUa-Client:2acae4d6.d9968c]  OPC UA connection error: ClientSecureChannelLayer => Socket is closed !
16 Sep 12:28:52 - [error] [OpcUa-Client:9917616f.a0cb]  OPC UA connection error: ClientSecureChannelLayer => Socket is closed !
16 Sep 12:28:52 - [error] [OpcUa-Client:8a6c895b.b5c728]  OPC UA connection error: ClientSecureChannelLayer => Socket is closed !
16 Sep 12:28:52 - [error] [OpcUa-Client:6f98c623.7d2798]  OPC UA connection error: ClientSecureChannelLayer => Socket is closed !
16 Sep 12:28:52 - [error] [OpcUa-Client:416fb27b.489eec]  OPC UA connection error: ClientSecureChannelLayer => Socket is closed !
16 Sep 12:29:06 - [error] [OpcUa-Client:9a06372.2cfe7c8]  OPC UA connection error: ClientSecureChannelLayer => Socket is closed !
16 Sep 12:29:06 - [error] [OpcUa-Client:816bcae5.0b9908]  OPC UA connection error: ClientSecureChannelLayer => Socket is closed !
16 Sep 12:29:06 - [error] [OpcUa-Client:5d75ac31.955ba4]  OPC UA connection error: ClientSecureChannelLayer => Socket is closed !
16 Sep 12:29:06 - [error] [OpcUa-Client:2fba23ee.3efd5c]  OPC UA connection error: ClientSecureChannelLayer => Socket is closed !
16 Sep 12:29:06 - [error] [OpcUa-Client:b00e84a3.bff9a8]  OPC UA connection error: ClientSecureChannelLayer => Socket is closed !

mikakaraila, 2018-09-16:
It can be that PLC runs communication on lower priority. I/O and controllers are on higher priority. Have you checked server capabilities? I try to run just few sessions, not so many as you.

adamgranted, 2018-09-16:
No I have not but I'll do some digging - I'm very new to OPCUA and am learning as I go. I did find that a reboot of the Wago PLC seems to have temporarily resolved it - all 17 client connections are working properly now. My concern is that if this gets implemented permanently that the connections will fail when its critical for them not to.

mikakaraila, 2018-09-16:
Then it can take time wago releases closed sockets... if it works after reboot.
Or it cannot reuse sockets.

adamgranted, 2018-09-16:
I've staggered the delays before client subscription, set the max TCP connections on the PLC to 100, and rebooted the PLC. So far so good but it doesn't help me narrow down which part is causing the socket to close

mikakaraila, 2018-09-17:
Have you used UaExpert server diagnostics? If there is some error or others counter behaving strangely...

thegeterrdone, 2019-03-05:
Hello, i am experiencing similar problem with OPC connection. I am using Siemens S7-1500 PLC and trying to build OPC connection.

 Below is my log:

5 Mar 10:21:15 - [info] [mosca in:6b908830.f475d8] Binding mosca mqtt server on port: 1883
5 Mar 10:21:15 - [info] Started flows
5 Mar 10:21:15 - [info] [mqtt-broker:ea96380e.9e66d8] Connected to broker: mqtt://localhost:1883
5 Mar 10:21:52 - [info] [mqtt-broker:ea96380e.9e66d8] Disconnected from broker: mqtt://localhost:1883
5 Mar 10:22:17 - [info] Stopping flows
5 Mar 10:22:17 - [info] [mosca in:6b908830.f475d8] Unbinding mosca mqtt server from port: 1883
5 Mar 10:22:17 - [error] [OpcUa-Client:b5f5075b.c1fca8]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:bd5238a9.60a218]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:d9c7714c.0d784]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:9d3db891.90d8e8]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:faa432d8.7cdd]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:a954e8b0.fbf1a8]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:9ee673cf.62ac2]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:487caa6f.31da54]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:e002993c.cf1cd8]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:539dd643.87a1d8]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:6c5c8649.654688]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:19e58ed6.82da81]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:2cf8a29c.5e2eee]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:b6a5470c.d1d458]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:ed02939a.78c17]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:4284daef.39e634]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:ec21e39a.2cfbd]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:4303f18.6c4861]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:620c7ea6.b7ab4]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:3432ab33.918c34]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:9ad7b00f.00974]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:1848b177.e83bdf]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:7c1ed129.bd11c]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:66a4ac78.4004b4]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:8698a624.ae29d8]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:a1307fe9.0f8b8]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:9d411f76.b183a]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:da164ee9.636f]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:7a113dda.42d664]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:60caa3b5.1437ac]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:81c9a7fd.68b268]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:c3a88715.45bc58]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
5 Mar 10:22:17 - [error] [OpcUa-Client:f493aa2f.67aa58]  OPC UA connection error: Timeout in waiting for data on socket ( timeout was = 10000 ms )
[... cut ...]
5 Mar 10:22:32 - [info] Stopped flows
5 Mar 10:22:32 - [info] Starting flows
5 Mar 10:22:32 - [info] [mosca in:6b908830.f475d8] Binding mosca mqtt server on port: 1883
5 Mar 10:22:32 - [info] Started flows
5 Mar 10:22:32 - [info] [mqtt-broker:ea96380e.9e66d8] Connected to broker: mqtt://localhost:1883
5 Mar 10:23:15 - [info] [mqtt-broker:ea96380e.9e66d8] Disconnected from broker: mqtt://localhost:1883

[image]

It was working fine before and since today i am experiencing this problem. Could there be any solution?

Regards

adamgranted, 2019-03-06:
> Have you used UaExpert server diagnostics? If there is some error or other counter behaving strangely...

No I have not, it has been a couple months since running this code if I find new info worth reporting, I'll reply back with an update.

thegeterrdone, 2019-03-06:
Thanks for reply!

closed 2020-03-12 by stale[bot]