node-opcua/node-opcua #525: npm test fails

spike0xff, 2018-09-18:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 
**Current behavior**
npm test output ends with a failed test:
1) Benchmarking Date conversion routines
     should check that slow and fast method produce same result:

    AssertionError: expected Array [ 30344869, 4043595776 ] to equal Array [ 30344869, -251371520 ] (at '1', A has 4043595776 and B has -251371520)
    + expected - actual

     [
       30344869
    -  4043595776
    +  -251371520
     ]

    at Context.<anonymous> (node-opcua-date-time\test\test_date_to_64bits_conversion.js:187:30)

**Expected behavior**
npm test to report complete success.

**Minimal reproduction of the problem with instructions**
This is what I actually did, I assume the npm audit had no effect and can be omitted...
git fetch
git pull
npm install
(this listed a critical security issue)
npm audit
npm test

  
**What is the motivation / use case for changing the behavior?**
well, either the test is incorrect or the code has a bug that is revealed by the failing test.

**Please tell us about your environment:**

  * (x) I have installed node-opcua from source ( using git clone)
  * ( ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: ___MacBook Pro__
  * OS version: __MacOS 10.13.6___
    
    - (x) Windows : version : _8.1 Pro N x64_
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:__KEPServerEX___
     - Version:___5______
     - Manufacturer/Software vendor:__Kepware_______
     - link : https://
     
    
* **node-opcua version:** ``
commit cf5381f507357c52ca17c554b24442525399ab00

* **Node:** 
   `node --version` = v10.8.0
npm -v =  6.2.0

erossignon, 2018-09-25:
is this error consistantly reproduceable ?
On my side I have tried to repeatedly run the test, without any issue.
Are you using a barebone MacOS or MacOs in a vm ?

``` 
$ cd packages\node-opcua-date-time
$ watch --color -n 0 mocha --retries 1000 test/test_date_to_64bits_conversion.js
```

erossignon, 2018-09-26:
@spike0xff
The test that is failing is comparing  a old DateTime opcua encoding method based on BigNumber package with faster one that we are now using.
the correct value= [ 30344828, 137254912 ],
What it the version of bignumber.js used on your side ?

```
$npm list  bignumber.js
`-- bignumber.js@7.2.1
```

erossignon, 2018-09-26:
OK I managed to reproduce with TZ=GMT+1
```
/node-opcua/packages/node-opcua-date-time$ TZ=GMT+1 mocha test -g "should check that slow and fast method produce same result"
```
```
  Benchmarking Date conversion routines
[ 30344844, -877189120 ] [ 30344844, 3417778176 ]
    1) should check that slow and fast method produce same result

  0 passing (31ms)
  1 failing

  1) Benchmarking Date conversion routines
       should check that slow and fast method produce same result:

      AssertionError: expected Array [ 30344844, 3417778176 ] to equal Array [ 30344844, -877189120 ] (at '1', A has 3417778176 and B has -877189120)
      + expected - actual

       [
         30344844
      -  3417778176
      +  -877189120
       ]

      at Assertion.fail (/mnt/c/sterfive/NodeOPCUA/node-opcua/node_modules/should/cjs/should.js:275:17)
      at Assertion.value (/mnt/c/sterfive/NodeOPCUA/node-opcua/node_modules/should/cjs/should.js:356:19)
      at Context.<anonymous> (test/test_date_to_64bits_conversion.js:192:34)
```

commit 52c80d91 (2018-09-26): fixes #525 - bn_dateToHundredNanoSecondFrom1601 slow and fast methods
cannot compare their encoding directly due to different 32bits word
storing scheme.
commit f9f437b0 (2018-09-26): fixes #525 - bn_dateToHundredNanoSecondFrom1601 slow and fast methods
cannot compare their encoding directly due to different 32bits word
storing scheme.
commit 6e80be56 (2018-09-27): fixes #525 - bn_dateToHundredNanoSecondFrom1601 slow and fast methods
cannot compare their encoding directly due to different 32bits word
storing scheme.
commit 9a7c6541 (2018-09-27): fixes #525 - bn_dateToHundredNanoSecondFrom1601 slow and fast methods
cannot compare their encoding directly due to different 32bits word
storing scheme.
closed 2018-09-26 by erossignon