node-opcua/node-opcua #526: npm install node-opcua-samples reports 12 errors

spike0xff, 2018-09-18:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting

**Current behavior**
after npm install node-opcua, I did
npm install node-opcua-samples
The output contains 12 occurrences of
ERROR :  Command failed: "openssl" verify
and
error 23 at 0 depth lookup: certificate revoked

**Expected behavior**
I expected that command to execute without error.

**Minimal reproduction of the problem with instructions**
(in my environment as detailed below)
mkdir node-opcua
cd node-opcua
npm install node-opcua
npm install node-opcua-samples
  
**What is the motivation / use case for changing the behavior?**
basic install of sample code should not throw mysterious errors.

**Please tell us about your environment:**
  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
        
  * Device: __Intel NUC (x86-64)_
  * OS version: __Resin OS 2.13.1+rev1__
    
    - ( ) Windows : version : _________
    - (x) Linux   : version : __4.12.12-yocto-standard (from uname -r)___
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   surely not relevant to this issue?
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:** ``
0.4.5

* **Node:** 
   `node --version` = 8.11.2

erossignon, 2018-09-18:
You can ignore this message,
When installing the samples, node-opcua creates all sorts of certificates (valid, invalid, revoked etc ...)
This message comes from some verification that a revoked certificate is ...revoked.

Nethertless, the message on the screen is confusing, and something need to be done about it.

erossignon, 2018-09-26:
new version of node-opcua-pki now fixes this issue.

closed 2018-09-26 by erossignon