node-opcua/node-opcua #527: Missing statusCode 1024 (0x00000400)

hcognet, 2018-09-21:
**I'm submitting a ...**
 - (x) bug report

**Current behavior**
Sometimes, some values returned by an OPC UA server have a statusCode equal to 1024 (decimal), 0x00000400 (hex).
This code is not included in the client's statusCode list. (raw_status_codes.js)

When we retrieve data with this statusCode, the operation returns:

DataValue {
  value:
   Variant {
     '$dataType': { [Number: 11] key: 'Double', value: 11 },
     '$arrayType': { [Number: 0] key: 'Scalar', value: 0 },
     value: 49.980000000000004,
     dimensions: null },
  **statusCode: undefined,**
  sourceTimestamp: 2018-09-20T13:24:27.639Z,
  sourcePicoseconds: 0,
  serverTimestamp: 2018-09-20T13:24:27.994Z,
  serverPicoseconds: 0 }

**Expected behavior**
The expected attitude is to recognize error code 1024 (0x00000400) as "Good" data (same as code 0)

In binary 0x00000400 is equal to 0b00000000000000000000**01**0000000000

**01** are the bits which represent the "InfoBits" (OPC UA Part4 specification).
Associated with **00** for LimitBits (Bit range 8:9), the data could be considered as good (some others clients already mark as "Good" this type of data)

**Minimal reproduction of the problem with instructions**
Request a value data which have a statusCode equals to 1024.
  
**What is the motivation / use case for changing the behavior?**
Having this new statusCode permits to have a statusCode defined in the DataValue structure.

**Please tell us about your environment:**

  * (x) I have installed node-opcua as a package ( using npm install )
        
  * Device: _Dell Latitude_
  * OS version: 
    - (x) Windows : version : 10
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:** `0.4.5`

* **Node:** 
   `node --version` = 8.12.0

erossignon, 2018-09-23:
Not all combination of StatusCodes can be described explicitly, additional quality bits are managed by the ModifiableStatusCode type that can have all extra bits set/unset.

erossignon, 2018-09-23:
You can create a Good status with Limit bit  this way:
```
       const dataValue = new DataValue({
            statusCode: StatusCodes.makeStatusCode(StatusCodes.Good, "LimitHigh")
        });
        encode_decode_round_trip_test(dataValue, function (buffer/*, id*/) {
            buffer.length.should.equal(5);
        });
```
when you display it 
```
        console.log(dataValue.toString());
```

it should produce:
```
DataValue:
   value:            <null>
   statusCode:      Good#LimitHigh (0x0000200)
   serverTimestamp: null
   sourceTimestamp: null
```

erossignon, 2018-09-23:
1024 = bit 10, correspond to InfoType bits set to DataValue:
```
       const dataValue = new DataValue({
            statusCode: StatusCodes.makeStatusCode(StatusCodes.Good, 1024)
        });

        console.log(dataValue.toString());

        console.log(" value in hexa = 0x",dataValue.statusCode.value.toString(16));
```

erossignon, 2018-09-25:
@hcognet , did this help ?

hcognet, 2018-09-27:
Hello @erossignon,

Yes, thanks for this information, it's very helpful.
However, I use the changed event with "subscription.monitor" in my OPC client.
So, when I can get access to DataValue, it's too late because the statusCode is already to undefined (or I don't see the way to keep the statusCode raw value).

Is there a possibility to extends the default statusCode list (on OPC client creation for example)?

Thanks!

erossignon, 2018-09-27:
@hcognet ,
I don't really get it , can you provide a example code ? 
What server are you trying to connect to ? 
Are you trying to monitor event changes ? on which node Id ? which event ?

hcognet, 2018-09-28:
@erossignon ,
I'm dealing with the piece of code below.
I'm quering a private CODRA Panorama OPC server on a node of measurement with the client monitoring subscription.

```
// install monitored item
const monitoredItem  = the_subscription.monitor({
        nodeId: opcua.resolveNodeId("ns=2;s=/Application/ROOT/[...]"),
        attributeId: opcua.AttributeIds.Value
    },
    {
        samplingInterval: 100,
        discardOldest: true,
        queueSize: 10
    },
    opcua.read_service.TimestampsToReturn.Both
);
console.log("-------------------------------------");

monitoredItem.on("changed", function(dataValue) {
   << ISSUE / DIFFICULTY HERE >>
   console.log(dataValue.statusCode.value);
});
```

<< ISSUE / DIFFICULTY HERE >> :
At this point, if the value of the statusCode (dataValue.statusCode.value) is not defined in the existing  StatusCode list, the StatusCode object is not defined in the DataValue structure as shown in my first message.
The next instruction (```console.log([...]```) raises an error.

erossignon, 2018-10-10:
The statusCode shall be defined in all cases ...  It may be possible that the codra system do not provide the statusCode information with event.
Make sure your code check if (dataValue.statusCode is defined before using it and assume statusCode is Good if it is not defined.

closed 2018-12-04 by erossignon