node-opcua/node-opcua #530: pkg + node-opcua

magres, 2018-09-25:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

I'm trying to build self executable binary sample_server on windows with pkg tool (https://www.npmjs.com/package/pkg)
But what I get:
`d:\projects\opcua>sample_server-win.exe
certificates/server_selfsigned_cert_2048.pem
certificates/PKI/own/private/private_key.pem
-----------------------------------------------------------
Error: Certificate file must exist :D:\snapshot\opcua\node_modules\node-opcua-server\src\certificates\server_selfsigned_cert_2048.pem
    at assert (D:\snapshot\opcua\node_modules\node-opcua-assert\dist\index.js:6:21)
    at OPCUAServer.OPCUASecureObject.getCertificateChain (D:\snapshot\opcua\node_modules\node-opcua-common\src\opcua_secure_object.js:61:9)
    at new OPCUAServer (D:\snapshot\opcua\node_modules\node-opcua-server\src\opcua_server.js:409:32)
    at Object.<anonymous> (D:\snapshot\opcua\sample_server.js:0:0)
    at Module._compile (pkg/prelude/bootstrap.js:1252:22)
    at Object.Module._extensions..js (module.js:661:10)
    at Module.load (module.js:563:32)
    at tryModuleLoad (module.js:503:12)
    at Function.Module._load (module.js:495:3)
    at Function.Module.runMain (pkg/prelude/bootstrap.js:1307:12)
-----------------------------------------------------------
D:\snapshot\opcua\node_modules\node-opcua-assert\dist\index.js:10
        throw err;
        ^

Error: Certificate file must exist :D:\snapshot\opcua\node_modules\node-opcua-server\src\certificates\server_selfsigned_cert_2048.pem
    at assert (D:\snapshot\opcua\node_modules\node-opcua-assert\dist\index.js:6:21)
    at OPCUAServer.OPCUASecureObject.getCertificateChain (D:\snapshot\opcua\node_modules\node-opcua-common\src\opcua_secure_object.js:61:9)
    at new OPCUAServer (D:\snapshot\opcua\node_modules\node-opcua-server\src\opcua_server.js:409:32)
    at Object.<anonymous> (D:\snapshot\opcua\sample_server.js:0:0)
    at Module._compile (pkg/prelude/bootstrap.js:1252:22)
    at Object.Module._extensions..js (module.js:661:10)
    at Module.load (module.js:563:32)
    at tryModuleLoad (module.js:503:12)
    at Function.Module._load (module.js:495:3)
    at Function.Module.runMain (pkg/prelude/bootstrap.js:1307:12)`
**Expected behavior**
getting working server

the working server. The path "D:\snapshot..." has nothing to do with my case.
**Minimal reproduction of the problem with instructions**

  1. config to turn off PKI
  2. config all the necessary paths
  3. ________________
  ...
  
**What is the motivation / use case for changing the behavior?**

Binary executable opc-ua server is a good feature
**Please tell us about your environment:**

Windows 10

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: Leneovo T470
  * OS version: Windows 10
    
    - (x) Windows : version : 10_________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: no
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:** ``

* **Node:** 
   `node --version` = 8.11.1

mikakaraila, 2018-09-25:
pkg uses own snapshot filesystem, you have to consider it.
Example how to use it:
var server_certificate_privatekey_file = path.join(path.dirname(process.execPath), "certificates/PKI/own/private/private_key.pem");

magres, 2018-09-28:
> 
> 
> pkg uses own snapshot filesystem, you have to consider it.
> Example how to use it:
> var server_certificate_privatekey_file = path.join(path.dirname(process.execPath), "certificates/PKI/own/private/private_key.pem");

Thank you. In order to add these certificates to snapshot filesystem I added:

`const path = require('path')
var p1 = path.join(__dirname, "certificates/server_selfsigned_cert_2048.pem");
var p2 = path.join(__dirname, "certificates/PKI/own/private/private_key.pem");
`
to node_modules\node-opcua-server\index.js and everything works out.

I think It makes sence to operate with files more obvious according to the pattern above.

closed 2018-09-28 by magres