node-opcua/node-opcua #531: Why i dont find client.off event?

mariodiciccio, 2018-09-28:
How i remove event clinet('on', ....) ?

 - ( ) bug report => search github for a similar issue or PR before submitting
 - (X) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

Dont exists the method

**Please tell us about your environment:**
MacOS 10.13.6
 
  * ( ) I have installed node-opcua from source ( using git clone)
  * (X) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
        
  * Device: MacBook Pro
  * OS version: 10.13
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - (X) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:** ``
All

* **Node:** 
   `node --version` = 10.9.0

erossignon, 2018-10-01:
Please refer to https://nodejs.org/api/events.html , OPCUAClient derives from nodejs EventEmitter.
Use removeListenner

mariodiciccio, 2018-10-02:
thank you so much

closed 2018-10-02 by erossignon