node-opcua/node-opcua #535: start opcua server with the xml file got error

ericccccccc, 2018-10-05:
**I'm submitting a ...**  (check one with "x")

 - (x ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

hi ~ i use FreeOpcUa modeler to generate the xml file of my model.
and i exec my server with the model xml , but it got the error ~ 

this is my server code :
[image]

this is my xml code generate by opcua-modeler  :
[image]

**Expected behavior**

when i start the server then got this error ~ 
[image]

is this means i can't use opcua-modeler to create the xml file of my information model ?
or something i miss  ?

thx for your help!

**Minimal reproduction of the problem with instructions**

  1. ________________
  2. ________________
  3. ________________
  ...
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: PC
  * OS version: _____
    
    - ( ) Windows : version : _________
    - (x ) Linux   : version : ubuntu14.04
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:** ``

* **Node:** 
   `node --version` =v8.12.0

erossignon, 2018-10-05:
Please provide a copy of your XML file, It may be ill-formed.

erossignon, 2018-10-05:
You also need to make sure that the standard opc namespace is loaded... 
Please uncomment this line and try again:
````javascript
nodeset_filename:[
  opcua.standard_nodeset_file,  /// this line is need
  "mynodeset.xml
];
```

ericccccccc, 2018-10-06:
thx your suggestions ~
i have loaded the stardard opc namespace too~
[image]

but i got another error~
[image]

erossignon, 2018-10-06:
NodeId must be in namespace 1 not 0 ;
Try to replace ```i=20001``` with ````ns=1;i=20001``` for all nodeid related to your new namespace.
There is a issue in the way this nodeset file has been generated with FreeOpcUa modeler, please post an issue to them.

ericccccccc, 2018-10-09:
thx~your help! 
Just like what u say, i have add the ns=1 to the nodeid by freeopcua ~ then fix the problems!!
[image]

[image]

but i'm confused bout what is the features of namespace ?
and how can i use the node opc-ua client to control the opc-ua server ? 
i have tried the only example bout sending data to client-side by server-side ~

i'm very thankful bout your help!!
help me a lot!!

closed 2018-10-10 by erossignon