node-opcua/node-opcua #537: opc-server write method problem

ericccccccc, 2018-10-10:
**I'm submitting a ...**  (check one with "x")

 - (x ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

i loaded my information model by generated xml, but i want to update the variable value every 5 seconds. i've found that the ServerEngine has "write" method ~
[image]

So i use this method like this ~
[image]

**Expected behavior**

then when i run the server,i got the error~
[image]

is something i missed or use the in the wrong way ?

erossignon, 2018-10-10:
value must be a DataValue like object: try this;
 ```javascript
const nodeToWrite = {
     nodeId: "ns=2;i=2001",
     attributeId: opcua.AttributeIds.Value,
     value : {
            sourceTimestamp: new Date(),
            statusCode: opcua.StatusCodes.Good,
            value: {
                 dataType: opcua.DataType.Int32 // check this type in your context
                  value: 42
             }
     } 
}
```
PS: note that write takes either a single node to write or an array of node to write: make sure you name nodeToWrite ( without s) in your case.

erossignon, 2018-10-10:
I have requalified the issue as a question / Support request rather than bug

ericccccccc, 2018-10-10:
i have tried what you say~ still get this problems
[image]

i think is that i didn't add this params with  " serverEngine Write method "?
[image]

but what is context ?
how to define it ?

erossignon, 2018-10-10:
Sorry, I didn't catch you're trying to set the value of a UAVariable Node from within a server.
In this case, you need to 
a/ get the variable node from its node id
b/ use setValueFromSource to change its value

``` javascript
const node = server.engine.addressSpace.findNode("ns=1,i=2001");
node.setValueFromSource({ dataType: opcua.DataType.Float, value: 4.2});
```

ericccccccc, 2018-10-11:
thx ! ! it works!!

closed 2018-10-11 by erossignon