node-opcua/node-opcua #539: how check the client is connected

mariodiciccio, 2018-10-11:
- ( ) bug report => search github for a similar issue or PR before submitting
 - ( x) feature request
 - ( ) support request => Please do not submit support request here, instead see use...

it is possible add opcuaClient.isConnected()?

thanks

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( x ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( x) MacOs   : version : 10.13.6
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :

erossignon, 2018-10-28:
```
const client = new opcua.OPCUAClient();
await client.connect(endpointurl);

if (client.isConnected) {
   console.log("client is connected"); 
} else {
   console.log("client may be trying to  reconnected"); 
}
```
alternatively , you can listen to some event that are raised when connection is changing

````
 client.on("connection_reestablished",function() {
            });
        client.on("backoff", function (retryCount, delay) {
         });
        client.on("start_reconnection", function () {
         });
```
in next version there will be also:
```
        client.on("connection_lost", function() {
         });
```

mariodiciccio, 2018-10-29:
thank you very much, i will wait next version

mariodiciccio, 2018-10-30:
Why the command client.isConnected return "undefined"?

panManfredini, 2019-04-02:
Hi,
it looks like "isConnected" is not a member of OPCUAClient. This would be actually an interesting feature to add.
My solution up to now is:
```javascript

client.isConnected = false;
client.on('close',()=>{ client.isConneted = false;});
client.on('connection_lost',()=>{client.isConneted = false;});
client.on("connection_reestablished", () => {client.isConneted = true;});

try{
   await client.connect(endpointUrl) ;
   client.isConnected= true;
}
catch(e){
// do something
}
```

erossignon, 2019-10-06:
In fact, `isConnected` flag applies to `session` object

closed 2019-10-06 by erossignon

nur455, 2023-07-18:
Property 'isConnected' does not exist on type 'OPCUAClient'

erossignon, 2023-07-18:
use  !client.isReconnecting instead