node-opcua/node-opcua #540: Configurable request timeouts

crussell52, 2018-10-12:
**I'm submitting a ...**  (check one with "x")

 - ( ) bug report => search github for a similar issue or PR before submitting
 - (x) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

Read value, write, and keep-alive requests take a hard-coded 60 seconds to time out.

**Expected behavior**

Various requests should have a configurable timeout.  The immediate needs are for read value, write value, and keep alive but the concept may be widely applied to any operation.

**Minimal reproduction of the problem with instructions**

  1.  Establish a connection
  2. Interrupt the network
  3. Perform a read-variable
  4. Observe the 60 second timeout
  
**What is the motivation / use case for changing the behavior?**

In my environment, the observed device and network is very performant and fast-feedback of a failed request is valuable.  Waiting 60 seconds for indication of a failure in an environment which has a typical response time of < 100 ms creates unnecessary delay in failure detection.

It is good that there are sensible defaults that cover a broad range of high latency systems.  However, the developer should be considered the 'expert' of the domain and have the ability to make adjustments where necessary.

Inspecting the source, I found that the timeouts are further clamped to a minimum of 30 seconds for any type of request (`ClientSecureChannelLayer._performMessageTransaction).  I would ask that if a developer chooses to specify a timeout (under this new feature), they should have full control over the range except in cases where it violates the OPC UA spec or is subject to technical limitations.

**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - (x) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:** ``

node-opcua@0.4.5

* **Node:** 
   `node --version` = 8.9.4

stevewillard, 2018-11-30:
@crussell52 Did you ever find a way to increase the timeout for a transaction?

erossignon, 2019-03-17:
@crussell52 , it is possible to  detect a lost of connection almost immediately by monitoring the on("connection_lost") event on the session object

closed 2020-03-23 by stale[bot]