node-opcua/node-opcua #542: how to use node-opcua-client-proxy ?

ericccccccc, 2018-10-16:
**I'm submitting a ...**  (check one with "x")

 - ( ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( x) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

hi ~ i want to transfer different protocol on opcua-client.
this is the structure i thought ~

                                                                                
node-opcua                                           node-opcua         --http-->            
   Server                <--tcp/bin-->               Client               --coap-->             Maybe Cloud/IoT Standard 
                                                                                           --mqtt-->

then i found there is a library called node-opcua-client-proxy,but doesn't have the example. 
how to use the proxy ?
can i use the proxy like what i thought ?

thx ~

erossignon, 2018-10-28:
@ericccccccc node-opcua-client-proxy provides a different API to access the addressSpace of a server and the object defined in the serve. It is not related to network related proxy.
There are some libraries to do coap/mqtt/http in nodejs and you will be able to design something that  (browse/read/subscribe to) an OPCUA Server with a node-opcua (tcp/bin protocol) client and redirect the relevant information somewhere else using  coap/mqtt/http.

ericccccccc, 2018-11-20:
thx your suggestion!!