node-opcua/node-opcua #543: Problem reading XML-definition with specialization of variable types

zenonpelle, 2018-10-18:
**I'm submitting a ...**  (check one with "x")

 - ( x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

**Expected behavior**

**Minimal reproduction of the problem with instructions**

  1. read the submitted XML file and the error is obvious
  2. ________________
  3. ________________
  ...
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( x)  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: surface pro 4_____
  * OS version: ____windows 10_
    
    - (x ) Windows : version : ___10______
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:**  :

* **Node:** 
   `node --version` = 0.4.6
[vtype.txt](https://github.com/node-opcua/node-opcua/files/2492188/vtype.txt)

0

erossignon, 2018-10-25:
How has this xml file be generated ?

zenonpelle, 2018-10-25:
I has been generated with the Atvise Scada software version 3.2.1.
https://www.atvise.com/en/

erossignon, 2018-10-25:
To complete the previous description the error is : 
```
-----------------------------------------------------------
Error:  UAVariableType already declared
    at assert (\node-opcua-45\packages\node-opcua-assert\dist\index.js:6:21)
    at _registerVariableType (\node-opcua-45\packages\node-opcua-address-space\src\namespace.js:191:5)

```
Investigation inside the provided xml file shows that "Anka" VariableType is declared twice.

[image]

[image]

The NodeSet.xml file is incorrect and shall be fixed.

zenonpelle, 2018-10-25:
Yes, it is the same name but different parents. This is possible in Atvise Scade and we are using this feature when we are importing a flat node structure from Kepserver,

erossignon, 2018-10-25:
OK, this make sense ... we shall do something about it

zenonpelle, 2018-10-25:
I see your point but this is just nodes in the tree as the variables are. The behavior is accepted in Atvise and UAExpert.  It might be hard to identify the types cause you must know their parent. But still I think this is allowed in OPCUA

erossignon, 2018-10-25:
Here is what I did to pursue the investigation using UAModeler from UnifiedAutomation.
UAModeler is **the** reference NodeSet editor & code generator and is used by the OPC Foundation to design the standard nodeset file ( such as the UA namesapce, DI , ADI, PLC namesapce etc )

  * I have tried to load the xml file inside UAModeler and it fails to load too inside UAModeler;
  * I then try to reproduce the Variable Type hierachie  inside UAModeler

As as expected, UAModeler refused to create a second Anke Variable Type....
[image]

Two ObjectType or VariableType cannot have the same name in the same namespace.

I would really recommend you to change the name of the VariableType to AnkaKalle and AnkaArne for instance; If you don't do so , this will create complication to other client for instance that would use automatique code generation of C++ / C# / Java classes based on the XML file as this will cause the code not to compile.