node-opcua/node-opcua #549: S7-1500 secure conversation is closed with simple_client.js

zombinary, 2018-10-30:
**I'm submitting a ...**  (check one with "x")

 - (X) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**
It is not possible to connect the S7-1500 OPC UA Server with an security mode.
Indepedent from the Security settings the PLC could not create a subscription over a long time.

i suspect that a tcp-package in the list 'request_data' cannot be assigned correctly by the 'client_secure_channel_layer' or the security token is not renewed.

```
client_secure_channel_layer   :223   request id =  7 Error: Security token has expired : tokenId 1
    at MessageBuilderBase._report_error (/opt/node-red/nodes/hilscher/opcuaclient/node_modules/node-opcua-transport/src/message_builder_base.js:221:24)
    at MessageBuilder._decrypt_MSG (/opt/node-red/nodes/hilscher/opcuaclient/node_modules/node-opcua-secure-channel/src/message_builder.js:266:14)
    at MessageBuilder._decrypt (/opt/node-red/nodes/hilscher/opcuaclient/node_modules/node-opcua-secure-channel/src/message_builder.js:338:21)
...
 message was
undefined
^C^C^C^C^C^C
^C^C^C^C message was 2: { /*PublishRequest*/
 requestHeader                 /* RequestHeader                    */: {
   authenticationToken         /* NodeId                           */: ns=0;i=843466748
   timeStamp                   /* UtcTime                          */: 2018-10-30T14:18:41.135Z
   requestHandle               /* IntegerId                        */: 8               0x8
   returnDiagnostics           /* UInt32                           */: 0               0x0
   auditEntryId                /* UAString                         */:
   timeoutHint                 /* UInt32                           */: 20000               0x4e20
   additionalHeader            /* ExtensionObject                  */: null
 }
 subscriptionAcknowledgements  /* SubscriptionAcknowledgement   [] */: [ /* empty */ ]
};
```

**Expected behavior**
The simple_client application could subcribe to the Curren Time on the S7 OPC UA Server.

**Minimal reproduction of the problem with instructions**
  1. enable the OPC UA Server in the S7-1500 PLC
  2. $ npm install node-opcua-samles@0.4.5
  3. $ cd node_modules/node-opcua-samples/bin
  4. $ DEBUG=ALL node simple_client -e opc.tcp://192.168.0.1:4840 -P=Basic256 -s=SIGN -n="ns=0;i=2258" --inspect=0.0.0.0:9229________________
  
**What is the motivation / use case for changing the behavior?**
It is Possible to connect the S7-1500 PLC with secure Settings.
currently you can only connect to none secure.

**Please tell us about your environment:**

  * (X) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (X) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: hilscher netIOT GW
  * OS version: yocto (the same behaviour is reproduce on a win32 machine)
    
    - (X) Windows : version : win7
    - (X) Linux   : version : yocto
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:Siemens OPC UA Server S7-1500
     - Version: Siemens S7-1500 CPU 1511-1 PN/6ES7 511-1AK01-0AB0/V2.0
     - Manufacturer/Software vendor: Siemens
     - link : https://www.siemens.com/global/de/home/produkte/automatisierung/systeme/industrie/sps/simatic-s7-1500.html
     
    
* **node-opcua version:**  :
node-opcua@0.4.5

* **Node:** 
   `node --version` = v8.11.4

mikakaraila, 2018-10-30:
I has similar kind of problem, but actual cause was the PLC did not update current time if it was subscribed. I has to read it to get new value. It can be that realtime Clock usage is optimized or minimized on PLC.

zombinary, 2018-11-01:
I found out that the behavior of Node-Opcua depends on the time in the PLC.
If the UTC time runs from server 1houre to (Siemens OPC UA Server: 7:00 o'clock, Client: 20:00 o'clock).
then node-opcua cannot establish a connection.
If the time deviates from <3min (server: 7:58 o'clock, client: 20:00 o'clock), it will be interrupted sporadically or not at all.
In comparison, the UaAnsiC server can establish and maintain the connection in all cases.

What I still don't understand is why a second GetEndpointRequest() should be used in the connection setup after sending the 2 HelloMessage.
I suspect that the bug is here and the client can't correctly assign the request from the server (which is being sent).
I haven't found anything here that would indicate that the timestamp in the response packets is being evaluated.
I'm not sure what this has to do with the time settings in the PLC?
```
This is what uaExpert done:
...
 --> Hello message
 <-- Acknowledge message
 --> OpenSecureChannel message: ServiceId0
 <-- OpenSecureChannel message: ServiceId0
 --> UA Secure Conversation Message: CreateSesseionRequest
 <-- UA Secure Conversation Message: (Message fragment 52)
 <-- UA Secure Conversation Message: [Malformed Packet]
 --> UA Secure Conversation Message: ActivateSessionRequest
 <-- UA Secure Conversation Message: ActivateSessionResponse
 --> UA Secure Conversation Message: ReadRequest
 <-- UA Secure Conversation Message: ReadResponse
 --> UA Secure Conversation Message: BrowseRequest
 <-- UA Secure Conversation Message: BrowseResponse
 ...
 
This is what node-opcua done:
 --> Hello message
 <-- Acknowledge message
 --> OpenSecureChannel message: ServiceId0
 <-- OpenSecureChannel message: ServiceId0
 --> UA Secure Conversation Message: GetEndpointRequest      | ?????
 <-- UA Secure Conversation Message: (Message fragment 52)
 <-- UA Secure Conversation Message: [Malformed Packet]
 --> OpenSecureChannel message: ServiceId0
 <-- OpenSecureChannel message: ServiceId0
 --> OpenSecureChannel message: ServiceId0
 <-- OpenSecureChannel message: ServiceId0
 ```

zombinary, 2018-11-02:
I found the Bug in the '_defineGetter_' of property expired.
The object 'createdAt' in the Server Response  is compared with the actual timestamp from the Client.
If the server time differs too much from the client time, then the client tries to update the secure token.

I have changed the condition to return the correct value.
Actual i could not download the source from git. I will create a pullrequest later.

**node-opcua-service-secure-channel/index.js**
```
/**
 * @property expired
 * @type {Boolean} - True if the security token has expired.
 */
ChannelSecurityToken.prototype.__defineGetter__("expired", function () {
    
    console.log('this.createdAt: '.red.bold, this.createdAt);
    console.log('Date: '.red.bold, Date());
    
    console.log('this.revisedLifeTime: '.red.bold, this.revisedLifeTime);
    console.log('this.createdAt.getTime(): '.red.bold, this.createdAt.getTime());
    console.log('Date.now(): '.red.bold, Date.now());
    
    //return (this.createdAt.getTime() + this.revisedLifeTime ) < Date.now();
    //console.log('(this.createdAt.getTime() + (this.createdAt.getTime()- Date.now()) + this.revisedLifeTime ) < Date.now(): '.red.bold, (this.createdAt.getTime() + (this.createdAt.getTime()- Date.now()) + this.revisedLifeTime ) < Date.now());
    var rslt = null;
    if(this.createdAt.getTime() < Date.now()){
      rslt = (this.createdAt.getTime() + (Date.now() - this.createdAt.getTime()) + this.revisedLifeTime ) < Date.now();
    }else{
      rslt = (this.createdAt.getTime() + this.revisedLifeTime ) < Date.now()
    }
    return rslt;
```

zombinary, 2018-11-05:
Unfortunately, I'm not making any progress with my solution.
Here the timestamp 'createdAt' is compared with the client system time. 
It is better to use the timestamp of the client's response, **not Date.now()**.

I don't know how to get to the ResponseHeader.Timestamp in context of Message_Builder.js (line 265), to pass this to the getter 'expired'?

[image]

mikakaraila, 2018-11-06:
Hmm, we have been using this with several S7-1500. Can you check Siemens firmware version?
Malformed packet was in one case caused by old OPC UA specification (old firmware).

zombinary, 2018-11-07:
I has download the actual Firmware v2.6.0 to the S7-1500 (update from: 2.0.5 => 2.6.0)
The Same Error Message will be return with a connection to UaAnsiC Server (V1.8.3)

With the actual node-opcua-samples@0.5.0 i got the message **BadCertificateUseNotAllowed**:

```
$node simple_client.js -e opc.tcp://192.168.0.1:4
840 -P=Basic256 -s=SIGN-n="ns=0;i=2258"
...
serviceResult               /* StatusCode                       */: BadCertificateUseNotAllowed (0x80180000)

````

mikakaraila, 2018-11-07:
Is PLC running something very fast? What about UaExpert, can it connect?

zombinary, 2018-11-12:
What do you mean with PLC running very fast?
I have no Modules configured in the PLC Project.  I used an empty Project with opcua-enable only.

Yes, UaExpert (v1.5.0 319) can connect the opcua-Server in the PLC without an Problem.

zombinary, 2018-11-14:
I **Solved** the Problem.
I think the problem is the certificate parameter ```Schluesselverwendung``` by creating the Certificate with ```node test_helpers/create_certificates.js certificate -s -o certificates/client_selfsigned_cert_1024.pem```.

[image]

I create the Certificate with my own openssl.cnf file and the openssl command and overwrite the certificate and key ```certificates/PKI/own/private/private_key.pem```, ```certificates/client_selfsigned_cert_1024.pem```:

```
#openssl.cnf
distinguished_name       = req_distinguished_name
default_md               = sha1
default_md                = sha256                      # The default digest algorithm

[ req ]
days                      = 365
req_extensions            = v3_req
#x509_extensions           = v3_ca

[ v3_req ]
basicConstraints       = CA:FALSE
#keyUsage               = critical, cRLSign, keyCertSign
keyUsage               = nonRepudiation, digitalSignature, keyEncipherment, dataEncipherment, keyCertSign
extendedKeyUsage       = critical, clientAuth, serverAuth
nsComment              = "SelfSigned Certificate generated by Node-OPCUACLIENT Certificate utility using openssl"
#subjectAltName        = $ENV::ALTNAME
subjectAltName          = URI:urn:NT0002A233E5F1:node-opcuaclient,DNS:node-opcuaclient

[ req_distinguished_name ]
countryName             = Country Name (2 letter code)
countryName_default     = DE
countryName_min         = 2
countryName_max         = 2
organizationName          = Organization Name (company)
organizationName_default  = node-opcuaclient
commonName                = Common Name (hostname, FQDN, IP, or your name)
commonName_max            = 256
commonName_default        = node-opcuaclient
emailAddress            = <email>
emailAddress_max        = 40
emailAddress_default    = <email>
#subjectAltName            = $ENV::ALTNAME
subjectAltName            = URI:urn:NT0002A233E5F1:node-opcuaclient,DNS:node-opcuaclient
```

```
  # Generate a 2048 bit RSA Key
  $ openssl genrsa -des3 -out key.pem 2048
  # remove passphrase from key
  $ openssl rsa -in key.pem -out key.pem
  $ openssl req -new -sha256 -newkey rsa:2048 -text  -extensions v3_req -config "./openssl.cnf" -key "key.pem" -out "cert.pem.csr" -subj "/CN=node-opcuaclient/O=<company>/OU=<department>/L=<city>/ST=<state>/C=DE"
  $ openssl x509 -req  -days 365 -extensions v3_req -extfile "openssl.cnf" -in "./cert.pem.csr" -signkey "./key.pem" -text  -out "./cert.pem" -text
  $ cp cert.pem ../node_modules/node-opcua-client/certificates/client_selfsigned_cert_1024.pem
  $ cp key.pem ../node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem
```

closed 2018-11-14 by zombinary