node-opcua/node-opcua #550: UserIdentityTokenType 4 is not supported yet

pramodwerea, 2018-11-02:
**I'm submitting a bug report or possible feature request to support for UserIdentityTokenType 4 **  (check one with "x")

 - ( x) bug report => search github for a similar issue or PR before submitting
 - ( x) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

We tested with pfiffner machine/server and  we have this problem when we run the simple_client.js it gives the following error

`Error: cannot  coerce value=4 to Enum
at _decode_enumeration (/Users/xxx/apps/node-opcua-sample/node_modules/node-opcua-factory/src/factories_enumerations.js:52:23)
at UserTokenPolicy.decode (/Users/xxx/apps/node-opcua-sample/node_modules/node-opcua-service-endpoints/_generated_/_auto_generated_UserTokenPolicy.js:139:22)
at /Users/xxx/apps/node-opcua-sample/node_modules/node-opcua-service-endpoints/_generated_/_auto_generated_EndpointDescription.js:181:12
at exports.decodeArray (/Users/xxx/apps/node-opcua-sample/node_modules/node-opcua-basic-types/src/array.js:40:18)
at EndpointDescription.decode (/Users/xxx/apps/node-opcua-sample/node_modules/node-opcua-service-endpoints/_generated_/_auto_generated_EndpointDescription.js:179:31)
at /Users/xxx/apps/node-opcua-sample/node_modules/node-opcua-service-endpoints/_generated_/_auto_generated_GetEndpointsResponse.js:105:12
at exports.decodeArray (/Users/xxx/apps/node-opcua-sample/node_modules/node-opcua-basic-types/src/array.js:40:18)
at GetEndpointsResponse.decode (/Users/xxx/apps/node-opcua-sample/node_modules/node-opcua-service-endpoints/_generated_/_auto_generated_GetEndpointsResponse.js:103:22)
at MessageBuilder._safe_decode_message_body (/Users/xxx/apps/node-opcua-sample/node_modules/node-opcua-secure-channel/src/message_builder.js:403:20)
at MessageBuilder._decode_message_body (/Users/xxx/apps/node-opcua-sample/node_modules/node-opcua-secure-channel/src/message_builder.js:450:18)`

   NOTE: with prosys opcua client it works

**Expected behavior**

It should connect to server and create session

**Minimal reproduction of the problem with instructions**

  1.  we are trying to connect OPC UA  based  pfiffner server
  2.  using simple_client.js code to test connection 
  3.  and getting error as shown above
  ...
  
**What is the motivation / use case for changing the behavior?**

we need to read data from pfiffner server

**Please tell us about your environment:**

  * (x ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device:  mac (also tested  with raspberry pi)
  * OS version: ios and liunx
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version :10.12.1
    - ( ) Raspbian: version :  3B+
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    NOTE: with prosys opcua client it works
* **node-opcua version:**  :

* **Node:** 
   `node --version` = 8.11.4

erossignon, 2018-11-04:
t looks like UserIdentityTokenType = 4 is not a valid value for UserIdentityTokenType

In node-opcua this enum is defined this way
```javascript
const EnumUserIdentityTokenType_Schema = {
  name: "EnumUserIdentityTokenType",
  enumValues: {
    ANONYMOUS: 0,
    USERNAME: 1,
    CERTIFICATE: 2,
    ISSUEDTOKEN: 3
  }
};
```

OPC Specification 1.04 specifies also only 4 possible values from 0 to 3

<opc:EnumeratedType Name="UserTokenType" LengthInBits="32">
   <opc:Documentation>The possible user token types. 
   </opc:Documentation>
  <opc:EnumeratedValue Name="Anonymous" Value="0" />
  <opc:EnumeratedValue Name="UserName" Value="1" />
  <opc:EnumeratedValue Name="Certificate" Value="2" />
  <opc:EnumeratedValue Name="IssuedToken" Value="3" />
</opc:EnumeratedType>

There seems to be an issue with the  ```pfiffner server``` .

This being said, node-opcua should probably handle this case more gracefully.

erossignon, 2018-11-04:
@pramodwerea could you fill with more details the section "Description of the other OPCUA system I am trying to connect to:"

pramodwerea, 2018-11-07:
Description of the other OPCUA system I am trying to connect to:

Name: Pfiffner
Version: Bosch Rexroth DAS Runtime Version : 02T05.12.0
Manufacturer/Software vendor:Bosch Rexroth AG,
ProductUri=urn:UnifiedAutomation:DataAnalyticsServerNet
Bosch Rexroth DAS Runtime Version : 02T05.12.0
link : http://documentation.unified-automation.com/uasdkdotnet/2.5.8/html/index.html

Following is the startup trace

`Following the startup trace:
09:25.43|0010|00|0|Log created at 11/7/2018 9:25:43 AM with TraceLevel Error
09:25.43|0010|00|0|Entry Assembly: DAS.Service, Version=2.5.1.0, Culture=neutral, PublicKeyToken=null
09:25.43|0010|00|0|Stack Assembly: UnifiedAutomation.UaBase, Version=2.0.0.0, Culture=neutral, PublicKeyToken=asdfasd
09:25.43|0010|01|0|MODULE LOADED [1] UnifiedAutomation.Stack
09:25.43|0010|02|0|MODULE LOADED [2] UnifiedAutomation.Server
09:25.43|0010|00|0|***VALID***UaSdkNetBundle-2.5.8-Win32-Binary -  ClientSDK enabled -  ServerSDK enabled - Sebastian Müller - Bosch Rexroth AG -
09:25.43|0010|03|0|MODULE LOADED [3] UnifiedAutomation.Client
09:25.45|0015|04|0|MODULE LOADED [4] UnifiedAutomation.ServerInternal`

Also getting following error while trying to connect and read node e.g. ns=0;i=431

`cannot decode message  for valid object of type ns=0;i=431 GetEndpointsResponse
 message was 
{ request: 
   GetEndpointsRequest {
     requestHeader: 
      RequestHeader {
        authenticationToken: [NodeId],
        timeStamp: [Date],
        requestHandle: 2,
        returnDiagnostics: 0,
        auditEntryId: null,
        timeoutHint: 0,
        additionalHeader: null },
     endpointUrl: '',
     localeIds: [],
     profileUris: [] },
  msgType: 'MSG',
  callback: [Function: modified_callback],
  bytesWritten_before: 189,
  bytesWritten_after: 258,
`

erossignon, 2018-11-24:
Hi,
I suggest you raise an issue to the provider of Pfiffner/Bosch Rexroth DAS Runtime Version : 02T05.12.0 to ask them what is the meaning of UserTokenType = 4 which doesn't not exist in the OPCUA Standard and let us know about the answer

closed 2019-12-01 by stale[bot]