node-opcua/node-opcua #552: Cannot set Enumeration Values

hamza765, 2018-11-09:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting

**Current behavior**

Trying to create a server where Enumeration variables can be added and be manipulated from a client, but writing from client fails. Even with test code.  

**Expected behavior**

I expect to be able to set a value from a OPCUA client. 

**Please tell us about your environment:**

  * (x )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x ) I have installed node-opcua as a package ( using npm install )

So I created my own server that adds variables from multiple namespaces. All datatypes except for Enums work. I can set Ints, Doubles, etc from a client. (I'm using Prosys). But when I try to set an enum, it says `datatype 1:MyEnumType2 must be registered` even though I have already registered it and it shows up in the Client with that nodeId. 

I thought it might have been my own code, but I tested it with test code from the `add_enumeration_type` test file and that gives me the same error:

```
function post_initialize() {
	console.log("initialized");

	function construct_my_address_space(server) {

		const addressSpace = server.engine.addressSpace;
		const namespace = addressSpace.getOwnNamespace();

		// declare a new object
		const device = namespace.addObject({
			organizedBy: addressSpace.rootFolder.objects,
			browseName: "MyDevice"
		});

		const myEnumType = namespace.addEnumerationType({
			browseName: "MyEnumType2",
			enumeration: ["RUNNING", "BLOCKED", "IDLE", "UNDER MAINTENANCE"]
		});
		const enumerationType = addressSpace.findDataType("Enumeration");

		const browseDescription = new BrowseDescription({
			referenceTypeId: null,
			browseDirection: opcua.BrowseDirection.Forward,
			resultMask: 0x3F
		});
		const r = enumerationType.browseNode(browseDescription, null);
		const names = r.map(function (x) {
			return x.browseName.toString();
		});

		const e = namespace.addVariable({
			propertyOf: device,
			dataType: myEnumType,
			browseName: "RunningState",

		});
		e.setValueFromSource({dataType: opcua.DataType.Int32, value: 1});
		console.log(e.readValue())
	}
	construct_my_address_space(server);
	server.start(function () {
		console.log("Server is now listening ... ( press CTRL+C to stop)");
		console.log("port ", server.endpoints[0].port);
		const endpointUrl = server.endpoints[0].endpointDescriptions()[0].endpointUrl;
		console.log(" the primary server endpoint url is ", endpointUrl);
	});
}
server.initialize(post_initialize);
```

```
EXCEPTION CAUGHT WHILE PROCESSING REQUEST !! WriteRequest
{ /*WriteRequest*/
 requestHeader                 /* RequestHeader                    */: {
   authenticationToken         /* NodeId                           */: ns=0;b=ca9c2369059d9241681630dbba1d4a82
   timeStamp                   /* UtcTime                          */: 2018-11-09T21:31:37.707Z
   requestHandle               /* IntegerId                        */: 22223               0x56cf
   returnDiagnostics           /* UInt32                           */: 0               0x0
   auditEntryId                /* UAString                         */: null
   timeoutHint                 /* UInt32                           */: 30000               0x7530
   additionalHeader            /* ExtensionObject                  */: null
 }
 nodesToWrite                  /* WriteValue                    [] */: [
   { /*0*/
     nodeId                    /* NodeId                           */: ns=1;i=1003
     attributeId               /* IntegerId                        */: 13               0xd
     indexRange                /* NumericRange                     */: NumericRange:<Empty>
     value                     /* DataValue                        */: {
       value                   /* Variant                          */: {
         dataType              /* DataType                         */: Int32 ( 6)
         arrayType             /* VariantArrayType                 */: Scalar ( 0)
         value                 /* Any                              */: 3
         dimensions            /* UInt32                        [] */: null []
       }
       statusCode              /* StatusCode                       */: Good (0x00000)
       sourceTimestamp         /* DateTime                         */: 2018-11-09T21:31:37.708Z
       sourcePicoseconds       /* UInt16                           */: 0
       serverTimestamp         /* DateTime                         */: 2018-11-09T21:31:37.708Z
       serverPicoseconds       /* UInt16                           */: 0
     }
   }
 ]
};
 display_trace_from_this_project_only = 
datatype 1:MyEnumType2 must be registered

 messages  EXCEPTION CAUGHT WHILE PROCESSING REQUEST !!! WriteRequest
datatype 1:MyEnumType2 must be registered
Error: datatype 1:MyEnumType2 must be registered
    at findBuiltInType (/Users/hamza/Projects/opcuaTest/node_modules/node-opcua-factory/src/factories_builtin_types.js:231:15)
    at validateDataType (/Users/hamza/Projects/opcuaTest/node_modules/node-opcua-address-space/src/ua_variable.js:133:23)
    at UAVariable._validate_DataType (/Users/hamza/Projects/opcuaTest/node_modules/node-opcua-address-space/src/ua_variable.js:625:12)
    at UAVariable.isValueInRange (/Users/hamza/Projects/opcuaTest/node_modules/node-opcua-address-space/src/data_access/ua_variable_data_access.js:28:15)
    at UAVariable.writeValue (/Users/hamza/Projects/opcuaTest/node_modules/node-opcua-address-space/src/ua_variable.js:815:29)
    at UAVariable.writeAttribute (/Users/hamza/Projects/opcuaTest/node_modules/node-opcua-address-space/src/ua_variable.js:895:18)
    at ServerEngine.writeSingleNode (/Users/hamza/Projects/opcuaTest/node_modules/node-opcua-server/src/server_engine.js:1142:13)
    at performWrite (/Users/hamza/Projects/opcuaTest/node_modules/node-opcua-server/src/server_engine.js:1172:16)
    at /Users/hamza/Projects/opcuaTest/node_modules/async/dist/async.js:1135:9
    at eachOfArrayLike (/Users/hamza/Projects/opcuaTest/node_modules/async/dist/async.js:1069:9)

```

erossignon, 2018-11-13:
bug confirmed

hamza765, 2018-12-20:
Any chance there's an update on this?

erossignon, 2018-12-21:
The fix is currently available in @next branch node-opcua@2.0.0-alpha.8 but has not been back ported to master.

erossignon, 2019-03-09:
Fixed in node-opcua@0.5.6

Graos, 2020-08-20:
I am using version 2.11.0 but i still see this issue. Do you think this fix is already resolved in 2.11.0 node-opcua package version?

commit f8b4c720 (2019-01-04): fixes #552 write issue with enumeration variable